jasonacox/Powerwall-Dashboard #252: Add solar self-consumption %

GrimmiMeloni, 2023-04-23:
**Problem**
Currently there is no indication of how much of the solar generated power is being self-consumed (vs. sent to the grid)

**Enhancement**
Similar to the % Solar Powered for the Home, I would like to also see another stat element that shows the percentage of the solar power being self consumed.
The query can be more or less taken 1:1 from the existing % solar powered element, just changing the values like this.

```
SELECT (solar - to_grid) / solar FROM 
  (SELECT integral("solar")  AS solar, integral("to_grid")  AS to_grid 
      FROM "autogen"."http" WHERE $timeFilter)
```

My main interest in this number stems from being interest in how the solar installation fares financially.

**Additional context**
Here's an example of how it looks locally for me (it's the 78.2%). I don't like that it messes with the overall alignment a bit. Maybe someone has a better idea on how to make this fancy.
[image]

jasonacox, 2023-05-08:
Thanks for this, @GrimmiMeloni !  I always struggle knowing how much of this we should include in the base install.  I tend to try to keep the base minimal, though I think it is fair to say that it is already busy.  However, I'm more than happy to add a new dashboard in our [Alternative Dashboard](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/dashboards#alternative-dashboards) folder if you want to submit a PR for this.

jasonacox-sam, 2026-09-06:
This proposed a solar self-consumption % stat (solar used on-site vs exported). Jason was supportive but wanted to keep the base dashboard lean, and invited it as an [Alternative Dashboard](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/dashboards) submission — no PR was ever filed.

Closing due to inactivity. The InfluxQL sketch in this issue is still valid, and alternative dashboard contributions remain very welcome — re-open with a PR if you build it.

— Sam 🌊

closed 2026-09-06 by jasonacox-sam