jasonacox/Powerwall-Dashboard #263: Move hard-coded avg cost per kWh into a variable

emptywee, 2023-05-01:
**Problem**
Every time there's an update to the Grafana dashboard(s) users need to go and change each query where 0.19 is used.

**Enhancement**
0.19 can be moved to variables and later used in queries making it easier to adjust after updating dashboards.

**Additional context**
Something like this can be added to the dashboard model and $avg_per_kwh can be used in queries where appropriate.
```
  "templating": {
    "list": [
      {
        "description": "Average Cost Per kWh",
        "hide": 2,
        "label": "kWh cost",
        "name": "avg_per_kwh",
        "query": "0.19",
        "skipUrlSync": false,
        "type": "constant"
      }
    ]
  }
```

jasonacox, 2023-05-02:
That's brilliant!  Do you have this working?  If so, would you be interested in submitting your `dashboard.json` updates as a pull request?

emptywee, 2023-05-02:
> 

I think so. I'll try to open a PR.

jasonacox, 2023-05-03:
For aesthetics, I vote "hidden" unless we can put it under the "Savings" graph.

closed 2023-05-03 by jasonacox

jgleigh, 2023-05-07:
@emptywee @jasonacox I mentioned this to Jason, but it'd be great if we could store off the two new buy/sell variables (similar to timezones) so people don't have to re-input them each they import an updated dashboard. Not sure the best way to handle that, but hopefully one of you will know.

emptywee, 2023-05-07:
Hmm,  the setup script can probably rewrite it just the same way as it does
for the time zone variable. Should be fairly easy to implement.

jasonacox, 2023-05-07:
The key tasks will be similar to timezone:
1. The `setup.sh` script ask user for timezone and stores this as local `tz` file and then uses [tz.sh](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/tz.sh) to update `dashboard.json` and the other files with timezones references.
2. The `upgrade.sh` process starts by reverting the above (also using [tz.sh](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/tz.sh)) and replacing the timezone values with the "default" value (`America/Los_Angeles`) so that the `git pull` will work.
3. Pulls latest files from repo (`git pull`)
4. Replaces the timezone values again with the "user selected" versions from local `tz` file.

It is possible, but I think this will be more challenging. Using the time zone variable example, the timezone string is unique enough that setting and unsetting it is trivial (search/replace `America/Los_Angeles` to `Australia/Sydney` and vice versa) but setting the costs values will be more challenging since they are just numbers.  It would be better if there is a way to have Grafana use env or external data source for variables. I couldn't find anything with a quick search, but someone else may know.

emptywee, 2023-05-07:
Hmm, yes, but the initial values in the dashboard.json file can be made
unique strings and then they'd get replaced by the scripts. That makes it a
little harder for those who don't use these scripts and import it manually.

In other words, we move the dashboard.json file in a subfolder, maybe
change its extension from .json to .json.tpl and generate importable
dashboard with the help of the setup scripts.

Another approach is store the values in the database and these variables
will be queried from the database instead of using constants you define
each time you import.

In other words, we add a few sql inserts for influxdb with initial values
(which can be adjusted by the setup script, or manually by folks who prefer
it this way) and then executed as part of installation. Grafana will query
for them and use as variables.

I think the second approach is more favorable for users. However, we'll
need a convenient way to change these values in the database, since prices
change from time to time. Maybe, we'll need to add a new script that
manages it or incorporate it into some other, already existing script.

PR #266 (2023-05-03): Make costs/credit per kWh and timezone to be variables in Grafana
This closes #263 

However, I do not differentiate between costs and credits here, assuming both are same. I do understand that buying electricity may include delivery costs, which you probably don't get compensated for when you export electricity, so as part of further improvements, we can set up two variables: price per kWh when you import (buy) and price per kWh when you export (sell) and use them accordingly in the queries. This is up for discussion.

Now, when you import a dashboard, you get prompted like this:

[image]

Se [... cut]