jasonacox/Powerwall-Dashboard #273: Backup script improvements

YesThatAllen, 2023-05-07:
**Problem**

The existing backup script requires users to make their own copies, and as such isn't as easy as it could be, and changes to the process aren't reliably enforceable.

**Enhancement**

- Use user-definable variables vs requiring users to copy & make changes. 
- Use variables like `$0` to find the path to the repo, eg `powerwall_repo="$0"`
- Make the script work regardless of where it's called from ( eg prefix all paths with `$powerwall_repo`)
- use `mkdir -p` to ensure folders exist
- limit the need for `sudo`... invoking sudo within the script as needed is safer than running the whole script as root.

**Additional context**

I'd start with moving the existing `backups/backup.sh.example` to the repo root as `backup.sh` so that it's at the same level as other tooling, and to prevent conflicts with any existing tooling people have made

jasonacox, 2023-05-08:
The backup script and process has been problematic for a while. Ideally it is not self contained w/in the project folder but is getting backed up to a separate partition or remote storage. But in any case, suggestions here are valid and improvements need to be made. I would welcome any help. 😁

YesThatAllen, 2023-05-08:
> Ideally it is not self contained w/in the project folder 

We'll have a more unified experience if the script itself is in the repo

> but is getting backed up to a separate partition or remote storage.

Absolutely. I envision a file where users define all sorts of variables.. timezone, backup destination, etc.
pypowerwall.env currently hosts some values including `TZ`, is it fair to say that's where we can put things like 

BACKUP_DESTINTATION
BACKUP_DAYS_TO_KEEP

etc?

the nice part about how you laid out that file already is that we can call:
pseudocode here:

```bash
pypowerwall_repo_home=$( cd -- "$( dirname -- "${BASH_SOURCE[0]}" )" &> /dev/null && pwd )
pypowerwall_env_file="$pypowerwall_repo_home/pypowerwall.env"

echo "pypowerwall_env_file is $pypowerwall_env_file"

if [[ -f $pypowerwall_env_file ]]; then
  source "$pypowerwall_env_file"
else
  echo "You must have messed up somewhere."
fi
```

Which will make the values stored there available in scripts like `$TZ`