jasonacox/Powerwall-Dashboard #276: Time of use beta - request for comment/feedback

BuongiornoTexas, 2023-05-13:
Hi All, 

At long last, I have a first pass time of use engine up and running and would like some comment/feedback on it. It is still a work in progress, but I can already create output like this without problem:

[image]

If you are interested, my development branch is at https://github.com/BuongiornoTexas/pypowerwall/, and more importantly, the development README is at https://github.com/BuongiornoTexas/pypowerwall/blob/main/proxy/REAME-usage.md.

Right now, the documentation is mostly workable for setting up a test server, but doesn't cover the details of the configuration file (I'll do it in the next few days and will put an update in this issue - I'm pushing this out today to give anyone who wants to play with it a bit of weekend time). But you can get a reasonable idea what is going on by looking at the example `usage.json` file. 

If you are feeling enthusiastic (or brave depending on your perspective), you can give it a try immediately by following the test server steps in the README and editing the influx_url and timezone fields in the `usage.json` file.

**BIG NOTE** The engine does not modify any influx data. It reads data from influx, performs usage calculations and passes the output to grafana.

BuongiornoTexas, 2023-05-13:
Discussions and issues that this may be relevant to: #244, #225, #68 and #82,  (post once so people following those can pick up here if interested).

jasonacox, 2023-05-13:
Very cool @BuongiornoTexas !  Great progress!

I have one suggestion for consideration... I see you embedded TOU computations and data feed into the pypowerwall proxy server, which makes sense for speed. However, this feels a be a bit outside the scope of pypowerwall (a simple Powerwall proxy) which doesn't use or need influxdb and other things but you would need for TOU computation.  It would make more sense for us to add another microservice that is focused on just computing TOU data in context of our `Powerwall-Dashboard` project.  We can add to our docker compose definition (similar to weather411) and it can be developed, updated and expanded outside of impact to pypowerwall.

BuongiornoTexas, 2023-05-13:
> It would make more sense for us to add another microservice that is focused on just computing TOU data.

@jasonacox - Absolutely no problem doing this - it makes my development effort a lot easier to focus on the single function of ToU rather than wedging it into pypowerwall. 

However, the reason I went down this path was not speed, but because I assumed people would prefer to minimise the number of services hanging on their Powerwall Dashboard hosts. This would add another proxy server, this time between InfluxDB and grafana. That said, the overhead should not be huge, as I'd expect it to spend a lot of time idle - I expect most people aren't interested in cost data at the minute level, and probably not even hourly (which is why I wasn't thinking about speed).

I don't have a view on this, so if it's your preference (from the above, I'm assuming so), I'll switch to a dedicated ToU service. 

And, if this is your preference - can you point me at a tutorial for creating docker images for python services (which I'm hoping will be less painful than pandas).

BJReplay, 2023-05-14:
> We can add to our docker compose definition (similar to weather411) and it can be developed, updated and expanded outside of impact to pypowerwall.

And, further to that, there's already a way of doing this without breaking the standard approach - see https://github.com/jasonacox/Powerwall-Dashboard/tree/main/weather/contrib/ecowitt#readme that describes using powerwall.extend.yml to add a new component that is picked up by compose-dash.sh

> And, if this is your preference - can you point me at a tutorial for creating docker images for python services (which I'm hoping will be less painful than pandas).

Again, have a look at https://github.com/jasonacox/Powerwall-Dashboard/tree/main/weather/contrib/ecowitt as an example (not necessarily a good example - this is me learning python, so it is probably a good anti-pattern), but is an example of a python service.

Jason has kindly published the docker image (the script is upload.sh in that folder), but I've since signed up to docker and could probably publish the docker image myself.

BuongiornoTexas, 2023-05-14:
Thanks Bernard - I'll follow up later in the week.

BuongiornoTexas, 2023-05-17:
`usage.json` is now detailed in the documentation. 

I'll work on a stand-alone micro-service over the next few days/weeks.

BJReplay, 2023-05-17:
> `usage.json` is now detailed in the documentation.

Great - having a look, I had a question, where I think I know the answer:

You have said that each plan **must** have a name.

I think (but it isn't explicitly stated) that each plan **must** have **at least one** season (and, as you've said, each plan can contain an arbitrary number of "seasons", provided all seasons in the plan have share the same basic tariff schedule structure).

Further, I suspect that each season **must** have **at least one** tariff schedule, and each schedule **must** have **at least one** period.

Would you like PRs on the documentation, or just comments? I could fork your repo and do a PR, but I'm guessing at the rules rather than reading the code here ;)

BuongiornoTexas, 2023-05-18:
> Great - having a look, I had a question, where I think I know the answer:

I've pushed the update. As I'm now going to move to the new repo, I'll accumulate additional comments and address them when I'm done with the move.

BJReplay, 2023-05-18:
> I've pushed the update.

Looks good.

BuongiornoTexas, 2023-05-23:
The new repo is now up at: https://github.com/BuongiornoTexas/PW-Dashboard-usage-proxy

I've stripped back the server module considerably and done a bit of tidy up elsewhere. The last major steps are cleaning up aggregation, sorting out the docker image, and doing an improved example dashboard. 

For anyone moving from the old version, notes and changes:

- Documentation continues to improve slowly.
- `usage.json` structure is unchanged, but now has full support for loading from a file specified in the environment variable `USAGE_JSON` - I'd suggest using this and keeping the configuration file outside the python folder, as it will make the transition to
a docker service much less painful later.

BuongiornoTexas, 2023-05-31:
The latest version is now up at https://github.com/BuongiornoTexas/pwdusage.

Unfortunately, yet another name change in preparation for python packaging (hopefully the last time) and then dockerisation.  

The readme outlines new features, with the main thing being sensible resampling and the ability to return summary numbers only. This runs dramatically better than my flux based concept, returning 8 months of data in about a second or two, 90 days very quickly, and shorter intervals almost instantly.

The next version will break the current file structure, but json configuration files should not be impacted (on the upside, the next version should be installable from pypi). 

I'm putting improved dashboards to the back of the queue on the assumption most users would prefer a docker container as a first priority. If anyone has different views, please let me know.

BuongiornoTexas, 2023-06-01:
@jasonacox - A quick query on this - As I'm doing the packaging, I've realised the grafana dashboards don't really fit well as part of the repo (as in, they should not be installed as part of the python package). Are you happy to host these in a subdirectory of the Powerwall-Dashboard `dashboard` folder? If not, I'll set up a tiny repo of my own to hold them.

jasonacox, 2023-06-02:
Absolutely!  We can set up a folder for anything you want in [/tools](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools) (pick a folder name that is descriptive, maybe savings-service, utility-cost-service, time-of-use, pwdusage, etc?).  You're also welcome to put your code or other things for the microservice there too.

BuongiornoTexas, 2023-06-02:
> Absolutely! We can set up a folder for anything you want in [/tools](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools)

Excellent - thanks! I think the things to drop in will be:
- Sample configuration file for the microservice, 
- Sample dashboards. 
- Maybe the dockerfile, as it doesn't really belong in the python package?

I'll do a PR in due course.

BuongiornoTexas, 2023-06-05:
Microservice now at v0.9.4 now available as a package on pypi. https://github.com/BuongiornoTexas/pwdusage for more details.

The core system is  feature complete so changes up to v1.0 will be bug fixes and documentation cleanup.

Moving on to containerisation now.

BuongiornoTexas, 2023-06-11:
Powerwall-Dashboard usage service is now v1.0 – [https://github.com/BuongiornoTexas/pwdusage](https://forums.whirlpool.net.au/go?https%3A%2F%2Fgithub.com%2FBuongiornoTexas%2Fpwdusage)

When @jasonacox approves the PR, all of the Powerwall-Dashboard setup components will be available from the tools/usage-service folder of the main Powerwall-Dashboard repo. If this takes a while and you want to experiment now, you can get them from my fork in the interim:

[https://github.com/BuongiornoTexas/Powerwall-Dashboard/tree/main/tools/usage-service](https://forums.whirlpool.net.au/go?https%3A%2F%2Fgithub.com%2FBuongiornoTexas%2FPowerwall-Dashboard%2Ftree%2Fmain%2Ftools%2Fusage-service)

@jasonacox, can I suggest we keep this issue open for a bit for feedback on the micro-service? Happy to close it if you would prefer issues go straight to the pwdusage repo.

BuongiornoTexas, 2024-01-06:
No follow up for 6 months. Closing as resolved.

closed 2024-01-06 by BuongiornoTexas