jasonacox/Powerwall-Dashboard #279: Weather/temperature stopped showing up.

Jot18, 2023-05-14:
As of yday, the temperature and wind has stopped working and isn't showing any data. Can anyone else confirm if they have this issue? I have solar only.

[image]

YesThatAllen, 2023-05-15:
I'm seeing the same thing today.

I just restarted the computer I run Powerwall-Dashboard on. I'm seeing data in the other areas, but nothing in `Temperature, Humidity and Clouds` or `Wind, Pressure and Precipitation`

YesThatAllen, 2023-05-15:
I'm seeing the same thing today.

I just restarted the computer I run Powerwall-Dashboard on. I'm seeing data in the other areas, but nothing in `Temperature, Humidity and Clouds` or `Wind, Pressure and Precipitation`

It looks like they've moved from API 2.5 to 3.0.

https://docs.openweather.co.uk/api/one-call-3

So this setup will have to do the same.

mcbirse, 2023-05-16:
Hi @Jot18 & @YesThatAllen 

For myself, everything still appears to be working as normal.

> It looks like they've moved from API 2.5 to 3.0.

OpenWeather have not moved to the OneCall 3.0 API for the free calls to get current weather, from what I can tell.

The 3.0 API has been around a while and is their paid subscription plan that gives you additional information (e.g. historical weather data + granular forecast weather data down to per minute etc.)

The 3.0 API should not be required for the weather411 service to work. The [current OpenWeather API documentation](https://openweathermap.org/current) is still showing the 2.5 API call to get current weather data for free accounts.

Can you test the below by browsing directly to this link (replace "your_api_key" with your own API Key) and see if it is working?

https://api.openweathermap.org/data/2.5/weather?lat=-33.868&lon=151.207&units=metric&appid=your_api_key

You should get a response similar to below:

```json
{"coord":{"lon":151.2068,"lat":-33.8715},"weather":[{"id":800,"main":"Clear","description":"clear sky","icon":"01d"}],"base":"stations","main":{"temp":20.34,"feels_like":19.71,"temp_min":18.6,"temp_max":23.12,"pressure":1017,"humidity":49},"visibility":10000,"wind":{"speed":6.17,"deg":170},"clouds":{"all":0},"dt":1684214913,"sys":{"type":2,"id":111111,"country":"AU","sunrise":1684183231,"sunset":1684220563},"timezone":36000,"id":1111111,"name":"Sydney","cod":200}
```

I have just signed up with another free account to test this myself, however need to wait a few hours before active.

It is possible my current account is still working (for both 2.5 and 3.0 calls) as I have a paid OneCall 3.0 API subscription, which I use with the weather-history tool to get historical weather data (I use the same API key for both 2.5 & 3.0 calls).

I wonder if OpenWeather could have possibly broken something for free accounts. I received an e-mail from them about upcoming new features they will be adding (snippet below), so it is possible I suppose if they are making changes.

> Hello, it’s OpenWeather!
> 
> We are pleased to inform you that the new feature will be released soon in One Call API 3.0 and will be available for all product users by default.
> 
> We will add the proper human-readable text description of the daily forecast, for example, “The day will start with rain through the late morning hours, transitioning to partly cloudy”.
>

mcbirse, 2023-05-16:
Update: I just re-tested this with a newly created free account and calls to the 2.5 API are working correctly. This is the API call used by the weather411 service, e.g.:

https://api.openweathermap.org/data/2.5/weather?lat=-33.868&lon=151.207&units=metric&appid=your_api_key

If it is not working for you, perhaps try generating a new API key for your account at the link below?

https://home.openweathermap.org/api_keys

After waiting a couple of hours for it to become active, re-test above and if it is working, then edit the `weather411.conf` file and replace APIKEY with the new one.

Finally, restart the weather411 docker container and hopefully you should see weather data again? (you may need to wait 10mins or so)

```bash
# Restart weather411
docker restart weather411
```

Other items to try - you could turn on debug output of weather411 and check the logs.

Edit `weather411.conf` and set `DEBUG = yes`, then:

```bash
# Restart weather411
docker restart weather411
```

```bash
# View weather411 log output
docker logs -f weather411
```

The debug output should show the calls to the OpenWeather API and writes to InfluxDB if working correctly (and perhaps some clues as to what is wrong if not working correctly).

```
DEBUG:https://api.openweathermap.org:443 "GET /data/2.5/weather?lat=xxxx&lon=xxxxx&units=metric&appid=xxxxxxx HTTP/1.1" 200 511
DEBUG:Weather data loaded
DEBUG:Writing to InfluxDB
```

YesThatAllen, 2023-05-16:
In my case, my `weather/weather411.conf` file had what looked to be a typo...

```
# Enter your location in latitude and longitude
LAT = 30.1234567
LON = -91.1234567,15
```

probably something from when I copied the info from google maps.

I fixed that by removing the `,15` and used upgrade.sh to get to the recently released 2.9.8… et voilà!.. 
weather is showing again.

Jot18, 2023-05-20:
> Update: I just re-tested this with a newly created free account and calls to the 2.5 API are working correctly. This is the API call used by the weather411 service, e.g.:
> 
> https://api.openweathermap.org/data/2.5/weather?lat=-33.868&lon=151.207&units=metric&appid=your_api_key
> 
> If it is not working for you, perhaps try generating a new API key for your account at the link below?
> 
> https://home.openweathermap.org/api_keys
> 
> After waiting a couple of hours for it to become active, re-test above and if it is working, then edit the `weather411.conf` file and replace APIKEY with the new one.
> 
> Finally, restart the weather411 docker container and hopefully you should see weather data again? (you may need to wait 10mins or so)
> 
> ```shell
> # Restart weather411
> docker restart weather411
> ```
> 
> Other items to try - you could turn on debug output of weather411 and check the logs.
> 
> Edit `weather411.conf` and set `DEBUG = yes`, then:
> 
> ```shell
> # Restart weather411
> docker restart weather411
> ```
> 
> ```shell
> # View weather411 log output
> docker logs -f weather411
> ```
> 
> The debug output should show the calls to the OpenWeather API and writes to InfluxDB if working correctly (and perhaps some clues as to what is wrong if not working correctly).
> 
> ```
> DEBUG:https://api.openweathermap.org:443 "GET /data/2.5/weather?lat=xxxx&lon=xxxxx&units=metric&appid=xxxxxxx HTTP/1.1" 200 511
> DEBUG:Weather data loaded
> DEBUG:Writing to InfluxDB
> ```

-Docker restart weather411 did the trick. It's updating now. Didn't change anything else. :)

mcbirse, 2023-05-20:
> -Docker restart weather411 did the trick. It's updating now. Didn't change anything else. :)

Good to hear @Jot18 - it must have gotten into a bad state or something? Very strange.

I wonder if the logs showed anything?

```bash
# View weather411 logs
docker logs -f weather411
```

Jot18, 2023-05-20:
> > -Docker restart weather411 did the trick. It's updating now. Didn't change anything else. :)
> 
> Good to hear @Jot18 - it must have gotten into a bad state or something? Very strange.
> 
> I wonder if the logs showed anything?
> 
> ```shell
> # View weather411 logs
> docker logs -f weather411
> ```

I tried to find errors but couldn't really see anything wrong....If it happens again, I'll check again.

closed 2023-05-20 by Jot18