jasonacox/Powerwall-Dashboard #286: Solar-only history is saving zero values when the API returns no data

youzer-name, 2023-06-01:
@mcbirse 

I've been running the solar-only history in a Docker container for about a week now. I noticed that yesterday and today there were zero kWh production values during times where there shouldn't have been.

Just now I saw this:

[image]

The yellow line is based on the tesla-history solar-only data. The orange line is the total solar production seen by my Gateway from both of my solar arrays. As you can see, the production didn't actually drop out. If it had, the orange line would also have dropped.

The raw influx data shows the zero point at 13:30Z:
```
2023-06-01T10:00:00Z 0         0       0    Jun   0     cloud  0       0     2023
2023-06-01T10:15:00Z 0         0       0    Jun   160   cloud  0       0     2023
2023-06-01T10:30:00Z 0         0       0    Jun   200   cloud  0       0     2023
2023-06-01T10:45:00Z 0         0       0    Jun   240   cloud  0       0     2023
2023-06-01T11:00:00Z 0         0       0    Jun   280   cloud  0       0     2023
2023-06-01T11:15:00Z 0         0       0    Jun   320   cloud  0       0     2023
2023-06-01T11:30:00Z 0         0       0    Jun   440   cloud  0       0     2023
2023-06-01T11:45:00Z 0         0       0    Jun   480   cloud  0       0     2023
2023-06-01T12:00:00Z 0         0       0    Jun   520   cloud  0       0     2023
2023-06-01T12:15:00Z 0         0       0    Jun   1240  cloud  0       0     2023
2023-06-01T12:30:00Z 0         0       0    Jun   2000  cloud  0       0     2023
2023-06-01T12:45:00Z 0         0       0    Jun   2040  cloud  0       0     2023
2023-06-01T13:00:00Z 0         0       0    Jun   2920  cloud  0       0     2023
2023-06-01T13:15:00Z 0         0       0    Jun   3280  cloud  0       0     2023
2023-06-01T13:30:00Z 0         0       0    Jun   0     cloud  0       0     2023
2023-06-01T13:45:00Z 0         0       0    Jun   3840  cloud  0       0     2023
2023-06-01T14:00:00Z 0         0       0    Jun   4040  cloud  0       0     2023
2023-06-01T14:15:00Z 0         0       0    Jun   0     cloud  0       0     2023
2023-06-01T14:30:00Z 0         0       0    Jun   0     cloud  0       0     2023
```
I just re-ran the download manually with the --force option and the debug output shows this:

```
        {
            "timestamp": "2023-06-01T08:45:00-04:00",
            "solar_power": 2040,
            "battery_power": 0,
            "grid_power": 0,
            "grid_services_power": 0,
            "generator_power": 0
        },
        {
            "timestamp": "2023-06-01T09:00:00-04:00",
            "solar_power": 2920,
            "battery_power": 0,
            "grid_power": 0,
            "grid_services_power": 0,
            "generator_power": 0
        },
        {
            "timestamp": "2023-06-01T09:15:00-04:00",
            "solar_power": 3280,
            "battery_power": 0,
            "grid_power": 0,
            "grid_services_power": 0,
            "generator_power": 0
        },
        {
            "timestamp": "2023-06-01T09:45:00-04:00",
            "solar_power": 3840,
            "battery_power": 0,
            "grid_power": 0,
            "grid_services_power": 0,
            "generator_power": 0
        },
        {
            "timestamp": "2023-06-01T10:00:00-04:00",
            "solar_power": 4040,
            "battery_power": 0,
            "grid_power": 0,
            "grid_services_power": 0,
            "generator_power": 0
        },
```
It has no data point for 09:30 local time / 13:30Z.

The chart in the Tesla phone app doesn't show this as a dropout. When I use the "Download my data" from the phone app, it is also missing the data for that interval but it doesn't show a zero:
```
2023-06-01T07:45:00-04:00	0.5
2023-06-01T08:00:00-04:00	0.5
2023-06-01T08:15:00-04:00	1.2
2023-06-01T08:30:00-04:00	2
2023-06-01T08:45:00-04:00	2
2023-06-01T09:00:00-04:00	2.9
2023-06-01T09:15:00-04:00	3.3
2023-06-01T09:45:00-04:00	3.8
2023-06-01T10:00:00-04:00	4
```

It seems the API returns zero values when there is no production. Is there a way to get the history script to only write values for the times that there is data returned?

youzer-name, 2023-06-01:
@mcbirse 

I also just realized I can deal with this by adding 'where solar > 0' to my queries, ignoring all the zero values whether they occur at night or due to the missing data points, so if fixing it is complicated and you don't see any other issues it is causing, it may not need attention.

mcbirse, 2023-06-02:
Hi @youzer-name - the script only writes datapoints for the timestamps returned.... it is odd that you have ended up with some zero datapoints.

First off - easy fix is as follows:

Use the tesla-history script with the "--remove" option to remove data between the --start & --end time range where you have zero data.

Then, run the script as normal to retrieve history for that period and it should fill in the missing data.

Let us know if this fixes it and you no longer see zero data points? (if you do, it means the API is returning a timestamp with zero value data points, however from your debug output and testing above, it appears this shouldn't be the case).

So, as to why you ended up with zero value datapoints in the first place... I'll have to speculate on this:

I noticed during testing, when requesting recent history data up until "now", the data returned could include datapoints for a timestamp with all values within set to zero.

On subsequent calls, the values for that specific timestamp would then be filled in with values (they would change or get updated). It is as if there can be lag for the data to be available from the cloud. This is why I added an option and default to e.g. always retrieve 60 minutes of history on each poll period - subsequent calls would then replace any previously written data within that history period with the updated data.

In your case though, I'll speculate that perhaps instead of updating values for the timestamp, on subsequent calls for history data the timestamp is then not returned! However, it would have been written to InfluxDB previously already, with all values set to zero.

It's a bit of a guess as I have not observed the behaviour myself. I think it seems like a plausible explanation though, so I will have a look at the script and see if I can make some changes to account for it.

youzer-name, 2023-06-02:
I've seen the behavior you mentioned with the last 2 or 3 data points being zero and it then updating. I think you've correctly identified the cause.

If you see a way to fix it, great. If not, selecting only non zero values in the Grafana query works for my use case.

mcbirse, 2023-06-12:
@youzer-name - let us know how the latest version works for you when you have had a chance to test for a while. 

Instead of updating data points in InfluxDB, the script will now remove previously written data just before writing new data.

In theory therefore, timestamps that were initially included in the response (i.e. with zero values) but not being returned in subsequent responses (i.e. no data) would be removed if they are within the history period (60 mins be default).

youzer-name, 2023-09-16:
@mcbirse - It took me a while to get around to installing the updated script, but I've been running it for a while and it seems to be working as expected. I'm also able to do a --remove and re-download of data for past dates to clear older zero values. Thanks again for your work on this tool... I'm going to close this issue.

closed 2023-09-16 by youzer-name

mcbirse, 2023-09-17:
Thanks @youzer-name for the feedback, great to hear it is working well!