jasonacox/Powerwall-Dashboard #292: Missing Strings and/or Configuration issue

ShawnKimble, 2023-06-09:
I have 2 PowerWall+ 1 Tesla Inverter for a total of 3 Inverters (3 x 7.6kw = 22.8kW max) fed with 9 strings, three strings to each inverter all fed out through a Tesla Gateway.  I started using this Dashboard to further investigate the proper setup of the system as I experianced 10-15 different electricians (yeah it was crazy) durring my install, and had to resolve numerious issues.  So I'm still not 100% sure everything is configured correctly, and unclear if the default configuration of this dashboard should pick up all 9 strings or not.

**System Details**
I have the dashboard up and running on Ubuntu 18.04.6, the Dashboard seems to be working fine, but under 
localhost:8675/strings, I only see 8 Strings, of which two read zeros.

Why do I not see all 9 strings, and what configuration edits might I need to capture everything?
I also find it odd that I see 8 strings, with 6 active, unless this dashboard only can see the 6 strings of data from the PowerWall+ and nothing from the standalone Tesla Inverter.

{
    "A": {
        "Connected": true,
        "Current": 1.95,
        "Power": 836.0,
        "State": "PV_Active",
        "Voltage": 431.20000000000005
    },
    "A1": {
        "Connected": true,
        "Current": 1.22,
        "Power": 403.0,
        "State": "PV_Active",
        "Voltage": 333.70000000000005
    },
    "B": {
        "Connected": true,
        "Current": 7.88,
        "Power": 2540.0,
        "State": "PV_Active",
        "Voltage": 328.1
    },
    "B1": {
        "Connected": true,
        "Current": 3.73,
        "Power": 1369.0,
        "State": "PV_Active",
        "Voltage": 382.6
    },
    "C": {
        "Connected": true,
        "Current": 8.41,
        "Power": 2257.0,
        "State": "PV_Active",
        "Voltage": 271.2
    },
    "C1": {
        "Connected": true,
        "Current": 8.99,
        "Power": 2904.0,
        "State": "PV_Active",
        "Voltage": 338.90000000000003
    },
    "D": {
        "Connected": false,
        "Current": 0.0,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": 0.10000000000000142
    },
    "D1": {
        "Connected": false,
        "Current": 0.0,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -2.3
    }
}

Thanks for any help!

ShawnKimble, 2023-06-09:
I've learned that the Gateway operates on 192.168.91.1, and the standalone inverter is on 192.168.92.1.  Is there a way to connec to both of these and get all the data from the three other strings?

I think in the above, D strings are the fourth string that is missing from each inverter, and that then makes sense about this setup.

jasonacox, 2023-06-10:
Hi @ShawnKimble, that's an elaborate setup!  

The issue is that Powerwall-Dashboard and the pypowerwall API is designed to only talk to one gateway.  I made the (bad) assumption that Tesla would aggregate all Powerwalls into the same gateway interface.  This has come up before and is tracked in https://github.com/jasonacox/Powerwall-Dashboard/issues/148. 

There is a way for you to customize your setup to see everything and have it store everything in the same dashboard. It will require customizing the setup... thinking through it:
* You will need two pypowerwall containers running (one for each gateway) 
* The telegraf config will need to tag the data based on which gateway it is using
* The InfluxdDB continuous queries will need to be updated to handle both gateway data feeds
* The dashboard will need to be adjusted to display both gateways (some can be combined, but others like alert data will need separate panels because the alerts will be specific to the gateway.

Ok, before we we go too far, It would be interesting to confirm that the string data is separate like we suspect. You could run this python script to pull string data for each:

```python
import pypowerwall

# Connect to Powerwalls
pw1 = pypowerwall.Powerwall("192.168.91.1","password","email")
pw2 = pypowerwall.Powerwall("192.168.92.1","password","email")

print("String Data from GW 192.168.91.1: %s\n\n" % pw1.strings(True))
print("String Data from GW 192.168.92.1: %s\n\n" % pw2.strings(True))

# Or show raw vitals
# print(pw1.vitals(True))
# print(pw2.vitals(True))
```

ShawnKimble, 2023-06-10:
Thanks Jason, here's what I get....

When on my local network, the script fails to connect.
If I connect to the Gateway as my wifi, I get this
````
String Data from GW 192.168.91.1: {    "A": {
        "Connected": true,
        "Current": 0.9400000000000001,
        "Power": 362.0,
        "State": "PV_Active",
        "Voltage": 369.40000000000003
    },
    "A1": {
        "Connected": true,
        "Current": 4.14,
        "Power": 1220.0,
        "State": "PV_Active",
        "Voltage": 269.90000000000003
    },
    "B": {
        "Connected": true,
        "Current": 5.78,
        "Power": 2017.0,
        "State": "PV_Active",
        "Voltage": 321.40000000000003
    },
    "B1": {
        "Connected": true,
        "Current": 3.2600000000000002,
        "Power": 1462.0,
        "State": "PV_Active",
        "Voltage": 346.70000000000005
    },
    "C": {
        "Connected": true,
        "Current": 5.62,
        "Power": 1489.0,
        "State": "PV_Active",
        "Voltage": 242.4
    },
    "C1": {
        "Connected": true,
        "Current": 7.8100000000000005,
        "Power": 2339.0,
        "State": "PV_Active",
        "Voltage": 250.3
    },
    "D": {
        "Connected": false,
        "Current": 0.0,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -0.5999999999999996
    },
    "D1": {
        "Connected": false,
        "Current": 0.08,
        "Power": 0.0,
        "State": "PV_Active",
        "Voltage": -1.0999999999999996
    }
}

String Data from GW 192.168.92.1: None

```

If I connect to the Tesla standalone inverter as my wifi connection, I get this

```
StringData from GW 192.168.91.1: None
String Data from GW 192.168.92.1: None
```

jasonacox, 2023-06-10:
A few things:

1. I would recommend accessing your Powerwall gateways via their local IP instead of their built in WiFi access point (I found those access points to be very unstable but maybe they are better now).  You can find your Powerwalls' local IPs using the scan feature of pypowerwall:

  ```bash 
  python3 -m pypowerwall scan
  ```

2. I suspect your second gateway ("standalone inverter") is not a Powerwall+.  that means you will not get string data from that unit. The Powerwall+ is the only one that reports individual string data. Sorry about that. I don't know why Tesla did that.

Based on that new info, I do wonder what you are currently getting with the Dashboard.  Do the solar production values (kW) seem to match what you see in the Telsa App?  If so, that would indicate that the Powerwall+ is aggregating the solar data from the standalone inverter too.  If so, that is good news and your current setup will work.  Unfortunately, as mentioned in 2 above, you won't see individual string details from your standalone inverter.

ShawnKimble, 2023-06-10:
I'm not sure why this is the result I got, as I'm on my local network, and I know that the wifi's in the Gateway and Inverter are also set to connect to this wifi, yet scan picks up nothing.  

On your dashboard I match the app, so the standalone inverter must be rolling it's string data up to the gateway somehow.
The six strings that show though, don't total the grand total, so I guess you might be able to subtract those two and make a "Hidden String" that represents the strings that are not part of the Powerwall+ 

```pyPowerwall Network Scanner [0.6.2]
Scan local network for Tesla Powerwall Gateways

    Your network appears to be: 192.168.1.0/24

    Enter Network or press enter to use 192.168.1.0/24:

    Running Scan...
      Host: 192.168.1.1 ... OPEN - Not a Powerwall
      Host: 192.168.1.7 ... OPEN - Not a Powerwall
      Host: 192.168.1.33 ... OPEN - Not a Powerwall
      Done

Discovered 0 Powerwall Gateway

ShawnKimble, 2023-06-10:
To clarify the "Current State" box on the Dashboard matches the overall "Power Flow", which matches the app.   The other graphs, including the primary "Energy Useage" are all off by the missing string data.

jasonacox, 2023-06-10:
The 192.168.9x.x addresses look like the ones that you get when you attach to the Powerwall Gateway Wifi access point directly.  Based on the scan, it seems like your local network should be assigning addresses within the 192.168.1.x range instead.  If you are able to get pypowerwall to work, try to browse to `http://grafana:8675/api/networks` (change 'grafana' to the address of your dashboard) and see what IP addresses the gateway has assigned to the various networks (it will be a JSON payload).

> To clarify the "Current State" box on the Dashboard matches the overall "Power Flow", which matches the app. The other graphs, including the primary "Energy Useage" are all off by the missing string data.

That's confusing.  The "Current State" box and "Power Flow" use the same data as the "Energy Usage" graph (it is just a time series plot of that current state data).  Can you post a screenshot?

ShawnKimble, 2023-06-12:
I've tried to get these all at the same timestamp or close, this shows the tesla app, the overall dashboard, the dashboard detail at 13:19 timestamp, the string detail and the stand alone tesla inverter that is not part of the dataset of the dashboard.   

The strings add up to 8,216kW from the PowerWall Plus (2x) and then 1,730kW from the Tesla Inverter = ~9,946kW.
So all of the string data does seems to roll up into the primary graph.  (I think my prior comment was a timing issue and clouds making it hard to get static numbers)

[image]
[image]
[image]
[image]
[image]

ShawnKimble, 2023-06-12:
Cycling all 3 inverters on and off, I can confirm that the third standalone inverter is piggybacked onto the PW2+ data feed.

jasonacox, 2023-06-13:
Nice! 👍  That's great news that you are getting the aggregate data.  It is a shame about the not getting all the string data, but at least you know.

By the way, I wonder what is happening to your power flow animation?  It doesn't look like pypowerwall is proxying and translating it correctly:

[image]

I wonder if has to do with the odd networking where your Powerwalls are not showing up on your local network? In my case, I hardwired my gateway (ethernet cable) since Wifi was too finicky.  😄

jasonacox, 2023-06-14:
Awesome @ShawnKimble !  Having the dashboard has helped me debug several issues as well. It's also gratifying to see it all working. 😁 

On the stand alone inverter, I wonder if that has a local LAN IP address as well?  Since the string data is showing up here, there may be a way to scrape that from the site (essentially that is how we determined the API endpoints for the Powerwall too).

If you click on the WiFi setting in the graphic, does it give you the LAN IP address?  You might try pointing pypowerwall at that IP address to see if you get any data, or less trivial, examine the web page source code/calls to see what APIs it is using to get the string data. My only fear is that it is using a protobuf payload which would require a .proto translation file (like we have for the gateway).

[image]

ShawnKimble, 2023-06-14:
The Inverter has a local IP, but it states you can't configure from that access point (see below).  It also requires a physical power cycle to login, otherwise the interface locks connections out.  

It seems to pool the data from a "v1" POST request to this endpoint, but I can't figure out how to get the json payload, as when I explore this url I get a response error.   "wsgi handler failed"
http://192.168.92.1/tedapi/v1

This one works, and give me back the Part ID and Serial Number, which seem to be needed to pull the above request.
http://192.168.92.1/tedapi/din

[image]

[image]

[image]

ShawnKimble, 2023-06-14:
Found a few more endpoints
http://192.168.1.5/api/1/vitals
This also provides "wsgi handler failed"
 
http://192.168.1.5/service
[image]

jasonacox, 2023-06-15:
Wow, nice discovery!  Keep trying all the URLs you find in the Chrome developer tool "Network" tab.

> wsgi handler failed

That is the "web server gateway interface" - likely indicating that you need to authenticate to be authorized to access that endpoint.  You might try combinations of https://192.168.1.5/api/login/Basic which is the endpoint on the gateway used to log in (what pypowerwall uses).