jasonacox/Powerwall-Dashboard #308: Time zone local to device not server?

dkerr64, 2023-06-20:
Traveling overseas and checking on my system at home and noticed that all times are converted to the time zone of my local device. 

So as my iPhone sets itself to local my sunrise and sunset time report as 11:20am and 2:30am which is 6 hours wrong.  Similarly my graphs representing “today” start at midnight local time not midnight server time. 

Can this be configured somewhere? 

Thanks

youzer-name, 2023-06-20:
Click on the time range selector at the top/right of the Grafana page (where it says "Last 24 hours" or "Today" etc.)

[image]

At the bottom, click "Change time settings"

It will show a box where you can set the time zone.

[image]

dkerr64, 2023-06-20:
Ah, thank you.  I suspected there was a setting in Grafana somewhere that I needed to change.  You saved me a lot of time by pointing me directly to it.

Thanks
David

closed 2023-06-20 by dkerr64