jasonacox/Powerwall-Dashboard #309: Displayed Powerwall/Gateway firmware in dashboard is stale

ibmaster, 2023-06-20:
What is the refresh rate for the animation pane showing the current Tesla firmware?
My system updated to 23.4.2-2 yesterday (as confirmed via Tesla's iOS app) but the dashboard is saying I'm still on 23.4.2-1:

[image]

ibmaster, 2023-06-20:
Well, if I completely reload the dashboard's web page it will refresh the latest firmware version string correctly... good enough for me, closing.

closed 2023-06-20 by ibmaster

jasonacox, 2023-06-20:
It should refresh every 60s or when you reload your browser as you discovered. The code for the version refresh is here if you are using a recent dashboard.json version (you can also see it by editing the animation panel):

https://github.com/jasonacox/Powerwall-Dashboard/blob/18ff67bcb433eeaa00731c2d417bbe267202e4a8/dashboards/powerflow-animation.html#L44-L53