jasonacox/Powerwall-Dashboard #310: Telegraf Error on Synology NAS

jasonacox, 2023-06-21:
Reported by user:

> I’ve been using your powerwall docker successfully since March and it’s awesome!  Unfortunately, I made the stupid mistake of updating the telegraf container to latest version and now it crashes constantly with the error “Error running agent: no outputs found, did you provide a valid config file?”  I’m hoping there is an easy fix for this rather than having to completely reinstall everything and lose 4 months of data.

jasonacox, 2023-06-21:
> lose 4 months of data. 

Your data is stored in InfluxDB so as long as you are not changing that, you should be in good shape.

> no outputs found 

This indicates that the new telegraf is not loading the telegraf configuration file ([telegraf.conf](https://github.com/jasonacox/Powerwall-Dashboard/blob/18ff67bcb433eeaa00731c2d417bbe267202e4a8/telegraf.conf#L91-L99)) which should include this output:

https://github.com/jasonacox/Powerwall-Dashboard/blob/18ff67bcb433eeaa00731c2d417bbe267202e4a8/telegraf.conf#L91-L99

I don't know why an upgrade would stop using the config file. You can try:
* Check logfile to see if an error on startup indicates why it isn't loading the config (`docker logs telegraf)
* Downgrade telegraf - We currently have the project pinned to telegraf:1.26.1 though I don't know why new version wouldn't work.
* Verify your telegraf.conf file still exists in the same directory where you start telegraf (e.g. `compose-dash.sh up -d`)

homeontherocks, 2023-06-21:
Thanks Jason.  I've checked the telegraf.conf file and it seems fine.  However, the container telegraf appears to be trying to load the telegraf.conf file from the /etc/telegraf/ path and there isn't one.  Sorry for the basic simple question.  Thanks!

[image]

jasonacox, 2023-06-22:
Hi @homeontherocks !  I don't have a Synology NAS so I'm not entirely sure how your configuration is set up.  If you used `setup.sh` to get your setup built, it would have used the docker compose file `powerwall.yml`.  This section of the compose defines how telegraf is built:

https://github.com/jasonacox/Powerwall-Dashboard/blob/18ff67bcb433eeaa00731c2d417bbe267202e4a8/powerwall.yml#L35-L59

Notice that the "volumes" section maps the local `./telegraf.conf` file to `/etc/telegraf/telegraf.conf`.  A few quetions

* How did you "upgrade" telegraf?
* What Dashboard version are you running (look for the VERSION file)?

My suggestion would be to completely remove telegraf and let docker compose rebuild it:

```bash
# remove all traces of telegraf
docker stop telegraf
docker rm telegraf
docker images | grep telegraf | awk '{print $3}' | xargs docker rmi -f

# reinstall
./compose-dash.sh up -d
# if the above command doesn't exist, you have an older version and could try
# docker-compose -f powerwall.yml up -d
```

homeontherocks, 2023-06-22:
Hi Jason, Thanks for the reply!  I just went back in and ran the ./setup.sh and kept all the same configurations I had and it seems to have overwritten the telegraf and seems to be working normally now (minus the hours of data I’ve missed).  I’ll let you know tomorrow if that is an easy fix for other idiots that go messing with the working config.  BTW, Synology docker informed me that telegraf could be updates and I just clicked the update button. Have a great night!B From: Jason Cox ***@***.***>Date: Wednesday, June 21, 2023 at 10:07 PMTo: jasonacox/Powerwall-Dashboard ***@***.***>Cc: homeontherocks ***@***.***>, Mention ***@***.***>Subject: Re: [jasonacox/Powerwall-Dashboard] Telegraf Error on Synology NAS (Issue #310)Hi @homeontherocks ! I don't have a Synology NAS so I'm not entirely sure how your configuration is set up. If you used setup.sh to get your setup built, it would have used the docker compose file powerwall.yml. This section of the compose defines how telegraf is built:https://github.com/jasonacox/Powerwall-Dashboard/blob/18ff67bcb433eeaa00731c2d417bbe267202e4a8/powerwall.yml#L35-L59Notice that the "volumes" section maps the local ./telegraf.conf file to /etc/telegraf/telegraf.conf. A few quetionsHow did you "upgrade" telegraf?What Dashboard version are you running (look for the VERSION file)?My suggestion would be to completely remove telegraf and let docker compose rebuild it:# remove all traces of telegrafdocker stop telegrafdocker rm telegrafdocker images | grep telegraf | awk '{print $3}' | xargs docker rmi -f # reinstall./compose-dashsh up -d# if the above command doesn't exist, you have an older version and could try# docker-compose -f powerwall.yml up -d

jasonacox, 2023-06-22:
Fantastic news!  Thanks for the update and warning for others. 👍

homeontherocks, 2023-06-22:
FYI for those that make the same mistake I did (DO NOT update any of the containers of a working configuration!), running ./setup.sh again and leaving the configuration data the same should overwrite the upgraded containers and get back to a working configuration.

homeontherocks, 2023-06-22:
Can confirm normal data collection resumed after simply rerunning setup.sh

jasonacox, 2023-06-23:
Thanks @homeontherocks !  I'll close this issue but feel free to respond or open it back up if you have any other questions.

closed 2023-06-23 by jasonacox