jasonacox/Powerwall-Dashboard #317: Influx DB Error: Unable to find Timezone Chicago

oralallen82, 2023-06-30:
Recently upgraded to latest version, during final stages of the install I noticed in the dialog:

Add downsample continuous queries to InfluxDB...
2023/06/30 03:10:24 error: error parsing query: unable to find time zone Chicago
2023/06/30 03:10:24 error: error parsing query: unable to find time zone Chicago

After installing importing the new dashboard I noticed that a few items weren't loading, hovering over the box I got the error:

InfluxDB Error: error parsing query: unable to find time zone Chicago

[image]

closed 2023-06-30 by oralallen82

jasonacox, 2023-07-01:
Hi @oralallen82  - I see you closed this. Were you able to get it fixed?

oralallen82, 2023-07-01:
I was yes.