jasonacox/Powerwall-Dashboard #320: Missing String Voltage, Current and Power pypowerwall running

maticks81, 2023-07-02:
Any suggestions of how to get String Voltage, Current and Power in the dashboard.
Everything else is working fine, pypowerwall is running and collecting other stats.

[image]

BuongiornoTexas, 2023-07-02:
Those are only available with a Tesla inverter and panels (well, barring people other than @jasonacox  doing the hard yards to add non-Tesla apis to pypowerwall ...). I know I'd like to add data from our Fronius, but it's also a long way down my to do list.

jasonacox, 2023-07-02:
Hi @maticks81 - @BuongiornoTexas is correct. The string data appears to only be available on Powerwall+ systems.  The string data is pulled from the device vitals API (e.g http://pypowerwall:8675/vitals) and shows up in the PVAC and PVS elements.

aesculus, 2023-08-24:
@jasonacox Notice the OP still has access to the Powerwall Temps panel even though they do not have a Powerwall+.

Can this be moved out of the Powerwall+ panel and into it's own panel like frequency, voltage and capacity panels? I do that but it makes updating the dashboard.json file a PITA. I also delete what remains in the Powerwall+ panel but I could just push the down arrow to hide it.

jasonacox, 2023-08-25:
Good suggestion @aesculus - something like this?

[image]

aesculus, 2023-08-25:
Perfect. And I can just close that row and leave the label and save me 5 minutes every time you update the dashboard. :-)

thenitz, 2024-05-07:
@jasonacox Hi Jason, I recently added more ct's on my neurio to be able to monitor all 3 invertors separately vs the combined view. The ct info does populate on the vitals page CT0, CT1, CT2 how can I use this to populate string data on my non plus setup?
[image]

jasonacox, 2024-05-08:
Hi @thenitz !  First, I have to mention that you are running Firmware 23.12.2 based on the screenshot. I have some bad news. When your system upgrades to anything beyond 23.44.3, you will lose access to the `vitals` data, unfortunately. We are working on a way to get the data back, but it won't be in the same format as the vitals. However, it does look like the tedapi payload that we are exploring does have the "neurio" data, but not sure what it would look like for you.  You can check out the `status.py` script at https://github.com/jasonacox/pypowerwall/tree/main/tools/tedapi:

```json
   "neurio": {
        "isDetectingWiredMeters": false,
        "pairings": [],
        "readings": [
            {
                "dataRead": [
                    {
                        "currentA": 0.9900000095367432,
                        "reactivePowerVAR": -30.65999984741211,
                        "realPowerW": -9.609999656677246,
                        "voltageV": 121.20999908447266
                    },
                    {
                        "currentA": 0.009999999776482582,
                        "reactivePowerVAR": -0.05999999865889549,
                        "realPowerW": 0.05000000074505806,
                        "voltageV": 121.75
                    },
                    {
                        "currentA": 0.009999999776482582,
                        "reactivePowerVAR": -0.029999999329447746,
                        "realPowerW": -0.05999999865889549,
                        "voltageV": 121.80000305175781
                    },
                    {
                        "currentA": 0.009999999776482582,
                        "reactivePowerVAR": -0.10000000149011612,
                        "realPowerW": -0.019999999552965164,
                        "voltageV": 121.20999908447266
                    }
                ],
                "serial": "VAH4600AB2155",
                "timestamp": "2024-02-20T21:06:35-08:00"
            }
        ]
    },
```

Having said all that, if you still want to use that data while you have it, you should see the NEURIO data in the `raw` bucket if you create a new dashboard panel and edit it.

[image]

My system upgraded so I can't confirm, but poke around in the dropdowns to see if you can find it. You could use that to create a Neurio panel.

thenitz, 2024-05-08:
Interesting are you in Cali?
I just manually forced an update the other day (replacement pw) so this is
the latest in our area (Tucson AZ)

jasonacox, 2024-05-09:
Yes, I'm in SoCal. I'm just jealous. I think a lot of us would have been happy to stay on that version. 😉

thenitz, 2024-05-09:
more info on versions:
I had a quick look at the rest of our firmware versions. my sister's house
is running 24.4.0 (she does not have a span panel)
my house is running 23.12.2 and my parents are running 23.44.0 the two of
us are using span panels and all three of these are non pw+. My boss's pw+
is running 24.4.0 also no span panel. Wondering if Span is maybe preventing
that update or if they're working with Tesla in some way.

jasonacox, 2024-05-10:
Interesting! I hope that is the case and they make a future version that brings back the vitals. I know, wishful thinking. :) 

🤞 that you get to stay on 23.12.2.

thenitz, 2024-05-13:
@jasonacox 
when adding to page cta and ctb are populating data but ctc reads 0 here is what i see from the vitals page right now. but screenshot shows raw data

 "NEURIO--": {
        "NEURIO_CT0_InstRealPower": 4958.68017578125,
        "NEURIO_CT0_Location": "solar",
        "NEURIO_CT1_InstRealPower": 1784.8399658203125,
        "NEURIO_CT1_Location": "solar",
        "NEURIO_CT2_InstRealPower": 3311.199951171875,
        "NEURIO_CT2_Location": "solar",
        "componentParentDin": "STSTSM--1232100-00-H--",
        "firmwareVersion": "1.7.2-Tesla",
        "lastCommunicationTime": 1715615696,
        "manufacturer": "NEURIO",
        "meterAttributes": {
            "meterLocation": [

   "METER_X_CTA_I": 8.7535,
        "METER_X_CTA_InstReactivePower": -333.0,
        "METER_X_CTA_InstRealPower": -35.0,
        "METER_X_CTB_I": 8.492,
        "METER_X_CTB_InstReactivePower": -622.0,
        "METER_X_CTB_InstRealPower": 54.0,
        "METER_X_CTC_I": 0.0,
        "METER_X_CTC_InstReactivePower": 0.0,
        "METER_X_CTC_InstRealPower": 0.0,
        
        
        
        
[image]

jasonacox, 2024-05-14:
Good job getting the graphs working!

>"METER_X_CTA_I": 8.7535,
"METER_X_CTA_InstReactivePower": -333.0,
"METER_X_CTA_InstRealPower": -35.0,
"METER_X_CTB_I": 8.492,
"METER_X_CTB_InstReactivePower": -622.0,
"METER_X_CTB_InstRealPower": 54.0,
"METER_X_CTC_I": 0.0,
"METER_X_CTC_InstReactivePower": 0.0,
"METER_X_CTC_InstRealPower": 0.0,

That shows that the CTC meter is not working or disable.

thenitz, 2024-05-15:
double checked the wiring and reran the tesla setup the ct is reading but
it doesn't look like its getting passed through?
also wasn't sure if the data was multiplied by 2 already or if it needed
the multiplier added to it (since only 1 leg (1ct) is attached per inverter)

jasonacox, 2024-05-15:
Are you checking the CT's on the Solar Inverters, or your house CT's (main utility power)?  The `METER` readings are for the utility power and unless you have 3 phase, you will only see 2 working.

The Neurio's are the ones usually set up to monitor the solar energy. You show 3:

```
"NEURIO_CT0_InstRealPower": 4958.68017578125,
"NEURIO_CT0_Location": "solar",
"NEURIO_CT1_InstRealPower": 1784.8399658203125,
"NEURIO_CT1_Location": "solar",
"NEURIO_CT2_InstRealPower": 3311.199951171875,
"NEURIO_CT2_Location": "solar",
```

You should see these matching what you get at http://localhost:8675/pod  (change localhost to your Dashboard address) as something like this: 

```
"PW1_p_out": 460,
"PW1_q_out": -20,
"PW1_v_out": 242.9,
"PW1_f_out": 59.991,
"PW1_i_out": -11,

"PW2_p_out": 440,
"PW2_q_out": -10,
"PW2_v_out": 243.10000000000002,
"PW2_f_out": 59.992000000000004,
"PW2_i_out": -10.700000000000001,

...

```

thenitz, 2024-05-15:
> Are you checking the CT's on the Solar Inverters, or your house CT's (main utility power)? The `METER` readings are for the utility power and unless you have 3 phase, you will only see 2 working. [...]

oh I see my mistake I was searching RAW for CT and found the CTA CTB CTC and assumed it was for my solar. I just went through all options and couldn't find any Neurio or CTs in raw or vitals. 

[image]

jasonacox, 2024-05-15:
Look in the next dropdown for "neurio" - there may be several:

[image]

thenitz, 2024-05-15:
https://drive.google.com/file/d/10hSp-fpW_mEXkq-mFbmRIQyBtdPb9IhD/view?usp=sharing

been through all 3 dropdowns - also tried typing in neurio in 2nd field
after raw or vitals
and for the 3rd field tried these
NEURIO_CT0_InstRealPower; NEURIO_CT1_InstRealPower; NEURIO_CT2_InstRealPower

"NEURIO_CT0_InstRealPower": 5418.97998046875,
        "NEURIO_CT0_Location": "solar", (tesla 7.x)
        "NEURIO_CT1_InstRealPower": 2530.780029296875,
        "NEURIO_CT1_Location": "solar", (tesla 3.x)
        "NEURIO_CT2_InstRealPower": 6427.89990234375,
        "NEURIO_CT2_Location": "solar", (solar edge 11.x)

jasonacox, 2024-05-16:
I'm sorry @thenitz - without having access to vitals myself, I'm sort of poking around in dusty memory. I finally remember that we are NOT capturing any of the neurio data. If you look at `telegraf.conf` it will show you the URLs that are used to pull in data. You won't see any URL that will contain the vitals Neurio data. 

```
[[inputs.http]]
	urls = [
		"http://pypowerwall:8675/aggregates",
		"http://pypowerwall:8675/soe",
		"http://pypowerwall:8675/strings",
		"http://pypowerwall:8675/temps/pw",
		"http://pypowerwall:8675/freq",
		"http://pypowerwall:8675/pod"
	]
	method = "GET"
	insecure_skip_verify = true
	timeout = "4s"
	data_format = "json"
```

You could add a URL for vitals in there and I would even suggest adding it to `telegraf.local`.

thenitz, 2024-05-16:
thanks was going nuts thinking I'm blind LOL