jasonacox/Powerwall-Dashboard #322: User Token Not Found Error

aesculus, 2023-07-03:
**Problem**
Every few hours I get Grafana log error and an alert on the Dashboard

logger=context traceID=00000000000000000000000000000000 t=2023-07-03T17:01:14.560304748Z level=error msg="Failed to look up user based on cookie" error="user token not found"

On the Dashboard an alert popup says Unknown Error but it clears after a few seconds.

Any thoughts on what might be causing this? I looked at the other containers for the dashboard and did not notice any other errors at the same times.

aesculus, 2023-07-03:
Update: I just saw this error on the % Solar Powered Meter line and it's currently displaying N/A

Unexpected token 'L', "Logged in" is not valid JSON

Not sure if this is related to #302 but the solar is producing 4 kW at the time.

Cleared when I set the timer to 4s.

mcbirse, 2023-07-06:
> Unexpected token 'L', "Logged in" is not valid JSON

I see this exact same error as well occasionally.

It happens intermittently/randomly and not on any specific Grafana panel/element - it could be any of them.

Usually just hitting refresh will clear it, or it will clear on the next auto-refresh (default 5m period).

I would love to know what is causing this... if anyone has any ideas or thoughts on how to investigate the cause, that would be great. Due to the intermittent nature of the problem I haven't been able to work it out, or bother investing much time in it to be honest.

I do wonder if it is some sort of issue with Grafana sending too many simultaneous query requests to the InfluxDB host for the server to be able to handle sometimes?

aesculus, 2023-07-06:
Thanks for confirming its not just my system. I was worried it was a portent for something worse. It does clear quickly and does not seem to have any lasting impact.

jasonacox, 2023-07-07:
Interesting. I haven't seen this. A few questions to see if we can build some connection to cause...

* Are you using the latest dashboard.json and have you seen this before the recent update? I want to rule out any recent changes.
* What is the Dashboard stack (grafana, influx) running on (e.g. I have it on a RPi4 and a Ubuntu 20 host)?
* What browser are you using (e.g. I'm using Chrome)?

It may be interesting to see if there are any errors in the grafana or influxdb logs or what the dev console says in the browser.

aesculus, 2023-07-07:
I am still using the previous dashboard.json

My stack is the infamous QNAP NAS with everything on the same instance

Using Chrome Version 114.0.5735.199 (Official Build) (64-bit)

aesculus, 2023-07-07:
I wiped my system and went to a Git install vs doing it by hand.

I am getting these messages in pypowerwall every so often. Maybe this gives more depth? The same message in Grafana as above seems to correlate to it but hard to be sure as there are no timestamps in pypowerwall log.

Exception occurred during processing of request from ('172.29.0.4', 36808)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 155, in _get_session
    self.auth = {'AuthCookie': r.cookies['AuthCookie'], 'UserRecord': r.cookies['UserRecord']}
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 334, in __getitem__
    return self._find_no_duplicates(name)
  File "/usr/local/lib/python3.10/site-packages/requests/cookies.py", line 413, in _find_no_duplicates
    raise KeyError(f"name={name!r}, domain={domain!r}, path={path!r}")
KeyError: "name='AuthCookie', domain=None, path=None"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/local/lib/python3.10/http/server.py", line 433, in handle
    self.handle_one_request()
  File "/usr/local/lib/python3.10/http/server.py", line 421, in handle_one_request
    method()
  File "/app/server.py", line 232, in do_GET
    vitals = pw.vitals()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 308, in vitals
    stream = self.poll('/api/devices/vitals')
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 233, in poll
    self._get_session()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 164, in _get_session
    raise LoginError("Invalid Powerwall Login")
pypowerwall.LoginError: Invalid Powerwall Login

jasonacox, 2023-07-08:
> pypowerwall.LoginError: Invalid Powerwall Login

This is the cause of the stack trace. The pypowerwall library will raise this exception to indicate that login to the Powerwall failed, but it will continue (allowing the proxy to retry and eventually reconnect).  I sometimes get network related errors but they look like this:

```
07/07/2023 03:21:04 PM [proxy] [ERROR] Socket broken sending response [doGET]
07/07/2023 03:21:04 PM [proxy] [ERROR] Socket broken sending response [doGET]
07/07/2023 03:21:04 PM [proxy] [ERROR] Socket broken sending response [doGET]
```

However, I haven't seen it impact login to the Powerwall.  Is your Powerwall connected via WiFi or hardware (ethernet)? I recall early on that I would see long periods of network errors and data gaps (up to 10m at a time) when I had the Powerwall connected via Wifi.  I hardwired it into my network switch and haven't seen that since.

The "login" errors could be due to too many connections to the Powerwall.  One option to try would be to decrease the pypowerwall `PW_POOL_MAXSIZE` value from 15 to 10. This is done by appending this to the `pypowerwall.env` file:

```bash
# append setting to pypowerwall env file
cat "PW_POOL_MAXSIZE=10" >> pypowerwall.env
cat pypowerwall.env

# restart
docker stop pypowerwall
docker rm pypowerwall
./compose-dash.sh up -d
```

You can tweak that number to see what works best.  Watch the pypowerwall logs to see what happens.  Other settings like `PW_CACHE_EXPIRE` (defaults to 5 -- five seconds) could be adjusted (increased) as well to reduce load on the Powerwall.

aesculus, 2023-07-08:
My PW was connect via WiFi and that was a complete disaster. But I cannot connect it directly to ethernet so I had to settle for a wireless extender to ethernet. Works very well know but this could have exposed the fact that timing could be an issue. I'll try changing the PW_POOL_MAXSIZE parm.  

Truthfully is NBD as it does self correct very quickly. Really just wanted you to be aware of it.

I'll check the Telegraf logs next time I see it but looking at the log now there was nothing post last startup 24 hours ago.

aesculus, 2023-07-22:
I noticed today that my NAS was basically locked up for a few minutes after one of these errors. I could not get into the Admin panel. After a few minutes it cleared and all was ok.

In the Telegraph logs I saw these:

2023-07-22T19:03:25Z W! [inputs.http] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:03:29Z W! [agent] ["outputs.influxdb"] did not complete within its flush interval
2023-07-22T19:03:48Z W! [inputs.kernel] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:03:48Z W! [inputs.http] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:03:48Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/freq]: Get "http://pypowerwall:8675/freq": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
2023-07-22T19:03:48Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/aggregates]: Get "http://pypowerwall:8675/aggregates": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
2023-07-22T19:03:48Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/temps/pw]: Get "http://pypowerwall:8675/temps/pw": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
2023-07-22T19:03:48Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/strings]: Get "http://pypowerwall:8675/strings": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
2023-07-22T19:03:48Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/pod]: Get "http://pypowerwall:8675/pod": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
2023-07-22T19:03:48Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/alerts/pw]: Get "http://pypowerwall:8675/alerts/pw": dial tcp 172.29.4.2:8675: i/o timeout (Client.Timeout exceeded while awaiting headers)
2023-07-22T19:03:48Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8675/soe]: Get "http://pypowerwall:8675/soe": dial tcp 172.29.4.2:8675: i/o timeout (Client.Timeout exceeded while awaiting headers)
2023-07-22T19:03:48Z W! [inputs.http] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:03:48Z W! [inputs.system] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:03:48Z W! [inputs.mem] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:03:48Z W! [inputs.disk] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:03:48Z E! [outputs.influxdb] When writing to [http://influxdb:8086]: failed doing req: Post "http://influxdb:8086/write?db=powerwall&rp=raw": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
2023-07-22T19:05:02Z W! [inputs.swap] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.mem] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.processes] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [agent] ["outputs.influxdb"] did not complete within its flush interval
2023-07-22T19:05:02Z W! [agent] ["outputs.influxdb"] did not complete within its flush interval
2023-07-22T19:05:02Z W! [inputs.diskio] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.diskio] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.http] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.http] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.disk] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.disk] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.cpu] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.system] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.system] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z W! [inputs.http] Collection took longer than expected; not complete after interval of 5s
2023-07-22T19:05:02Z E! [agent] Error writing to outputs.influxdb: could not write any address

jwvanderbeck, 2024-10-01:
Saw this Unexpected token myself today.  Running on RPi5 using Arc browser (which is chromium) however I highly doubt the browser has anything to do with this as the error appears to be data that is appearing in one of the REST requests

[image]

jasonacox, 2024-10-04:
What do you see in the pypowerwall logs?

jwvanderbeck, 2024-10-08:
Unless I'm missing something or looking in the wrong place, the only thing really in my logs is just a zillion lines of "Max retries exceeded with url: /tedapi/v1"

jasonacox, 2024-10-09:
Ah, that can happen if the password is not set correctly. Confirm that the password you used for the gateway which is typically the WiFi password listed on the QR code on the gateway.

jwvanderbeck, 2024-10-09:
The password is fine.  I get data constantly.  The max retries seems to be about throttling, IE it requests data too often.  But it doesn't seem to be an issue 99.9% of the time.

jasonacox, 2024-10-11:
Is the gateway connected via WiFi or a hardline? It could be that pypowerwall is having trouble connecting with the Powerwall Gateway. Are you able to ping the gateway?

Would you also mind posting the output of the `verify.sh` script?

jwvanderbeck, 2024-10-11:
It is wifi.  What I'm saying is I don't think this is a "you can't connect error" but a "You're connecting too often" error.

```
john@raspberrypi:~/Powerwall-Dashboard $ ./verify.sh
Verify Powerwall-Dashboard 4.5.3 on Linux - Timezone: America/New_York
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.11.1 Proxy t64
 - Powerwall State: CONNECTED - Firmware: 24.28.2
 - Site Name: Solar Gateway 1
 - Gateway TEDAPI: Available (192.168.91.1)
 - TEDAPI Vitals: Connected - Mode: full
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 76.42}
 - Version: 0.2.3

All tests succeeded.
```

jasonacox, 2024-10-12:
> What I'm saying is I don't think this is a "you can't connect error" but a "You're connecting too often" error 

Understood. The way the proxy works is that it creates a pool of connections to the Powerwall Gateway and leaves them open. This eliminates the TCP overhead for each subsequent call, reducing the load on the PW. The only reason it would want to reconnect is if the connection drops. That is why I'm wondering if it is due to a poor connection, possibly WiFi related. I had issues on my system when I was on WiFi but it would just go offline for 5-10m at a time before the WiFi would reconnect.  I have 4 different systems running the Dashboard simulatenously, all polling the gateway and don't see that error. However, my setup is is hardwired so it may not be a fair comparision.

## Proxy Tuning

There are tunings you can make to help optimize for a weak connection or if you think the system is polling too often.

* PW_CACHE_EXPIRE - Default is 5 seconds, this is how long pypowerwall will cache results before trying to pull from the Powerwall.
* PW_TIMEOUT - Default is 5 seconds, this is how long pypowerwall will wait to see if the Gateway is available before retrying. Increase to have it wait longer.
* PW_POOL_MAXSIZE - Default is 15, this is how many concurrent connections pypowerwall will make to the Gateway to process requests (API calls). This allows it to process multiple requests (e.g. telegraf calls) at the same time.  Some have had luck reducing this number on weaker connection or when they suspect the Gateway is having issues.

You can set these values in the `pypowerwall.env` file and run `./compose-dash.sh up -d` to activate the settings.

## Troubleshooting

You can also check the status of pypowerwall, including the number of timeouts it has detected by looking at:
http://localhost8675/help 

## Network Latency

Finally, if you ping 192.168.91.1 from your dashboard host, what is the average latency? I had a few cases where the ARP cache on the host was reflecting the traffic off the wrong bit of the network, driving the latency up.  If it is local it should be 10ms or less. My example is 1.32ms:

```
$ ping -q -c 10 192.168.91.1
PING 192.168.91.1 (192.168.91.1) 56(84) bytes of data.

--- 192.168.91.1 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9014ms
rtt min/avg/max/mdev = 1.392/1.450/1.499/0.037 ms
```

jwvanderbeck, 2024-10-12:
> Understood. The way the proxy works is that it creates a pool of connections to the Powerwall Gateway and leaves them open. This eliminates the TCP overhead for each subsequent call, reducing the load on the PW. The only reason it would want to reconnect is if the connection drops. 

Ok I got you there, just seems odd that I don't notice any problems with data not being available 99% of the time.  I could be looking at the wrong times :p  

Unfortunately Wifi was the only option at the time, as the installers weren't interested in pulling a line across the ceiling.  I either had to have one there waiting for them or nothing.

```
john@raspberrypi:~/Powerwall-Dashboard $ ping -q -c 10 192.168.91.1
PING 192.168.91.1 (192.168.91.1) 56(84) bytes of data.

--- 192.168.91.1 ping statistics ---
10 packets transmitted, 10 received, 0% packet loss, time 9011ms
rtt min/avg/max/mdev = 7.213/31.455/87.004/29.198 ms
```

jasonacox, 2024-10-12:
>  min/avg/max/mdev = 7.213/31.455/87.004/29.198 ms

Wow... That's a big range (7ms to 87m) for a local connection, and for just 10 pings! Run a continual ping for a bit to see if it gets even worse (`ping 192.168.91.1`). 

I suspect it is "good enough" most of the time but any ambient EMF disturbance might make it worse and push it past the timeout amount (5s) even for a brief period, causing a thundering herd of reconnects. 😁 

> Unfortunately Wifi was the only option at the time

I completely understand. As long as the system recovers, you are only losing a few samples and can fill in those gaps with the [tesla-history script](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/tesla-history) that fills in any gaps.