jasonacox/Powerwall-Dashboard #330: Error in New PW Dashboard Solar Energy Year

aesculus, 2023-07-24:
I noticed that on the graph legend that the grid mean was really low compared to solar and home. It looks like its taking the mean over the entire year vs amounts that are > 0 like home and solar are doing.

[image]

Also found this error when inspecting the query:

{
  "request": {
    "method": "GET",
    "url": "api/datasources/proxy/1/query",
    "params": {
      "db": "powerwall",
      "q": "SELECT sum(\"home\") FROM \"kwh\".\"http\" WHERE (\"home\" > 0) GROUP BY time(1d) fill(null) tz('America\\/Los_Angeles');SELECT sum(\"solar\") FROM \"kwh\".\"http\" WHERE (\"solar\" > 0) GROUP BY time(1d) fill(0) tz('America\\/Los_Angeles');SELECT sum(\"from_grid\") - sum(\"to_grid\") as \"grid\" FROM \"kwh\".\"http\" WHERE time >= now() - 24h and time <= now() GROUP BY time(1d) fill(0) tz('America\\/Los_Angeles')",
      "epoch": "ms"
    },
    "data": null,
    "precision": "ms",
    "hideFromInspector": false
  },
  "response": {
    "error": "error parsing query: found \\/, expected identifier, string, number, bool at line 1, char 98",
    "message": "error parsing query: found \\/, expected identifier, string, number, bool at line 1, char 98"
  }
}

jasonacox, 2023-07-25:
Hi @aesculus  - thanks for this.  Are you using this dashboard? 
https://github.com/jasonacox/Powerwall-Dashboard/blob/main/dashboards/dashboard.json

> I noticed that on the graph legend that the grid mean was really low compared to solar and home. It looks like its taking the mean over the entire year vs amounts that are > 0 like home and solar are doing.

Both solar and home should NOT have negative values so we remove any noise that may be caused by meters reporting negative flow.  However, for grid, you need to account for negative flow because you can push energy back to the grid. The "mean" will account for this so you could potentially have a very low or even negative mean.

> America\\/Los_Angeles

That's a typo.  The new dashboard pulls the timezone from the $tz grafana variable rather than have it hardcoded.  You might check to ensure it has the right variable value set.

[image]

aesculus, 2023-07-25:
Makes sense on the negative values.

Here is my Grid Usage chart. You can see I am more of consumer and the net is way more than 863 Wh.

[image]

The tz variable is set the same as above which is appropriate for my area.

jasonacox, 2023-07-25:
Ah, I think I see what you are saying. The mean is averaging in the 0kWh days.  We could add `WHERE ("grid" = 0)` to the query.

aesculus, 2023-07-25:
Let me know what the query should be and I can change it on the dashboard and see if it works.

This is something you would not notice if you had a full year of data since 0 would happen like never. :-)

aesculus, 2023-07-25:
BTW here is the query and number of results:

[image]

jasonacox, 2023-07-25:
Ok, if we assume that we should always have either home, from_grid, or to_grid power usage data, we can use the sum of those to indicate the system is activated.  You can update the query by clicking "Edit":

[image]

Scroll down to C and update the query to be (the updated bit is highlighted here):

[image]

Here is the updated query:

```sql
SELECT sum("from_grid") - sum("to_grid") as "grid" FROM "kwh"."http" WHERE ("home" + "from_grid" + "to_grid" > 0) and $timeFilter GROUP BY time(1d) fill(0) tz('$tz')
```

Click Apply

[image]

Click "Save"

[image]

aesculus, 2023-07-25:
Unfortunately still selecting 366 rows, most 0.

jasonacox, 2023-07-25:
Bummer.  Thanks for the update. This may just be the way Grafana computes legend labels. The query doesn't seem to impact it. I'm not sure what try try next.  If you discover anything that works, let us know!

aesculus, 2023-07-25:
Do you have a problem with this?

SELECT sum("from_grid") - sum("to_grid") as "grid" FROM "kwh"."http" WHERE $timeFilter GROUP BY time(1d) fill(0) tz('$tz')

I just changed the fill from 0 to null

[image]

jasonacox, 2023-07-25:
That's brilliant!  Nice job @aesculus !  I'll update the dashboard.

aesculus, 2023-07-25:
Looks great. Another satisfied customer.

closed 2023-07-25 by aesculus