jasonacox/Powerwall-Dashboard #337: Use .7z for backups

s-crypt, 2023-08-24:
**Problem**
Backups can use a use quite a bit of space, especially as time goes on.

**Enhancement**
When I download the backups and re-compress the files as a default 7z file, I get around 25% space savings.

**Additional context**
7-Zip now has native linux binaries ([link](https://7-zip.org/download.html)), including for arm, though there is the out of date but apt based p7zip

I think it would be a good idea to maybe make a separate backup script for people to download that uses 7z files instead of tar.gz  
  
Below is my attempt after installing the 7zz binary from 7zip and installing it in /usr/local/bin. I get an error `line 41: 14540 Killed 7zz a -t7z ${BACKUP_FOLDER}/Powerwall-Dashboard.$STAMP.7z influxdb -r`. I believe it has to do with 7zz not liking the sh reference of ${BACKUP_FOLDER}.

<details>
<summary>My testing script backup-7z.sh</summary>

```sh                                                                  
#!/bin/bash
#
# Backup Script for Powerwall Dashboard Data
#   https://github.com/jasonacox/Powerwall-Dashboard
#   by Jason Cox - 27 Feb 2022 

# Daily Backup for Powerwall-Dashboard Data
if [ "$EUID" -ne 0 ]
  then echo "Must run as root"
  exit
fi

# Set values for your environment 
DASHBOARD="/home/pi/Powerwall-Dashboard"    # Location of Dashboard to backup
BACKUP_FOLDER="${DASHBOARD}/backups"          # Destination folder for backups
KEEP="5"                                      # Days to keep backup

# Check to see if directory exists
if [ ! -d "${DASHBOARD}" ]; then
  echo "Dashboard directory ${DASHBOARD} does not exist."
  exit
fi
if [ ! -d "${BACKUP_FOLDER}" ]; then
  echo "Backup directory ${BACKUP_FOLDER} does not exist."
  exit
fi

# Timestamp for Backup Filename
STAMP=$(date '+%Y-%m-%d')

# Optional: Ask InfluxDB to create a snapshot backup 
echo "Creating InfluxDB Backup"
cd ${DASHBOARD}
mkdir -p influxdb/backups
chmod g+w influxdb/backups
docker exec influxdb influxd backup -database powerwall /var/lib/influxdb/backups

# Backup Powerwall-Dashboard Data
echo "Backing up Powerwall-Dashboard Data (influxdb)"
cd  ${DASHBOARD}
7zz a -t7z ${BACKUP_FOLDER}/Powerwall-Dashboard.$STAMP.7z influxdb -r

# Cleanup Old Backups
echo "Cleaning up old backups"
rm -rf ${DASHBOARD}/influxdb/backups/*        # Delete InfluxDB snapshots after backup
find ${BACKUP_FOLDER}/Powerwall-Dashboard.*7z -mtime +${KEEP} -type f -delete
echo "Done"

```
</summary>

jasonacox, 2023-08-25:
Thanks @s-crypt !  This is a good recommendation.  It looks like the challenge would be making it work across all platforms we support for the community (Linux Ubuntu, RPi, MacOS, WinOS).  It seems like Linux and RPI use `7z` but not MacOS or Windows in my testing.

s-crypt, 2023-08-25:
Since the backup script is bash, wouldn't it not work on windows anyway? 
I was thinking since configuring the backup is sort of a manual process anyway, asking a user to install 7z wouldn't be that big of an ask, we just need to get the script working (which is a little challenging because I have never worked with command line archiving programs before and I am not really understanding the errors I've gotten)

BuongiornoTexas, 2023-08-25:
Installing new software is likely to be a barrier for people who just want out of the box functionality. Can I suggest this should be an option rather than default? I know I would rather not install 7zip for a one off application (I use it as a swiss army knife on windows, but on Linux I've got the already very capable gzip and tar).

Also, have you tested how gzip performs at maximum compression level? (I know it's likely to be worse than 7zip, but I'd be surprised if it's as bad as 25% when both options are pushing compression rather than speed)

s-crypt, 2023-08-26:
I totally agree this should be optional, although I think it should be noted that the backup solution is not "out of the box" as it requires manual configuration and a manual cronjob so I don't see adding this as an option as too far of a stretch.

In my testing
influxdb.tar: 248.03 MB
influxdb.tar.gz (default compression): 85.79 MB
influxdb.tar.gz (max compression): 85.07 MB
influxdb.7z  (default): 67.11 MB (using p7zip, the 7z command. the official 7 zip binary using 7zz crashed by raspberry pi)

BuongiornoTexas, 2023-08-26:
> In my testing

Huh - much better than I expected from 7z. I take it that 7z is doing both archiving and packing? More from curiosity than anything else, does 7zip perform as well on the tar file?

|  I think it should be noted that the backup solution is not "out of the box" as it requires manual configuration

Fair enough!

s-crypt, 2023-08-26:
The testing was all on the tar file which was packed with `tar`. When packing with `7z` the process keeps getting killed (and it takes significantly longer)

s-crypt, 2023-08-26:
I believe I have a working backup script if the user has install p7zip (installed by default on raspberry pi I believe) using the `7z` command.

<details>
<summary>A working backup-7z.sh</summary>

```bash
#!/bin/bash
#
# Backup Script for Powerwall Dashboard Data
#   https://github.com/jasonacox/Powerwall-Dashboard
#   by Jason Cox - 27 Feb 2022 

# Make sure to install 7-zip from your package manager from the website
# use 7zz instead of 7z on line 45 if you are using the official binary

# Daily Backup for Powerwall-Dashboard Data
if [ "$EUID" -ne 0 ]
  then echo "Must run as root"
  exit
fi

# Set values for your environment 
DASHBOARD="/home/user/Powerwall-Dashboard"    # Location of Dashboard to backup
BACKUP_FOLDER="${DASHBOARD}/backups"          # Destination folder for backups
KEEP="5"                                      # Days to keep backup

# Check to see if directory exists
if [ ! -d "${DASHBOARD}" ]; then
  echo "Dashboard directory ${DASHBOARD} does not exist."
  exit
fi
if [ ! -d "${BACKUP_FOLDER}" ]; then
  echo "Backup directory ${BACKUP_FOLDER} does not exist."
  exit
fi

# Timestamp for Backup Filename
STAMP=$(date '+%Y-%m-%d')

# Optional: Ask InfluxDB to create a snapshot backup 
echo "Creating InfluxDB Backup"
cd ${DASHBOARD}
mkdir -p influxdb/backups
chmod g+w influxdb/backups
docker exec influxdb influxd backup -database powerwall /var/lib/influxdb/backups

# Backup Powerwall-Dashboard Data
echo "Backing up Powerwall-Dashboard Data (influxdb)"
cd  ${DASHBOARD}
tar -cvf ${BACKUP_FOLDER}/Powerwall-Dashboard.$STAMP.tar influxdb
7z a -t7z ${BACKUP_FOLDER}/Powerwall-Dashboard.$STAMP.7z ${BACKUP_FOLDER}/Powerwall-Dashboard.$STAMP.tar

# Cleanup Old Backups
echo "Cleaning up old backups"
rm -rf ${DASHBOARD}/influxdb/backups/*        # Delete InfluxDB snapshots after backup
find ${BACKUP_FOLDER}/Powerwall-Dashboard.*7z -mtime +${KEEP} -type f -delete
echo "Done"
```
</details>

s-crypt, 2023-08-26:
I figured out that it was because it was running out of memory. The problem was solved by using 2 threads as to not exceed ram usage. 2 threads uses 192MB of ram for compressing.

<details>
<summary>A ACTUALLY working backup-7z.sh</summary>

```bash
#!/bin/bash
#
# Backup Script for Powerwall Dashboard Data
#   https://github.com/jasonacox/Powerwall-Dashboard
#   by Jason Cox - 27 Feb 2022 

# Make sure to install 7-zip from your package manager from the website
# use 7zz instead of 7z on line 45 if you are using the official binary

# Daily Backup for Powerwall-Dashboard Data
if [ "$EUID" -ne 0 ]
  then echo "Must run as root"
  exit
fi

# Set values for your environment 
DASHBOARD="/home/pi/Powerwall-Dashboard"    # Location of Dashboard to backup
BACKUP_FOLDER="${DASHBOARD}/backups"          # Destination folder for backups
KEEP="5"                                      # Days to keep backup

# Check to see if directory exists
if [ ! -d "${DASHBOARD}" ]; then
  echo "Dashboard directory ${DASHBOARD} does not exist."
  exit
fi
if [ ! -d "${BACKUP_FOLDER}" ]; then
  echo "Backup directory ${BACKUP_FOLDER} does not exist."
  exit
fi

# Timestamp for Backup Filename
STAMP=$(date '+%Y-%m-%d')

# Optional: Ask InfluxDB to create a snapshot backup 
echo "Creating InfluxDB Backup"
cd ${DASHBOARD}
mkdir -p influxdb/backups
chmod g+w influxdb/backups
docker exec influxdb influxd backup -database powerwall /var/lib/influxdb/backups

# Backup Powerwall-Dashboard Data
echo "Backing up Powerwall-Dashboard Data (influxdb)"
cd  ${DASHBOARD}
tar -cvf ${BACKUP_FOLDER}/Powerwall-Dashboard.$STAMP.tar influxdb
#create archive using the 7z format, 2 threads-which uses 192MB of RAM for compressing
7z a -t7z -mmt2 ${BACKUP_FOLDER}/Powerwall-Dashboard.$STAMP.7z ${BACKUP_FOLDER}/Powerwall-Dashboard.$STAMP.tar

# Cleanup
echo "Cleaning up old backups"
rm -rf ${DASHBOARD}/influxdb/backups/*        					# Delete InfluxDB snapshots after backup
rm ${BACKUP_FOLDER}/Powerwall-Dashboard.$STAMP.tar	  			# Delete created tarball
find ${BACKUP_FOLDER}/Powerwall-Dashboard.*7z -mtime +${KEEP} -type f -delete #delete backups older than the amount of days requested. Default 5 days.
echo "Done"

```
</details>

BuongiornoTexas, 2023-08-26:
> The testing was all on the tar file which was packed with `tar`. When packing with `7z` the process keeps getting killed (and it takes significantly longer)

 Huh - I guess it really is dramatically better!

rcasta74, 2023-09-19:
On linux you could consider to use also `xz`, which uses the same compression algorithm (LZMA) as 7zip.
It is supported by `tar` command: just use `-J` instead of `-z`.
It is most likely installed by default on recent Linux.

dngray, 2023-11-11:
I would use tar.xz that's far more standard and doesn't require extra tools in most cases. It's used for things like kernel images, pacman, and apt etc.

jasonacox, 2023-11-13:
That's a good suggestion @dngray . It looks like the default compression level is a third of the size of the tar.gz version in my test. It does take longer, but I didn't try using the xz threading options. 

```
-rw-r--r-- 1 root root 284M Nov 12 20:46 Powerwall-Dashboard.2023-11-12.tar.xz
-rw-r--r-- 1 root root 925M Nov 12 19:54 Powerwall-Dashboard.2023-11-12.tar.gz
```

This would just require a few minor changes to the backup script.

```bash
tar -Jcvf ${BACKUP_FOLDER}/Powerwall-Dashboard.$STAMP.tar.xz influxdb 
...
find ${BACKUP_FOLDER}/Powerwall-Dashboard.*tar.xz -mtime +${KEEP} -type f -delete
```

What distros automatically include xz without installing a package?

s-crypt, 2023-12-16:
Finally got around to testing this

```
tar: 344mb (tar extracted from standard tgz)
tgz: 104mb (default backup file)
txz: 73mb  (tar option suggested by @rcasta74)
7z : 84mb  (my modified script from above)
```

Using tar.xz definitely took longer than tgz, but 7z is much longer.

@jasonacox wouldn't tar.xz be supported on all platforms that the backup is currently running on because the current backup solution uses tar and xz is just an option?

Considering the space savings, I would definitely be in favor of switching to tar.xz as it is built in, standard, and saves more space than the external 7z tools. The only thing to possibly mess around with is additional options to speed up the process. I am unsure of how long it will take on larger databases on lower end hardware (like a raspberry pi) and if that would be a prohibitive factor.

jasonacox, 2023-12-16:
I haven't found a platform that doesn't support xz (`tar -Jcvf`).  I'm in favor of moving to that. Feel free to submit a PR if you would like to make the necessary changes.

closed 2024-01-26 by s-crypt