jasonacox/Powerwall-Dashboard #342: Legends Were Moved to Right

cwagz, 2023-08-27:
**Problem**
On the newer dashboard.json, I believe since we upgraded the chart types, it seems the legends were moved to the right side. When viewing the dashboard from the iPhone these charts now become compressed and unusable due to the entire legend being rendered on the right.

**To Reproduce**
Open the dashboard from an iPhone using Safari and view the charts. The 12-month chart will be rendered very small so there is room for the legend on the right.

**Additional context**
I have edited the affected panels manually to place the legend on the bottom. Some of the legends now require scrolling to view the entire legend. I will need to look into this further to see if I can easily change the size of the panel, so the scrolling goes away.

If I get this all figured out, are there any objections to changing the legends back to "bottom" from "right"? I believe they were set to "bottom" previously, but I am not sure if moving them to the right was needed to fix some other issue.

jasonacox, 2023-08-27:
Hi @cwagz !

I see the original "12 month solar" panel had minimized legend...
[image]

The new one includes Mean and Max...
[image]

We could move that one back to a minimized (list not table) or put it under the graph. I don't think the man or max table adds a lot of value.

For the other (frequency, voltage, etc.) those have always had the table to the right. Did the "old" graph panel time work better on the iPhone? I believe it had the same issue but I may not be remembering correctly (had to switch to landscape my rotating the phone).  

Post screenshots of what you come up with that look nice. Key for me is to still have this look good for the desktop as the primary dashboard most people will have running.

cwagz, 2023-08-27:
Hi @jasonacox - It was mainly the 12-month panel that seemed really messed up now on the iPhone. I can't remember on the other panels, but it seems like they are also worse now on the iPhone. I will mess around with it when I have the time and see if I can get something that works well on the iPhone without harming the desktop view. I am a little concerned that the old graph style panels just worked better on the iPhone.