jasonacox/Powerwall-Dashboard #346: Powerwall Capacity showing No Data

droolingtaz, 2023-08-31:
**Problem**
[image]

However it appears that the vitals pyPowerwall output is showing data (serial numbers masked):

{
    "NEURIO--OBB3553103388": {
        "NEURIO_CT0_InstRealPower": -222.0800018310547,
        "NEURIO_CT0_Location": "site",
        "NEURIO_CT1_InstRealPower": 34.41999816894531,
        "NEURIO_CT1_Location": "site",
        "NEURIO_CT2_InstRealPower": 2891.780029296875,
        "NEURIO_CT2_Location": "solar",
        "componentParentDin": "STSTSM--1118431-00-L--TG1200300XXXXX",
        "firmwareVersion": "1.7.2-Tesla",
        "lastCommunicationTime": 1693516514,
        "manufacturer": "NEURIO",
        "meterAttributes": {
            "meterLocation": [
                1,
                4
            ]
        },
        "serialNumber": "OBB3553103388"
    },
    "STSTSM--1118431-00-L--TG1200300XXXXX": {
        "STSTSM-Location": "Gateway",
        "alerts": [
            "GridCodesWrite",
            "GridCodesWrite",
            "FWUpdateSucceeded",
            "SystemConnectedToGrid",
            "BatteryComms",
            "RealPowerAvailableLimited",
            "PodCommissionTime"
        ],
        "firmwareVersion": "2022-10-24-g64e8c689f9",
        "lastCommunicationTime": 1693516515,
        "manufacturer": "TESLA",
        "partNumber": "1118431-00-L",
        "serialNumber": "TG1200300XXXXX",
        "teslaEnergyEcuAttributes": {
            "ecuType": 207
        }
    },
    "TEPINV--1081100-10-Q--T19K00XXXX1": {
        "alerts": [
            "PINV_a067_overvoltageNeutralChassis"
        ],
        "componentParentDin": "TETHC--1092170-05-J--TG1200310XXXX1",
        "firmwareVersion": "8cd8cac4b9cb48",
        "lastCommunicationTime": 1692295532,
        "manufacturer": "TESLA",
        "partNumber": "1081100-10-Q",
        "serialNumber": "T19K00XXXX1",
        "teslaEnergyEcuAttributes": {
            "ecuType": 253
        }
    },
    "TEPINV--1081100-10-Q--T19K00XXXX2": {
        "PINV_EnergyCharged": 12397800,
        "PINV_EnergyDischarged": 11095710,
        "PINV_Fout": 60.001000000000005,
        "PINV_GridState": "Grid_Compliant",
        "PINV_HardwareEnableLine": true,
        "PINV_PllFrequency": 59.991,
        "PINV_PllLocked": true,
        "PINV_Pout": -0.01,
        "PINV_PowerLimiter": "PWRLIM_POD_Power_Limit",
        "PINV_Qout": 0.31,
        "PINV_ReadyForGridForming": true,
        "PINV_State": "PINV_GridFollowing",
        "PINV_VSplit1": 120.2,
        "PINV_VSplit2": 120,
        "PINV_Vout": 240.20000000000002,
        "alerts": [
            "PINV_a067_overvoltageNeutralChassis"
        ],
        "componentParentDin": "TETHC--1092170-05-J--TG1200310XXXX2",
        "firmwareVersion": "8cd8cac4b9cb48",
        "lastCommunicationTime": 1693516514,
        "manufacturer": "TESLA",
        "partNumber": "1081100-10-Q",
        "serialNumber": "T19K00XXXX2",
        "teslaEnergyEcuAttributes": {
            "ecuType": 253
        }
    },
    "TEPOD--1081100-10-Q--T19K0003056": {
        "alerts": [
            "POD_f003_FW_Batt_OV",
            "POD_f007_SW_Batt_OC",
            "POD_w058_SW_App_Boot",
            "POD_w093_SW_Charge_Request",
            "POD_w105_SW_EOD"
        ],
        "componentParentDin": "TETHC--1092170-05-J--TG1200310XXXX1",
        "firmwareVersion": "8cd8cac4b9cb48",
        "lastCommunicationTime": 1692295531,
        "manufacturer": "TESLA",
        "partNumber": "1081100-10-Q",
        "serialNumber": "T19K00XXXX1",
        "teslaEnergyEcuAttributes": {
            "ecuType": 226
        }
    },
    "TEPOD--1081100-10-Q--T19K00XXXX2": {
        "POD_ActiveHeating": false,
        "POD_CCVhold": false,
        "POD_ChargeComplete": true,
        "POD_ChargeRequest": false,
        "POD_DischargeComplete": false,
        "POD_PermanentlyFaulted": false,
        "POD_PersistentlyFaulted": false,
        "POD_available_charge_power": 0,
        "POD_available_dischg_power": 7000,
        "POD_enable_line": true,
        "POD_nom_energy_remaining": 12394,
        "POD_nom_energy_to_be_charged": 0,
        "POD_nom_full_pack_energy": 12680,
        "POD_state": "POD_ACTIVE",
        "alerts": [
            "POD_w110_SW_EOC"
        ],
        "componentParentDin": "TETHC--1092170-05-J--TG120031XXXXX",
        "firmwareVersion": "8cd8cac4b9cb48",
        "lastCommunicationTime": 1693516514,
        "manufacturer": "TESLA",
        "partNumber": "1081100-10-Q",
        "serialNumber": "T19K00XXXX2",
        "teslaEnergyEcuAttributes": {
            "ecuType": 226
        }
    },
    "TESYNC--1118431-00-L--TG1200300XXXXX": {
        "ISLAND_FreqL1_Load": 59.99,
        "ISLAND_FreqL1_Main": 59.99,
        "ISLAND_FreqL2_Load": 59.99,
        "ISLAND_FreqL2_Main": 59.99,
        "ISLAND_FreqL3_Load": 0,
        "ISLAND_FreqL3_Main": 0,
        "ISLAND_GridConnected": true,
        "ISLAND_GridState": "ISLAND_GridState_Grid_Compliant",
        "ISLAND_L1L2PhaseDelta": -256,
        "ISLAND_L1L3PhaseDelta": -256,
        "ISLAND_L1MicrogridOk": true,
        "ISLAND_L2L3PhaseDelta": -256,
        "ISLAND_L2MicrogridOk": true,
        "ISLAND_L3MicrogridOk": false,
        "ISLAND_PhaseL1_Main_Load": 0,
        "ISLAND_PhaseL2_Main_Load": 0,
        "ISLAND_PhaseL3_Main_Load": -256,
        "ISLAND_ReadyForSynchronization": true,
        "ISLAND_VL1N_Load": 119.5,
        "ISLAND_VL1N_Main": 119.5,
        "ISLAND_VL2N_Load": 119,
        "ISLAND_VL2N_Main": 119,
        "ISLAND_VL3N_Load": 0,
        "ISLAND_VL3N_Main": 0,
        "SYNC_ExternallyPowered": false,
        "SYNC_SiteSwitchEnabled": true,
        "componentParentDin": "STSTSM--1118431-00-L--TG1200300XXXXX",
        "firmwareVersion": "8cd8cac4b9cb48",
        "lastCommunicationTime": 1693516514,
        "manufacturer": "TESLA",
        "partNumber": "1118431-00-L",
        "serialNumber": "TG1200300XXXXX",
        "teslaEnergyEcuAttributes": {
            "ecuType": 259
        }
    },
    "TETHC--1092170-05-J--TG120031XXXX2": {
        "THC_AmbientTemp": 28.80000000000001,
        "THC_State": "THC_STATE_AUTONOMOUSCONTROL",
        "componentParentDin": "STSTSM--1118431-00-L--TG1200300XXXXX",
        "firmwareVersion": "8cd8cac4b9cb48",
        "lastCommunicationTime": 1693516514,
        "manufacturer": "TESLA",
        "partNumber": "1092170-05-J",
        "serialNumber": "TG120031001W48",
        "teslaEnergyEcuAttributes": {
            "ecuType": 224
        }
    },
    "TETHC--1092170-05-J--TG120031XXXX2": {
        "THC_AmbientTemp": 31.80000000000001,
        "THC_State": "THC_STATE_AUTONOMOUSCONTROL",
        "componentParentDin": "STSTSM--1118431-00-L--TG1200300XXXXX",
        "firmwareVersion": "8cd8cac4b9cb48",
        "lastCommunicationTime": 1693516514,
        "manufacturer": "TESLA",
        "partNumber": "1092170-05-J",
        "serialNumber": "TG1200310XXXX2",
        "teslaEnergyEcuAttributes": {
            "ecuType": 224
        }
    }
}

jasonacox, 2023-09-01:
Hi @droolingtaz  -  Your vitals are missing some key data points - you need these:

```json
        "POD_nom_energy_remaining": 8369.0,
        "POD_nom_energy_to_be_charged": 4828.0,
        "POD_nom_full_pack_energy": 12926.0,
```

This should be in the TEPOD device vitals section like this:

```json
"TEPOD--1081100-10-U--x": {
        "POD_ActiveHeating": false,
        "POD_CCVhold": false,
        "POD_ChargeComplete": false,
        "POD_ChargeRequest": false,
        "POD_DischargeComplete": false,
        "POD_PermanentlyFaulted": false,
        "POD_PersistentlyFaulted": false,
        "POD_available_charge_power": 7000.0,
        "POD_available_dischg_power": 12000.0,
        "POD_enable_line": true,
        "POD_nom_energy_remaining": 8369.0,
        "POD_nom_energy_to_be_charged": 4828.0,
        "POD_nom_full_pack_energy": 12926.0,
        "POD_state": "POD_ACTIVE",
        "componentParentDin": "TETHC--3012170-05-B--x",
        "firmwareVersion": "452c76cbd7a6f3",
        "lastCommunicationTime": 1693536600,
        "manufacturer": "TESLA",
        "partNumber": "1081100-10-U",
        "serialNumber": "x",
        "teslaEnergyEcuAttributes": {
            "ecuType": 226
        }
    },
```

Looking at your vitals output, it looks like you have two Powerwall 2's of type 1092170, one of which is expressing a lot of alerts.  My Powerwall 2 is a 2012170.  I wonder if the 1092170 Powerwall even records the  "nominal energy" datapoints. Can anyone else confirm?

Unfortunately, we don't have enough information on the alerts your system is showing. Some have posted these during system failures, but they may still be nominal versus indicative of failure.

droolingtaz, 2023-09-01:
ok so it may be related to issues I am having with PW1.  Tesla is in the process of replacing it.  I should have the replacement in next week.  I will check this again after that and report if it is still an issue.

droolingtaz, 2023-09-01:
even though one pw is not reporting the data, should it not at least be graphing the data from the one functioning unit?

jasonacox, 2023-09-01:
Yes, it is confusing. Neither of you Powerwalls are sending the "nominal power" data.  They are both type 1092170 (which shows up in the TETHC sections, which I think is the first Powerwall 2 but I could be wrong). It could be that this version doesn't report on nominal energy.  In the TEPOD sections (you have two representing both of your Powerwalls), you will want to see these elements:

```json
        "POD_nom_energy_remaining": 8369.0,
        "POD_nom_energy_to_be_charged": 4828.0,
        "POD_nom_full_pack_energy": 12926.0,
```

The `POD_nom_full_pack_energy` is the key one to plot degradation over time.

Your vitals post doesn't show any of these. Since the elements are missing, the telegraf (the tool used by the Dashboard to fetch and store the data) will not post anything to the database (influxdb). Grafana renders that as "no data".

droolingtaz, 2023-09-07:
@jasonacox I just got that faulty PW replaced and now it is showing capacity data.

[image]

jasonacox, 2023-09-08:
That's great news, @droolingtaz !  Thanks for the update. Congrats on getting it fixed!

closed 2023-10-30 by droolingtaz