jasonacox/Powerwall-Dashboard #348: Utility Bill Comparison to Dashboard

aesculus, 2023-09-01:
Well after running the system for a month I just compared it to my utility bill. It was spot on. Maybe less than 1/2% off on some power figures but certainly well within tight tolerances.

This tool is going to be very valuable in monitoring and tuning optimal Powerwall and energy use.

What a great tool and thank you for putting this together.

jasonacox, 2023-09-02:
Hi @aesculus, I appreciate you taking the time to share your positive experience with the project and the measurement observation. Your kind words are a great source of motivation for all the dashboard contributors. Thank you! 🙏

hulkster, 2023-10-06:
It's not even close to my utility bill that just came in today. 

LOL (not really!) this is because Xcel is buffoons and despite PTO in late August, their billing department hasn't _"caught up"_ yet ... it _"takes a while"_ ... over a month - WTF?!?

In case my sarcasm isn't obvious, GREAT job @jasonacox for your fine work on a really nice tool.

jasonacox, 2023-10-07:
> it "takes a while" to "enable solar" on my account.

Keep an eye on it! We had a similar thing after our PTO. The bill didn't change. Our utility company said it could take a month or two. I check in with them after a month when it wasn't fixed and found out the "change" didn't go through so they had to do it again.  A few months later and it was fixed. It's a bit ridiculous, but in the end, they recomputed all the months to did the necessary adjustments.