jasonacox/Powerwall-Dashboard #350: Question about having two powerwall +

akotalik, 2023-09-03:
**Problem Description**
Installed the dashboard using the provided instructions without an issue, but the dashboard only shows data from a single inverter. I have 2 powerwall + with a single backup gateway 2. I thought I could just connect to the gateway, but it seems I need to connect to each inverter separately? Is there a way to combine into a single dashboard? Or how would I even see the other inverter?

**System Details**
Describe the host system you are trying to use for the Dashboard:
* Raspberry Pi
* Debian gnu/linux 11

akotalik, 2023-09-03:
Or I suppose a simple way to somehow install the dashboard twice on the same raspberry pi and be able to switch between inverters in graphana within single session? Only way I can see myself is buying another raspberry pi :D

akotalik, 2023-09-03:
Apologies- it does show both, never mind!

closed 2023-09-03 by akotalik

jasonacox, 2023-09-03:
Congratulations, @akotalik on getting it working! Any suggestions on how we could make it better (or less confusing) to setup?

akotalik, 2023-09-03:
Well for me the most confusing part is with 2 inverters and 1 gateway, each has its own password and it was not clear which one it wanted from me. Also the fact that it's only the last 5 characters. I ended up using the password from one inverter and somehow it lets me see the other too.

Also, the power flow animation on and off works and doesn't? Connection was reset error

akotalik, 2023-09-04:
Hm well I hardwired my gateway with an ethernet cable and sure enough, it still doesn't work. Not I can't get any data other than weather. I even tried to uninstall the dashboard and reinstall everything. Any ideas?

akotalik, 2023-09-04:
Hm well I hardwired my gateway with an ethernet cable and sure enough, it still doesn't work. Not I can't get any data other than weather. I even tried to uninstall the dashboard and reinstall everything. Any ideas?

logs:
Traceback (most recent call last):
  File "/app/server.py", line 108, in <module>
    pw = pypowerwall.Powerwall(host,password,email,timezone,cache_expire,timeout,pool_maxsize)
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 138, in __init__
    self._get_session()
  File "/usr/local/lib/python3.10/site-packages/pypowerwall/__init__.py", line 151, in _get_session
    raise ConnectionError(err)
pypowerwall.ConnectionError: Unable to connect to Powerwall at https://192.168.91.1
09/04/2023 12:08:48 PM [proxy] [INFO] pyPowerwall [0.6.2] Proxy Server [t26] - HTTP Port 8675
09/04/2023 12:08:48 PM [proxy] [INFO] pyPowerwall Proxy Started

jasonacox, 2023-09-04:
Hi @akotalik - check the IP address now that you are hardwired. It should not be the same as the WiFi address you show above (192.168.91.1) as that is what the Powerwall AP uses.  You can scan for the address using pypowerwall from the same network the ethernet cable is on:

```bash
pip install pypowerwall # install the Python package
python3 -m pypowerwall scan
```

It should scan your network (takes a while) and then outputs the address:

```
jason@pi:~$ python3 -m pypowerwall scan

pyPowerwall Network Scanner [0.6.2]
Scan local network for Tesla Powerwall Gateways

    Your network appears to be: 10.0.1.0/24

    Enter Network or press enter to use 10.0.1.0/24: 

    Running Scan...
      Host: 10.0.1.6 ... OPEN - Not a Powerwall
      Host: 10.0.1.8 ... OPEN - Found Powerwall 1232100-00-E--TGxxxxxxxxxxxG
                                     [Firmware 23.12.11 452c76cb]
      Host: 10.0.1.9 ... OPEN - Not a Powerwall
      Host: 10.0.1.13 ... OPEN - Found Powerwall 1232100-00-E--TGxxxxxxxxxxxG
                                     [Firmware 23.12.11 452c76cb]
      Done                           

Discovered 2 Powerwall Gateway
     10.0.1.8 [1232100-00-E--TGxxxxxxxxxxxG] Firmware 23.12.11 452c76cb
     10.0.1.13 [1232100-00-E--TGxxxxxxxxxxxG] Firmware 23.12.11 452c76cb
```

akotalik, 2023-09-05:
Yep, that sure fixed it. Works great now! Thanks a lot for your help, love the product