jasonacox/Powerwall-Dashboard #362: Grafana container keeps restarting due to volume issue

bparruck, 2023-10-04:
**Grafana container doesn't have its mount**

`docker logs grafana`
GF_PATHS_DATA='/var/lib/grafana' is not writable.
You may have issues with file permissions, more information here: http://docs.grafana.org/installation/docker/#migrate-to-v51-or-later
mkdir: cannot create directory '/var/lib/grafana/plugins': Permission denied
`
docker ps`
CONTAINER ID   IMAGE                            COMMAND                  CREATED         STATUS                         PORTS                          NAMES
963f28fddfb7   jasonacox/weather411:0.2.2       "python3 server.py"      7 minutes ago   Up 11 seconds                  0.0.0.0:8676->8676/tcp         weather411
4dfedc99ff30   telegraf:1.26.1                  "/entrypoint.sh tele…"   7 minutes ago   Up 11 seconds                  8092/udp, 8125/udp, 8094/tcp   telegraf
231ad69520dc   grafana/grafana:9.1.2-ubuntu     "/run.sh"                7 minutes ago   Restarting (1) 3 seconds ago                                  grafana
d138eb995342   jasonacox/pypowerwall:0.6.2t27   "python3 server.py"      7 minutes ago   Up 12 seconds                  0.0.0.0:8675->8675/tcp         pypowerwall
b27a0664b5d5   influxdb:1.8                     "/entrypoint.sh infl…"   7 minutes ago   Up 12 seconds                  0.0.0.0:8086->8086/tcp         influxdb

**To Reproduce**
Tried on a brand new installation of Ubuntu 22.04.3 with Docker Desktop
`lsb_release -a`
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy

**Host System**
`sudo lshw -short`

H/W path             Device     Class          Description
==========================================================
/0/3e                           processor      Intel(R) Core(TM) i7-4770R CPU @ 3.20GHz
/0/3e/3f                        memory         1MiB L2 cache
/0/3e/40                        memory         256KiB L1 cache
/0/3e/41                        memory         6MiB L3 cache
/0/42                           memory         16GiB System Memory
/0/100/1f.2          scsi4      storage        8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode]
/0/100/1f.2/0.0.0    /dev/sda   disk           500GB Samsung SSD 850
/0/100/1f.2/0.0.0/1             volume         511MiB Windows FAT volume
/0/100/1f.2/0.0.0/2  /dev/sda2  volume         465GiB EXT4 volume

**Verify Scripts thinks everything is hunky-dory:**

`./verify.sh `
Verify Powerwall-Dashboard - Version 2.9.12 - Timezone: America/Los_Angeles
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.6.2 Proxy t27
 - Powerwall State: CONNECTED - Firmware Version: 23.12.11
 - ERROR: Your pypowerwall settings are missing TZ.

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.26.1 (git: HEAD@8f421f25)

Checking influxdb
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD

bparruck, 2023-10-04:
**Here is the fix for the issue:**
```
./compose-dash.sh stop
chmod 777 grafana
./compose-dash.sh start
```

closed 2023-10-04 by bparruck

jasonacox, 2023-10-07:
@bparruck did you use the `setup.sh` script and still needed to run this `chmod` fix?   

TODO: might be good to figure out how to added permission setting checks to `verify.sh`.

bparruck, 2023-10-07:
@jasonacox The real issue was Docker Desktop. However, the verify.sh does have an issue.