jasonacox/Powerwall-Dashboard #365: Add more check to verify.sh

jasonacox, 2023-10-07:
Verify script is not checking for file system permission. Investigate possible options to have it help troubleshoot permission issues.

_Originally posted by @bparruck in https://github.com/jasonacox/Powerwall-Dashboard/issues/362#issuecomment-1751774549_

bparruck, 2023-10-07:
@jasonacox It would use to check that all containers are coming up properly.

jasonacox, 2023-10-07:
There is a "container running" check, but the issue is that the services that need to write to the disk (influxdb and grafana) appear to be "up" even if they can't modify the filesystem. I added a simple check for both to ensure they have write access to the folder they should own:

```bash
echo -e -n "${dim} - Filesystem (./$CONTAINER): "
rm -f ./grafana/WRITE
ERR=`docker exec -it grafana touch /var/lib/grafana/WRITE`
if [ -e "./grafana/WRITE" ]; then
    echo -e $GOOD
    rm -f ./grafana/WRITE
else
    echo -e "${alert}ERROR: Unable to write to filesystem - check permissions${normal}"
fi
```

BuongiornoTexas, 2023-10-07:
@jasonacox - I suspect this is part of the long running issue around grafana running a non-root user id in the container, and expecting an appropriately mapped sub userid/groupid in the host space - which is often not there and is why the chmod +777 is needed to allow universal write access. 

A long term fix would be to address this underlying problem of making sure that the grafana suid/guid in the user space are correct for the version of docker being run. It's on my long term backlog to look at, as it affects me each time I upgrade.

Stoobiedoo, 2024-10-02:
I think my problem is related to this, probably because I messed up permissions when trying to move my docker data to a different locations.
I am getting an error on the grafana container that makes it continually restart. The error is: "Error: ✗ failed to create temporary file: open /tmp/2853355834.zip: permission denied" or similar with different numbers.
Is that the actual root file system /tmp folder or somewhere in my /home/stuart/Powerwall-Dashboard folder? Or somewhere else like /var/lib/grafana/

jasonacox, 2024-10-03:
When you say you moved the data, did you re-run setup? The containers will need to be recreated so the volume mounts point to the new location.

Stoobiedoo, 2024-10-03:
Thank you. I brute forced read write permissions on everything in the new location for the docker data and it seems to be working now. Probably not the best way, but I'm still learning.