jasonacox/Powerwall-Dashboard #373: Update Versions of Grafana and Telegraf

jasonacox, 2023-10-15:
This is a proposal to upgrade the pinning to newer version of Grafana and Telegraf:

* Grafana (v9.4.14)
* Telegraf (v1.28.2)
* pyPowerwall (v0.6.2t28) - Required to support background color adjustments for newer Grafana versions

During testing, the "look" difference between Grafana v9.1.2 and v9.4.14 are minor. There is a subtle adjustment to the font and labeling (IMHO looks better). No issues with panels. The Power Flow Animation required an update to pyPowerwall to accommodate the grafana-dark background color (transparent settings were in use before, but no longer work).

Related discussions:
* https://github.com/jasonacox/Powerwall-Dashboard/discussions/371
* https://github.com/jasonacox/Powerwall-Dashboard/discussions/363
* https://github.com/jasonacox/Powerwall-Dashboard/issues/368

rcasta74, 2023-11-04:
Why not use Alpine-based images?
```
docker.io/grafana/grafana        9.4.14                f69ac1c180a5  7 weeks ago    332 MB
docker.io/grafana/grafana        9.4.14-ubuntu         76a0cce050ec  2 months ago   428 MB
```
```
docker.io/library/telegraf       1.28.2-alpine     3d231738b7db  2 weeks ago    210 MB
docker.io/library/telegraf       1.28.2            f10f8ba7295f  3 weeks ago    453 MB
```

jasonacox, 2023-11-04:
Good question. Is there a reason? Key is to make sure we target all architectures (amd64, arm32, arm64) to ensure it they have builds that run on things from Raspberry Pis, x86 to Macs.

BuongiornoTexas, 2023-11-05:
> Good question. Is there a reason? 

I know I ran into an issue when I was doing my ToU service - alpine does not include gcc (required by pandas). Not sure if this could affect any of the core modules in the dashboard. I also have a memory that there is an issue with RPi support for some of the python versions.

jasonacox, 2025-12-31:
Latest release moves us to Grafana 12. Closing this for now.

closed 2025-12-31 by jasonacox