jasonacox/Powerwall-Dashboard #382: Separate Car Charging Energy

droolingtaz, 2023-10-30:
**Problem**
Currently when I am charging my car, its energy is included in the house consumption.

**Enhancement**
In the tesla app it shows the car charging energy as a separate number from the house energy consumption
[image]

would it be possible to make same separation in the dashboard?

jasonacox, 2023-10-31:
Since it is 🎃  - Trick or treat... If someone will buy me a Tesla, I can probably make that happen. 😂  Yes, I got solar before getting a Tesla EV.

In all seriousness, the answer is a bit complicated.  The Tesla app has access to all your devices so it is bridging data.  The Powerwall likely only sees the power for the car charging as house "load" so since pypowerwall just pulls that data from the Powerwall, it isn't able to differentiate it. 

An option would be to pull the data from the Tesla cloud for your car charing data (or is there a local way to do that direct from the car... or does it even appear in the Powerwall gateway portal?) and inject that data into the InfluxDB that can then be subtracted back out of house load data during dashboard rendering (and new data points added for the car).

mcbirse, 2023-10-31:
For my Polestar 2 I had a myenergi Zappi installed to charge the car a while back.

I have been wanting to see the car charging power usage separate to the home usage as well.

Partly because I would like to calculate EV charging costs accurately, since the charge source could be coming from excess solar, Powerwall, grid, or a combination, and each have different costs. Also, because it would be cool to see in the dashboard of course!

So, I have been working on adapting my tesla-history script to pull the car charging data from the myenergi Zappi cloud API. A python module [pymyenergi](https://github.com/CJNE/pymyenergi) was available making this quite easy, thankfully.

I am basically finished, but just have an issue with spare time to finally complete this. I am getting the raw data into InfluxDB, but have not go around to setting up some Grafana panels or modifying the dashboard to include the data yet.

Here's an example of the script running as a docker container. It is very similar to the tesla-history solar-only version, so I'd done a lot of the ground work on the script already, making it reasonably easy to adapt for this purpose. 🎉 

```
zappi-history Server [0.0.9]
* Configuration Loaded [/var/lib/zappi-history/zappi-history.conf]
 + Server - Wait: 5m, Hist: 60m, Retry: 30s, Log: yes, Debug: no, Test: no
 + Myenergi - Hub: 99999999, Key: xxxxxxxxxxxxxxxxxxxxxxxx
 + InfluxDB - Host: influxdb, Port: 8086, DB: powerwall, FIELD: zappi, Timezone: Australia/Sydney
----------------------------------------
 Myenergi Hub: 99999999
----------------------------------------
    Zappi S/N: 12345678
    Site name: My home
     Timezone: Australia/Sydney
  System time: 2023-10-31 17:13:28+11:00
----------------------------------------
* Server Started
 + Searching InfluxDB for data gaps >= 5 minutes every 5 minutes
 + Retrieving up to 60 minutes of history
Searching InfluxDB for data gaps >= 5 minutes
* Found data gap: [2023-10-31 17:14:00+11:00] - [2023-10-31 17:20:05+11:00] (0:06:05s)
Retrieving data for gap: [2023-10-31 17:15:00+11:00] - [2023-10-31 17:20:05+11:00] (0:05:05s)
* Loading daily history: [2023-10-31] (Australia/Sydney)
Writing to InfluxDB
```

The raw data from myenergi is a bit weird, but workable. Below is an example of the fields.

[image]

The car charging data is the addition of "h1b" and "h1d" depending on charge modes (normal, boost, etc.), and will need a Grafana transform to combine multiple queries into a single value.

I could do this in the script itself, but would prefer any scripts like this just pull the raw data from whatever API and dump it into InfluxDB as is, where it could then be interpreted and displayed.

Here's data from today which I just quickly cobbled together. I have the zappi data going into a separate field in InfluxDB named "zappi" with the car charging data in green in the below graph. My car was charging from excess solar today.

[image]

The yellow line is the "home" power usage from the Powerwall, but would include the car charging. Ideally I think I'd like to see just the home usage portion not including my car charging, and should be able to calculate that since I have the zappi charge data (as mentioned, have not had time to work on actual Grafana panels/queries yet).

It would be interesting to find out, can we pull similar data from the Tesla cloud/app?

If so, then perhaps once I've finished the work on this zappi-history script, we can adapt it again for Tesla car charging.

I was already planning perhaps to adapt this for ChargePoint since it was mentioned in #242 (and @jasonacox even suggested adapting the tesla-history for this 😁 ). Just needs work investigating how to get data from the ChargePoint API.

youzer-name, 2023-10-31:
TLDR: There may be a simple solution for Tesla cars by using Teslamate and Node Red to get the charging data into InfluxDB. 

This discussion got me to think about how I could bring this data into my dashboards. I use a Grizzl-E Duo EVSE which has no WiFi or data reporting, so I can't get data directly from the EVSE or the Tesla cloud. I do however have Teslamate set up, and it tracks all the charge data in real-time and publishes a lot of data, including the charger_power in kW, to MQTT.

It is fairly simple to use Node Red to take the charger_power data that Teslamate publishes to MQTT and insert it into InfluxDB. While initially looking at the data I realized I should only include data when the car's geofence = 'home'. In Node Red I'm using an MQTT listener node to get the data,  passing it to a function node that checks whether the car is at home and formats the message for the database, and an InfluxDB node to store the data.

I started a charge on one of my cars and can see that there might be some rough edges. I only had it charging for a minute or two, so I had to zoom way in to see the data on the chart. The dashboard was showing home use as 5 kW while the charger power was showing 8 kW, so when looking at very short time spans it may not work well due to the mean values stored in the kwh measurement and the slight time delay between what the Powerwall gateway sees and what Teslamate reports. I expect that when zoomed out to look at 24 hours of data, that probably won't be an issue. After the cars charge up overnight I'll take a look at the data and see if I can do anything interesting with it in Grafana.

jasonacox, 2023-11-01:
@mcbirse That's brilliant!  I added Polstar to my wish list too. ;)

@youzer-name Nice work! Feel free to share too. The offset seem a little annoying. I suppose it  could be caused by mean value differences, but is it also due to sample rate?

youzer-name, 2023-11-01:
> @youzer-name Nice work! Feel free to share too. The offset seem a little annoying. I suppose it could be caused by mean value differences, but is it also due to sample rate?

I'm pretty sure that what i saw was due to the downsampling / mean values in kwh, because the car was drawing more power than the home use was showing. Over a longer time interval it won't matter.

After setting things up using Node Red and getting my dashboards set up, I realized that pulling the data directly from the Teslamate database was not nearly as difficult as I had originally thought, so I shifted to that method.

For anyone else who wants to do this with Teslamate, you'll need to either update the Teslamate stack so it publishes a port for the database, or if you run Teslamate and the Powerwall Grafana on the same Docker instance, you can create an overlay network in Docker that includes both Powerwall Dashboard's Grafana and Teslamate's PostgreSQL containers. I did the latter, and also installed a PGAdmin container (a web-based tool for interacting with the database) which is also in the overlay network.

Once you have the database accessible by either a published port or a common network, you need to add the PostgreSQL data source to Grafana and set up the connection to Teslamate's DB.

Here are some screenshots:

Last 30 days (from my "Powerwall History" dashboard):

[image]

Taking a slightly wider screenshot and looking at the last 90 days, this shows I could have powered the house entirely from solar if not charging the EVs*:

[image]

(* = there are a bunch of charges missing from Teslamate over the past few months due to a Tesla software glitch that was causing the cars to go offline when scheduled charging started, so the 'EV charging' number is likely an undercount, and the solar production compared to non-EV home use would be even higher.)

On my energy usage panel I added the car power usage as shown below. At the bottom of the chart are blue and white lines that I previously had added as indicators for when either car was charging (my cars are blue and white). Those are based on the 'state' of the car, either 'charging'  or <> 'charging'. I will probably hide or remove those since they're redundant now. What I just added are the blue and white areas on the main plot area that show EV charging over top of the home/grid/solar data. I set each with opacity at 50% and stacking turned on. When both cars are charging it will show them stacked. I always charge on grid power, using my home automation system to set my reserve to 100% when charging starts, so whatever grid use shows in purple above the blue or white area is the non-EV home use.  

[image]

<details>

<summary>Click to view the query for the energy usage panel:</summary>

SELECT date as time, (charger_voltage * charger_actual_current) as "fieldname"
FROM charges c
join charging_processes p 
ON p.id = c.charging_process_id 
WHERE $__timeFilter(date) AND p.car_id = 1 and geofence_id = 2

set the fieldname, car_id, and geofence_id based on your own Teslamate setup. Then set field overrides for fields with name 'carname'. 

I set the field name to "3charging" and '"ycharging" for my two cars, so my overrides for the Model Y look like this:
[image]

</details>

<details>

<summary>Click to view the query and JSON for the stat panel:</summary>

select  cast('1970-01-01' as timestamp), sum(charge_energy_added) * 1000 as "EV charging"
FROM charging_processes 
WHERE geofence_id = 2 and $__timeFilter(end_date)

Again, set the geofence_id to the correct value for your Teslamate setup.

The full panel JSON is:

{
  "__inputs": [
    {
      "name": "DS_INFLUXDB",
      "label": "InfluxDB",
      "description": "",
      "type": "datasource",
      "pluginId": "influxdb",
      "pluginName": "InfluxDB"
    },
    {
      "name": "DS_POSTGRESQL-PI_DATABASE_(TESLAMATE)",
      "label": "PostgreSQL-pi_database (Teslamate)",
      "description": "",
      "type": "datasource",
[... 220 lines cut ...]
      ],
      "title": "Home Usage",
      "transformations": [],
      "type": "stat"
    }
  ],
  "schemaVersion": 37,
  "style": "dark",
  "tags": [],
  "templating": {
    "list": []
  },
  "time": {
    "from": "now-6h",
    "to": "now"
  },
  "timepicker": {},
  "timezone": "",
  "title": "export 2023-11-01",
  "uid": "j__-YNmgz",
  "version": 1,
  "weekStart": ""
}

</details>

jasonacox, 2023-11-02:
This is great @youzer-name - thanks for sharing!

mcbirse, 2023-11-02:
Thanks @youzer-name - this is great, and gives me some ideas for Grafana panels for the myenergi Zappi charger data as well.

So I guess your approach should satisfy Tesla owners with Tesla chargers who want to see the car charging data in the dashboard.

Considering it appears the data needs to be obtained in "real time" for the Tesla charging, a script similar to the Tesla-History is not going to be possible.

This is somewhat different to what I am currently developing for the Zappi where the myenergi API provides historical data, which is nice.

In fact, the myenergi historical data is better than the Tesla cloud, and generally available in per minute intervals (as opposed to 5 minute for Tesla cloud). I have 3 CT's connected to my Zappi charger, monitoring home, solar, and Powerwall, so essentially have a more accurate cloud backup of all my power usage than what the Tesla cloud provides. 😄 

I will release a "zappi-history" tool once I'm finished, which will enable those with myenergi Zappi chargers to get the car charging data into InfluxDB to be able to display this in the dashboard.

Adapting the script to work with ChargePoint chargers could be possible too, but I will need to investigate the API and may need help/access to someone with a ChargePoint charger to develop and test this.

youzer-name, 2023-11-02:
> Thanks @youzer-name - this is great, and gives me some ideas for Grafana panels for the myenergi Zappi charger data as well.
> 
> So I guess your approach should satisfy Tesla owners with Tesla chargers who want to see the car charging data in the dashboard.

Just to clarify, the Teslamate approach works for anyone with a Tesla car, regardless of charger.

youzer-name, 2023-11-03:
My original Teslamate queries for the Energy Usage panel didn't work right when both cars were charging at once, as Grafana will only stack the values if the timestamps are identical. This is where the Influx group by and fill features make things much easier. There were also some cases where there was no 0 power value at the end of a charge, so Grafana would connect one charge to the next one when looking at certain time scales. 

After struggling with the queries all day, I was able to come up with something that works by ensuring there is a value for both cars on every row... either the current value for that car, or the previous value for the other car. This would have to be rewritten for more than 2 cars and would be much simpler with only one. 

Due to some other Grafana charting weirdness that I couldn't seem to work out, I went with one total EV Charging value on the Energy Usage chart, rather than color coding for each car and trying to stack. I'll just keep my indicators at the bottom of the chart to show which car is charging. 

Once I have some time to make sure there are no more surprises, I'll edit my earlier post to update the query.

youzer-name, 2023-11-06:
My earlier post has now been updated. I tried to add an explanation of how the query I wrote works, but I'm not sure I made a lot of sense.

gusinghbal, 2024-08-27:
> My earlier post has now been updated. I tried to add an explanation of how the query I wrote works, but I'm not sure I made a lot of sense.

hello @youzer-name 
Thanks for your efforts in integrating TeslaMate with powerwall dashboard. Can I please request some assistance in setting mine up, please. I have the pw grafana dashboard working perfect on :9000 and Teslamate grafana working perfect on :3000 individually. So I log onto one and check and the other one logs off and then check the other one.
I have both the dashboard running on the same RPi under one docker instance. I tried adding the postgres db to the pw grafana and vice versa but not able to see the data. I see that you mentioned creating a network for the two, but Im unable to get that setup properly. Can you please provide a bit more detailed steps on how to achieve that or you many have some detailed instructions already, if you can link me to those please.
@jasonacox : Thank you very much for this great tool, really appreciate all your efforts.
Best Regards,
Garry.

youzer-name, 2024-08-27:
TLDR: The easiest solution is to modify the TeslaMate compose file to expose the database port.

@gusinghbal  The issue is that the default TeslaMate setup doesn't expose any ports on the database container, so it is only accessible from within the network that Docker creates for the TeslaMate stack. An overlay network lets the two containers connect so you can set up your Grafana PostgreSQL connection to point to "teslamate-database-1:5432".

I originally had both TeslaMate and PWD running on the same Pi / Docker instance, so I was using an overlay network, but
I've since moved my TeslaMate setup to a separate Raspberry Pi from the one that runs PWD so that wouldn't work for me now. Instead, I just modified the TeslaMate installation so the database container exposes port 5432, and then I set up the Grafana database connections to connect to the other Pi using its IP on that port. 

That method should also work if the two containers are on the same Pi. If you set TeslaMate to expose port 5432 outside of the container, you can use the IP and port to connect from the Grafana running on the same machine.

I added the last two lines below to the TeslaMate docker-compose.yml file in the database section: 
```
  database:
    image: postgres:15
    restart: always
    environment:
      - POSTGRES_USER=${TM_DB_USER}
      - POSTGRES_PASSWORD=${TM_DB_PASS}
      - POSTGRES_DB=${TM_DB_NAME}
    volumes:
      - teslamate-db:/var/lib/postgresql/data
    ports:
      - 5432:5432
```

gusinghbal, 2024-08-27:
Thank You @youzer-name , So do I just run the one grafana that is on port :9000 being used for Powerwall and add the postgres datasource after exposing the port/overlay network and upload TeslaMate dashboards?
I will open the 5432 port in Teslamate and see how that goes. Will let you know here of the outcome.
Thanks.

youzer-name, 2024-08-27:
I still run both Grafana instances since I don't want to have to update things every time TeslaMate makes changes on their end. You could certainly copy everything over, but that is going to be a lot of work once you dig into it. I think I started to do that once and decided it wasn't worth it just to duplicate the panels that are already in TeslaMate.

So I'm only using the connection to create my own custom queries to combine the data, like showing the charging use on the Energy Usage panel, or showing the car status and charge level at the top of my main dashboard.

Since my original posts in this thread, I started using Tessie which is similar to TeslaMate but is cloud hosted (and not free). They have a great API where you can access the data that they're pulling from Tesla. Most of my panels are now using data that I pull from Tessie's API and store in either InfluxDB or MySQL rather than data from PostgreSQL/TeslaMate.

I still use TeslaMate and refer to it when I'm looking for charge/drive/trip information, but I'm getting my charging energy use data from Tessie.

gusinghbal, 2024-09-13:
@youzer-name 
Hello. 
Thanks for the setup help. I managed to have the powerwall dashbord and Teslamate running on the same RPi under different containers in docker. I have exposed the db port as you suggested and can log into each grafana (one at a time) and see the powerwall dashboard and Teslamate dashboards. 
Can I please seek your assistance with :
"
Once you have the database accessible by either a published port or a common network, you need to add the PostgreSQL data source to Grafana and set up the connection to Teslamate's DB.
"
for adding the PostgreSQL data source to Grafana and setting up the connection to Teslamate DB, please. I believe I'll be adding the PostgreSQL data source to the powerwall dashboard Grafana, right? connection to Teslamate DB?
Then I think I need to follow your query to have the charging data show in the powerwall dashboard. I think I'll get to that when the data is setup.
Any help is much appreciated.
Thanks.

youzer-name, 2024-09-15:
Correct, you need to add the PostgreSQL data source in the PWD Grafana. The URL is "RaspberryPi_IP:5432", assuming you exposed the default PostgreSQL port. The database name is teslamate. The username should be teslamate and the password is whatever you used when setting up Teslamate. You should be able to find that password in the Teslamate setup yml file. You should be able to leave all the other fields at their default values and click "Save & Test" to check that the connection is working.

Once you have the connection set up, you can add whatever data you want from Teslamate to any PWD panel or dashboard.

If you want to bring entire Teslamate dashboards over into the PWD Grafana, you can open the TeslaMate Grafana and use the "Share" button at the top of any dashboard. Click that and select the export tab. Turn on the switch for "Export for sharing externally". You can then "Save to file" or "View JSON". 

You can then go to the PWD Grafana... Dashboards -> New -> Import.

If you exported the file, you can import it here. If you copied the JSON to the clipboard, paste it into the "Import via dashboard JSON model" box. 

On the next screen, you can edit the name, pick a folder, and pick the database connection for the dashboard. Make sure you select the PostgreSQL connection.

You would need to do this for each dashboard in TeslaMate that you want to bring into the other Grafana instance.

gusinghbal, 2024-09-16:
Thanks @youzer-name 
I added the PostgreSQL data source to PWD grafana successfully. Just had to disable TSL/SSL Mode and tested connection OK.
Now to setup to show the charging data in the PWD dashboard.
I do not see the need to bring the entire Teslamate dashboards to Grafana yet, 
So, I just want to setup the charging data to show up in the usage data as your screenshots.
I have two tesla EVs and one wall connector and one 32Amp 3phase universal mobile charger, So need to set up the charging data of the two cars. I rarely have both the EVs charging at the same time, but there will be some times occasionally when both are charging.
basically just want to see what you showed in screenshots. 
I guess I will have to use the two car query that you linked, just not sure how and where this query needs to be pasted/set.
If you can give some direction that will be great.
Thanks.

youzer-name, 2024-09-19:
@gusinghbal  
For the energy usage panel query, you'll want to edit the existing energy usage panel. 

On older versions of Grafana, click the down arrow next to the title
[image]

On newer versions, click the three dots at the top right of the panel:
[image]

Then click edit.

Scroll down to the bottom of the list of queries at the bottom left of the screen.  Click the button to add a new query.

Older Grafana:
[image]

Newer Grafana:
[image]

Select the data source for PostgreSQL, then click the pencil/edit button at the right to go into text edit mode:

[image]

Paste the query there.

For the stat panel, the steps are similar, but you'll want to start by creating a new panel

Older Grafana: