jasonacox/Powerwall-Dashboard #383: Unable to pull weather41 image with podman

rcasta74, 2023-11-03:
I'm trying to update `weather41` image with podman but I get this error:
```
❯ podman pull docker.io/jasonacox/weather41:0.2.2
Trying to pull docker.io/jasonacox/weather41:0.2.2...
Error: initializing source docker://jasonacox/weather41:0.2.2: reading manifest 0.2.2 in docker.io/jasonacox/weather41: requested access to the resource is denied
```

It worked in the past, so it may be an issue on my system only, but since I can pull `pypowerwall` image, I wonder if there may be some differences between the two repositories.

Output with DEBUG:
```
DEBU[0000] Looking for TLS certificates and private keys in /etc/docker/certs.d/docker.io 
DEBU[0000] GET https://registry-1.docker.io/v2/         
DEBU[0000] Ping https://registry-1.docker.io/v2/ status 401 
DEBU[0000] GET https://auth.docker.io/token?scope=repository%3Ajasonacox%2Fweather41%3Apull&service=registry.docker.io 
DEBU[0001] GET https://registry-1.docker.io/v2/jasonacox/weather41/manifests/0.2.2 
DEBU[0002] Detected insufficient_scope error, will retry request with updated scope 
DEBU[0002] GET https://auth.docker.io/token?scope=repository%3Ajasonacox%2Fweather41%3Apull&scope=repository%3Ajasonacox%2Fweather41%3Apull&service=registry.docker.io 
DEBU[0002] GET https://registry-1.docker.io/v2/jasonacox/weather41/manifests/0.2.2 
DEBU[0002] Content-Type from manifest GET is "application/json" 
DEBU[0002] Discarding non-primary errors:               
DEBU[0002]   unauthorized: authentication required      
DEBU[0002] Accessing "docker.io/jasonacox/weather41:0.2.2" failed: reading manifest 0.2.2 in docker.io/jasonacox/weather41: requested access to the resource is denied 
DEBU[0002] Error pulling candidate docker.io/jasonacox/weather41:0.2.2: initializing source docker://jasonacox/weather41:0.2.2: reading manifest 0.2.2 in docker.io/jasonacox/weather41: requested access to the resource is denied 
Error: initializing source docker://jasonacox/weather41:0.2.2: reading manifest 0.2.2 in docker.io/jasonacox/weather41: requested access to the resource is denied
DEBU[0002] Shutting down engines                        
```

compared to output for `pypowerwall`:
```
DEBU[0000] Looking for TLS certificates and private keys in /etc/docker/certs.d/docker.io
DEBU[0000] GET https://registry-1.docker.io/v2/
DEBU[0000] Ping https://registry-1.docker.io/v2/ status 401
DEBU[0000] GET https://auth.docker.io/token?scope=repository%3Ajasonacox%2Fpypowerwall%3Apull&service=registry.docker.io
DEBU[0001] GET https://registry-1.docker.io/v2/jasonacox/pypowerwall/manifests/0.6.2t28
DEBU[0002] Content-Type from manifest GET is "application/vnd.docker.distribution.manifest.list.v2+json"
DEBU[0002] Using blob info cache at /home/roberto/.local/share/containers/cache/blob-info-cache-v1.boltdb
DEBU[0002] Source is a manifest list; copying (only) instance sha256:a10b8ccdad85db4752c2015868bd1c53a929f90bff9e51685a366352c5bae365 for current system
DEBU[0002] GET https://registry-1.docker.io/v2/jasonacox/pypowerwall/manifests/sha256:a10b8ccdad85db4752c2015868bd1c53a929f90bff9e51685a366352c5bae365
DEBU[0002] Content-Type from manifest GET is "application/vnd.docker.distribution.manifest.v2+json"
DEBU[0002] IsRunningImageAllowed for image docker:docker.io/jasonacox/pypowerwall:0.6.2t28
DEBU[0002]  Using default policy section
DEBU[0002]  Requirement 0: allowed
DEBU[0002] Overall: allowed
DEBU[0002] Downloading /v2/jasonacox/pypowerwall/blobs/sha256:998ac30eeb79245eb56f5bfa0db20349ecfe06c37b2f11e53f57250bc8a72842
```

Has anyone experienced the same issue?

mcbirse, 2023-11-04:
Hi @rcasta74 - could be a very simple fix...

Is the container name correct in your command, as it appears to be missing a "1" and should be `weather411`

Try this:

```bash
podman pull docker.io/jasonacox/weather411:0.2.2
```

rcasta74, 2023-11-04:
omg, what a stupid typo.
Thanks for pointing this out

closed 2023-11-04 by rcasta74