jasonacox/Powerwall-Dashboard #387: I have a Powerwall 3 and willing to test

jesaf00, 2023-11-18:
**Problem**
Powerwall 3 not compaticle with Powerwall-Dashboard

**Enhancement**
I would like to help make it work if I can

**Additional context**
I have 3x Powerwall 3s just installed with a tesla backup switch and would like to be able to use this tool.

jesaf00, 2023-11-18:
[image]
[image]

jasonacox, 2023-11-19:
Hi @jesaf00 -  Congrats on getting the new Powerwall!  

Do you know if it is on your local network? If so, can you log in to your Powerwall web portal?  Tesla has instructions ([link](https://www.tesla.com/support/energy/powerwall/own/connecting-network)) but I wonder if it works for your setup. The pypowerwall library uses the APIs on that web portal for Powerwall/2/+.

Can you also provide the model number / picture of the Powerwall?

jesaf00, 2023-11-19:
I can send the specific model info from the side tomorrow. But here are some pics. Sent from my iPhone

jesaf00, 2023-11-19:
Sorry, forgot to mention that yes it is on my local network. 2 of 3 are showing on my WiFi but only one switched to the correct network after I changed using the app. I am unable to log into the PW using the instructions for previous PWs. I get a 404 right away or after ignoring the self signed cert. Sent from my iPhone

jasonacox, 2023-11-19:
The key will be figuring out how to log in to the web portal as that is essentially what pypowerwall is doing.

You can try some of the URLs (replace the IP with the address of your Powerwall):

https://10.1.2.3/summary?mode=kiosk
https://10.1.2.3/api/site_info

jesaf00, 2023-11-19:
HI, those urls with my IP produce a 404 after bypassing the certificate. Attached are the photos and the part number is 1707000-25-G.
[image]
[image]
[image]
[image]

https://github.com/jasonacox/Powerwall-Dashboard/assets/136191207/918e5dea-0314-408d-9728-41b861c3b60e

jasonacox, 2023-11-19:
Thanks @jesaf00 !  The new Powerwalls look nice.  Thanks for the video.  Just to confirm, you also tried the landing page https://192.168.200.51/ by itself?

I'm worried. If you are able to connect to their local WiFi but still unable to get to any web portal, that could indicate that Tesla is removing the local access (and APIs) and switching to Cloud only control and monitoring similar to the solar-only configurations. The good news is that @mcbirse 's tesla-history code will likely be able to get that data, but the bad news is that it won't be at the same fidelity as local API provides and will be limited (local API for PW/2/+ can provide string, voltage, frequency, temp data). 

Since you are likely one of the first to get these, the local access may still be under development. Also, since you are getting a 404, we know it does still have a webserver running. They may not have a customer page, but there could still be APIs. You might try this:

https://192.168.200.51/api/login/Basic

There definitely isn't much on the internet about PW3 other than "they are coming in 2024."  I'll reach out to see if anyone else has discovered the APIs.

If you still have the contact with the installer, can you ask if there is a local web portal (or APIs) that customers can access/use?  Feel free to direct them to this dashboard project if they wonder why. :)

jesaf00, 2023-11-19:
Hi Jason, yes I tried just the IP and https://192.168.200.51/api/login/Basic also gives 404, arg! I am having a couple of issues with these also so hopefully I can just get the working correctly. I am running a port scanner to see what other ports might be open besides 443 and will let you know. 

https://teslamotorsclub.com/tmc/posts/7917791/

jasonacox, 2023-11-19:
Thanks for that link!

Good idea on the port scanner. Again, the 404 does provide hope that there is an underlying API (a new one) that we just don't know yet. However, without a UI, it will be hard to determine what they are unless Tesla is willing to provide it to us. Perhaps the installer application makes calls to this and we could pick it up by traffic sniffing.

jasonacox, 2023-11-19:
@jesaf00 - Great suggestion from @vls29 at https://github.com/vloschiavo/powerwall2/discussions/99#discussioncomment-7613279: If you can, try to run a verbose curl against the Powerwall 3. Even if it is a 404, there might be  payload or header data that could be helpful.

```bash
curl -v -k  https://192.168.200.51
```

jesaf00, 2023-11-19:
Here is the output

`Last login: Sun Nov 19 10:28:14 on ttys001
jerry@MBP14M3aceblack ~ % curl -v -k  https://192.168.200.51
*   Trying 192.168.200.51:443...
* Connected to 192.168.200.51 (192.168.200.51) port 443 (#0)
* ALPN: offers h2,http/1.1
* (304) (OUT), TLS handshake, Client hello (1):
* (304) (IN), TLS handshake, Server hello (2):
* (304) (IN), TLS handshake, Unknown (8):
* (304) (IN), TLS handshake, Certificate (11):
* (304) (IN), TLS handshake, CERT verify (15):
* (304) (IN), TLS handshake, Finished (20):
* (304) (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / AEAD-AES128-GCM-SHA256
* ALPN: server accepted h2
* Server certificate:
*  subject: C=US; ST=California; L=Palo Alto; O=Tesla; OU=Tesla Energy Products; CN=d9528c5155a2621cf41a465e671195cc
*  start date: Jul 14 12:33:48 2023 GMT
*  expire date: Jul  7 12:33:45 2048 GMT
*  issuer: C=US; ST=California; L=Palo Alto; O=Tesla; OU=Tesla Energy Products; CN=d9528c5155a2621cf41a465e671195cc CA
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
* using HTTP/2
* h2 [:method: GET]
* h2 [:scheme: https]
* h2 [:authority: 192.168.200.51]
* h2 [:path: /]
* h2 [user-agent: curl/8.1.2]
* h2 [accept: */*]
* Using Stream ID: 1 (easy handle 0x12900e200)
> GET / HTTP/2
> Host: 192.168.200.51
> User-Agent: curl/8.1.2
> Accept: */*
> 
< HTTP/2 404 
< content-type: text/plain; charset=utf-8
< x-content-type-options: nosniff
< content-length: 19
< date: Sun, 19 Nov 2023 21:09:20 GMT
< 
404 page not found
* Connection #0 to host 192.168.200.51 left intact
jerry@MBP14M3aceblack ~ % 
`

jasonacox, 2023-11-19:
Thanks Jerry. Not much here, sadly.  For completeness, it would be good to try some POSTs as well:

```bash
# Powerwall 2 Login Endpoint
curl -v -k -s -i -X POST -H "Content-Type: application/json" -d '{"username":"example@example.com","password":"example","force_sm_off":false}' https://192.168.200.51/api/login/Basic

# Root Endpoint
curl -v -k -s -i -X POST -H "Content-Type: application/json" -d '{"username":"example@example.com","password":"example","force_sm_off":false}' https://192.168.200.51/
```

jesaf00, 2023-11-20:
Not sure this gave anything different. Is the password my Tesla.com password?

` https://192.168.200.51/api/login/Basic
*   Trying 192.168.200.51:443...
* Connected to 192.168.200.51 (192.168.200.51) port 443 (#0)
* ALPN: offers h2,http/1.1
* (304) (OUT), TLS handshake, Client hello (1):
* (304) (IN), TLS handshake, Server hello (2):
* (304) (IN), TLS handshake, Unknown (8):
* (304) (IN), TLS handshake, Certificate (11):
* (304) (IN), TLS handshake, CERT verify (15):
* (304) (IN), TLS handshake, Finished (20):
* (304) (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / AEAD-AES128-GCM-SHA256
* ALPN: server accepted h2
* Server certificate:
*  subject: C=US; ST=California; L=Palo Alto; O=Tesla; OU=Tesla Energy Products; CN=d9528c5155a2621cf41a465e671195cc
*  start date: Jul 14 12:33:48 2023 GMT
*  expire date: Jul  7 12:33:45 2048 GMT
*  issuer: C=US; ST=California; L=Palo Alto; O=Tesla; OU=Tesla Energy Products; CN=d9528c5155a2621cf41a465e671195cc CA
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
* using HTTP/2
* h2 [:method: POST]
* h2 [:scheme: https]
* h2 [:authority: 192.168.200.51]
* h2 [:path: /api/login/Basic]
* h2 [user-agent: curl/8.1.2]
* h2 [accept: */*]
* h2 [content-type: application/json]
* h2 [content-length: 101]
* Using Stream ID: 1 (easy handle 0x13a00da00)
> POST /api/login/Basic HTTP/2
> Host: 192.168.200.51
> User-Agent: curl/8.1.2
> Accept: */*
> Content-Type: application/json
> Content-Length: 101
> 
* We are completely uploaded and fine
< HTTP/2 404 
HTTP/2 404 
< content-type: text/plain; charset=utf-8
content-type: text/plain; charset=utf-8
< x-content-type-options: nosniff
x-content-type-options: nosniff
< content-length: 19
content-length: 19
< date: Mon, 20 Nov 2023 01:58:28 GMT
date: Mon, 20 Nov 2023 01:58:28 GMT

< 
404 page not found
* Connection #0 to host 192.168.200.51 left intact
`

`https://192.168.200.51               
*   Trying 192.168.200.51:443...
* Connected to 192.168.200.51 (192.168.200.51) port 443 (#0)
* ALPN: offers h2,http/1.1
* (304) (OUT), TLS handshake, Client hello (1):
* (304) (IN), TLS handshake, Server hello (2):
* (304) (IN), TLS handshake, Unknown (8):
* (304) (IN), TLS handshake, Certificate (11):
* (304) (IN), TLS handshake, CERT verify (15):
* (304) (IN), TLS handshake, Finished (20):
* (304) (OUT), TLS handshake, Finished (20):
* SSL connection using TLSv1.3 / AEAD-AES128-GCM-SHA256
* ALPN: server accepted h2
* Server certificate:
*  subject: C=US; ST=California; L=Palo Alto; O=Tesla; OU=Tesla Energy Products; CN=d9528c5155a2621cf41a465e671195cc
*  start date: Jul 14 12:33:48 2023 GMT
*  expire date: Jul  7 12:33:45 2048 GMT
*  issuer: C=US; ST=California; L=Palo Alto; O=Tesla; OU=Tesla Energy Products; CN=d9528c5155a2621cf41a465e671195cc CA
*  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
* using HTTP/2
* h2 [:method: POST]
* h2 [:scheme: https]
* h2 [:authority: 192.168.200.51]
* h2 [:path: /]
* h2 [user-agent: curl/8.1.2]
* h2 [accept: */*]
* h2 [content-type: application/json]
* h2 [content-length: 101]
* Using Stream ID: 1 (easy handle 0x15a80d000)
> POST / HTTP/2
> Host: 192.168.200.51
> User-Agent: curl/8.1.2
> Accept: */*
> Content-Type: application/json
> Content-Length: 101
> 
* We are completely uploaded and fine
< HTTP/2 404 
HTTP/2 404 
< content-type: text/plain; charset=utf-8
content-type: text/plain; charset=utf-8
< x-content-type-options: nosniff
x-content-type-options: nosniff
< content-length: 19
content-length: 19
< date: Mon, 20 Nov 2023 02:00:01 GMT
date: Mon, 20 Nov 2023 02:00:01 GMT

< 
404 page not found
* Connection #0 to host 192.168.200.51 left intact`

jasonacox, 2023-11-20:
Thanks again! Unfortunately, it didn't give us much more.  Also, the user/password doesn't matter. If it was working it would have given us something like an access denied message instead of a 404. 

How did the installer configure the Powerwalls?  Was it the [Tesla Pro app](https://apps.apple.com/us/app/tesla-pros/id1625770308)?

Are you able to connect to each Powerwall separately? In my Powerwall+ setup, the backup gateway is a separate box and has its own WiFi access point which is where the web portal works. I wonder if there are other access points. If so, it would be good to run the `curl` commands against those.

longzheng, 2023-11-20:
It looks like Powerwall 3 can be [designed without a Backup Gateway 2](https://service.tesla.com/docs/Public/Energy/Powerwall/Powerwall-3-Owners-Manual-NA-EN/GUID-49E1A269-2EBC-4C2C-8AA1-44A08704065C.html), I'm guessing that's what OP has.

[image]

Because the dashboard/portal exists on the gateway at the moment, I'm guessing there's no dashboard/portal when connecting directly to a Powerwall 3?

zigam, 2023-11-20:
This was the case for PW2 as well, with a backup switch (i.e. no gateway) one of the Powerwalls is designated a "site controller" and acts as the gateway, with the same web and API interface.

However, the access seems to have changed with PW3.  Based on reports here and from [TMC forums](https://teslamotorsclub.com/tmc/threads/discussion-powerwall-3-speculation-discussion-etc.299492/page-24#post-7918474):
1. No access to web or /api using the local network.
2. No access to web using the internal wifi network (now named `TeslaPW_xxx`).
3. Access is possible using the Tesla Pros app, which uses the internal wifi network.  Tesla Pros seems to be using a new binary `/tedapi` endpoint which I posted about [here](https://github.com/jrester/tesla_powerwall/issues/20#issuecomment-1810848383).

jasonacox, 2023-11-20:
This is great @zigam !  It is possible the Powerwall 3 has this same binary to be compatible with the Tesla Pros app.

@jesaf00 could you try these `curl` commands to see if you get anything on the PW3?

```bash
# Log in to the TeslaPW_xxx wifi 

# First just check to see if the API is responsive
curl -v -k https://192.168.91.1/tedapi/din
curl -v -k https://192.168.91.1/tedapi/v1

# Store the gateway password
export GW_PWD="<FULL_GATEWAY_PWD>"

# Get the device din, e.g. 1232100-10-E--CN321329G1E123.
curl -k -u Tesla_Energy_Device:$GW_PWD https://192.168.91.1/tedapi/din
```

Next, if that works, download this and rename to request.bin - edit it to have your DIN.
[request.bin.txt](https://github.com/jrester/tesla_powerwall/files/13353587/request.bin.txt)

```bash
# Request the configuration (binary file - protobuf payload?)
curl -k -H 'Content-type: application/octet-string' -u Tesla_Energy_Device:$GW_PWD --data-binary @request.bin https://192.168.91.1/tedapi/v1
```

pbburkhalter, 2023-11-20:
@jasonacox  here's the output from my recently installed PW3's

```
curl -v -k https://192.168.1.73/tedapi/din
*   Trying 192.168.1.73:443...
* Connected to 192.168.1.73 (192.168.1.73) port 443
* schannel: disabled automatic use of client certificate
* schannel: using IP address, SNI is not supported by OS.
* ALPN: curl offers http/1.1
* ALPN: server accepted http/1.1
* using HTTP/1.1
> GET /tedapi/din HTTP/1.1
> Host: 192.168.1.73
> User-Agent: curl/8.4.0
> Accept: */*
>
* schannel: remote party requests renegotiation
* schannel: renegotiating SSL/TLS connection
* schannel: SSL/TLS connection renegotiated
< HTTP/1.1 403 Forbidden
< Content-Type: application/json
< X-Content-Type-Options: nosniff
< Date: Mon, 20 Nov 2023 18:27:08 GMT
< Content-Length: 102
<
{"code":403,"error":"Unable to GET to resource","message":"User does not have adequate access rights"}* Connection #0 to host 192.168.1.73 left intact
```

curl -v -k https://192.168.1.73/tedapi/v1
*   Trying 192.168.1.73:443...
* Connected to 192.168.1.73 (192.168.1.73) port 443
* schannel: disabled automatic use of client certificate
* schannel: using IP address, SNI is not supported by OS.
* ALPN: curl offers http/1.1
* ALPN: server accepted http/1.1
* using HTTP/1.1
> GET /tedapi/v1 HTTP/1.1
> Host: 192.168.1.73
> User-Agent: curl/8.4.0
> Accept: */*
>
* schannel: remote party requests renegotiation
* schannel: renegotiating SSL/TLS connection
* schannel: SSL/TLS connection renegotiated
< HTTP/1.1 405 Method Not Allowed
< Content-Type: text/plain; charset=utf-8
< X-Content-Type-Options: nosniff
< Date: Mon, 20 Nov 2023 18:29:00 GMT
< Content-Length: 19
<
Method Not Allowed
* Connection #0 to host 192.168.1.73 left intact

zigam, 2023-11-20:
@pbburkhalter you'll have to run these commands while on the `TeslaPW_xxx` wifi network.  The ip address on that network will likely be https://192.168.91.1

pbburkhalter, 2023-11-20:
Oops. I did just connect to that and got the same results.

zigam, 2023-11-20:
Those responses are actually expected because auth was not provided.  This is the one that will hopefully work:

```
# Full gateway password should be on the sticker inside the PW3.
export GW_PWD="<FULL_GATEWAY_PWD>"

# Get the device din, e.g. 1232100-10-E--CN321329G1E123.
curl -k -u Tesla_Energy_Device:$GW_PWD https://192.168.91.1/tedapi/din
```

pbburkhalter, 2023-11-20:
Yes, that does work.

zigam, 2023-11-20:
Great!  The next step would be an actual RPC call.  Goes without saying that this is uncharted territory, so proceed at your own risk :)

Download [request.bin.txt](https://github.com/jrester/tesla_powerwall/files/13353587/request.bin.txt), rename it to `request.bin`, and edit the file by replacing the din with your own din from the command above.  You have to do this carefully, the din in the file ends with 123 so the `z` character remains in the file.  Assuming your din is the same length as the one currently in the file (`1232100-10-E--CN321329G1E123`), the file should remain at 63 bytes.  Then:

```
# Request the configuration.
curl -k -H 'Content-type: application/octet-string' -u Tesla_Energy_Device:$GW_PWD --data-binary @request.bin https://192.168.91.1/tedapi/v1
```

If this works, you should get a response with the entire configuration of your system. 🤞

pbburkhalter, 2023-11-20:
Doesn't work from a command line or PowerShell but I'm open to ideas.

zigam, 2023-11-20:
@pbburkhalter what's the error?  I think the other way to investigate the protocol would be to install a debugging proxy on the phone and capture Tesla Pros traffic.  This is a bit more involved, but if you want to spend the time (and a bit of money, the proxy app costs $9) I'd be happy to help.  ziga.mahkovec AT gmail.com.

pbburkhalter, 2023-11-20:
Sorry I should have clarified I'm not using the Tesla pros app... I am just running it on my computer connected to the PowerWall wi-fi network

zigam, 2023-11-20:
@pbburkhalter right but these curl requests are replaying what the Tesla Pros app does.  So if they don't work for you, you'd have to inspect the requests that the app generates so we can learn how to reproduce them.

The error you got from curl would also be helpful.

pbburkhalter, 2023-11-20:
curl: (6) Could not resolve host: application

zigam, 2023-11-20:
> curl: (6) Could not resolve host: application

Ah, that looks like a shell issue with the single quotes.  Try this:
```
curl -k -H "Content-Type: application/octet-string" -u Tesla_Energy_Device:$GW_PWD --data-binary @request.bin https://192.168.91.1/tedapi/v1
```

jasonacox, 2023-11-20:
I updated [pypowerwall](https://github.com/jasonacox/pypowerwall) (v0.6.3) to detect Powerwall 3 devices on your local network. I haven't added any additional support for the /tedapi  APIs but this may help if you are looking for Powerwall 3 gateways on your local network.

```bash
# Install python package - or upgrade
pip3 install --upgrade pypowerwall

# Scan
python3 -m pypowerwall scan
```

pbburkhalter, 2023-11-20:
curl: (28) Failed to connect to 192.168.91.1 port 443 after 21035 ms: Couldn't connect to server

zigam, 2023-11-20:
> curl: (28) Failed to connect to 192.168.91.1 port 443 after 21035 ms: Couldn't connect to server

Just to confirm, this was on the PW3 wifi and the previous /din command still works?
```
curl -k -u Tesla_Energy_Device:$GW_PWD https://192.168.91.1/tedapi/din
```

pbburkhalter, 2023-11-21:
That is correct

zigam, 2023-11-21:
Thanks for testing it!  I'm curious if this works for others, regardless of whether they're running PW3 or PW2+.

stevecastaneda, 2023-11-21:
> I updated [pypowerwall](https://github.com/jasonacox/pypowerwall) (v0.6.3) to detect Powerwall 3 devices on your local network

With my 4x PW3 install, I notice the scan will find the `child` powerwalls, but not the one designated by the installers as the "main".

jesaf00, 2023-11-21:
Sorry for the delay. The updated pypowerwall found 2 of the 3 PW3s but did not locate the second one at .204. I am having issues with connectivity so that may be related. 

`pyPowerwall Network Scanner [0.6.3]
Scan local network for Tesla Powerwall Gateways

    Your network appears to be: 192.168.200.0/24

    Enter Network or press enter to use 192.168.200.0/24: 

    Running Scan...
      Host: 192.168.200.1 ... OPEN - Not a Powerwall
      Host: 192.168.200.6 ... OPEN - Not a Powerwall
      Host: 192.168.200.10 ... OPEN - Not a Powerwall
      Host: 192.168.200.51 ... OPEN - Found Powerwall 3 [Currently Unsupported]
      Host: 192.168.200.145 ... OPEN - Not a Powerwall
      Host: 192.168.200.213 ... OPEN - Not a Powerwall
      Host: 192.168.200.238 ... OPEN - Not a Powerwall
      Host: 192.168.200.246 ... OPEN - Found Powerwall 3 [Currently Unsupported]
      Done                           

Discovered 2 Powerwall Gateway
     192.168.200.51 [Powerwall-3] Firmware Currently Unsupported - See https://tinyurl.com/pw3support
     192.168.200.246 [Powerwall-3] Firmware Currently Unsupported - See https://tinyurl.com/pw3support`

jasonacox, 2023-11-21:
Thanks @jesaf00 and @stevecastaneda !

> With my 4x PW3 install, I notice the scan will find the 3 x child powerwalls, but not the last one designated by the installers as the main.

@stevecastaneda on the `main` PW3 that it didn't find, what data do you get back when you run this against its endpoint?  If you can, use it's local IP address (if you can figure it out) as well as connecting to its WiFI.

```
curl -v -k -u Tesla_Energy_Device:GW_PWD https://192.168.91.1/tedapi/din
```

jasonacox, 2023-11-21:
@zigam Testing this on my PW+ system using the script and modifying the request.bin with my DIN.

```bash
export GW_USER="Tesla_Energy_Device"
export GW_PWD="XXXXXXXXXX"
export GW_IP="192.168.91.1"

echo "-- DIN Payload --"
curl -v -k -u "$GW_USER:$GW_PWD" https://$GW_IP/tedapi/din

echo "-- Config Payload --"
curl -v -k -H 'Content-type: application/octet-string' -u "$GW_USER:$GW_PWD" --data-binary @request.bin https://$GW_IP/tedapi/v1
```
Connecting to TEG-xxx wifi:
* ✅  GET /tedapi/din - Works and returns HTTP/2 200 and the DIN  
* ❌  POST /tedapi/v1 - Responds with HTTP/2 400 - I wonder if the request.bin file has other elements that need to be changed. 

Connected to Local Network:
* ❌  GET /tedapi/din - Responds with HTTP/2 403 and `{"code":403,"error":"Unable to GET to resource","message":"User does not have adequate access rights"}`
* ❌  POST /tedapi/v1 - Responds with HTTP/2 403 and `{"code":403,"error":"Unable to GET to resource","message":"User does not have adequate access rights"}`

jesaf00, 2023-11-21:
According to Rockyproc at https://teslamotorsclub.com/tmc/posts/7919661/ , he was able to get more info from Home Assistant + HACS.

stevecastaneda, 2023-11-21:
> @stevecastaneda on the `main` PW3 that it didn't find, what data do you get back when you run this against its endpoint? If you can, use it's local IP address (if you can figure it out) as well as [...]

- Connected to Local Network: `{"code":403,"error":"Unable to GET to resource","message":"User does not have adequate access rights"}`
- Connected to PW Network: `{"code":403,"error":"Unable to GET to resource","message":"User does not have adequate access rights"}`

jasonacox, 2023-11-21:
> According to Rockyproc at https://teslamotorsclub.com/tmc/posts/7919661/ , he was able to get more info from Home Assistant + HACS.

Thanks @jesaf00 !  However, unless I'm reading it wrong, The Home Assistant "Tesla Custom Integration" that the post mentions, uses the Tesla Cloud. There is another one for powerwalls that uses https://github.com/jrester/tesla_powerwall. That library uses a similar approach to pypowerwall but would be interesting to verify.

The Tesla Custom Integration is similar to what the [tesla-history](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/tesla-history) and [solar-only](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/solar-only) tools do. You can get some of the main data that way but it is delayed and doesn't include some things like voltage, frequencies, temperatures, alerts, capacity-degredation, string data, etc.

> Connected to Local Network: {"code":403,"error":"Unable to GET to resource","message":"User does not have adequate access rights"}

Thanks @stevecastaneda ! The pypowerwall scan is looking for this exact string to find Powerwall 3 instances on your local network.  Can you run it again to see if it just happened to miss it the first time? I may need to expand the timeout.

stevecastaneda, 2023-11-21:
> Thanks @stevecastaneda ! The pypowerwall scan is looking for this exact string to find Powerwall 3 instances on your local network. Can you run it again to see if it just happened to miss it the first time? I may need to expand the timeout.

You're right - a second scan found the one missing in the first, but then didn't find a different one in the 2nd scan.  +1 to adding time to the timeout.

jasonacox, 2023-11-21:
@stevecastaneda  - The default is 400ms (0.4s) and you can adjust it via command line. I can change the default if you can help me find the sweet spot. Of course, the issue is that the higher we go, the longer the scan takes. Try a few of these:

```bash
# timeout set to 500ms = 0.5s
python3 -m pypowerwall scan 0.5

# timeout set to 1s
python3 -m pypowerwall scan 1

jesaf00, 2023-11-21:
.5 didn't get all three and neither did .1, however when I query that one specifically it worked. 

`pyPowerwall Network Scanner [0.6.3]
Scan local network for Tesla Powerwall Gateways

    Your network appears to be: 192.168.200.0/24

    Enter Network or press enter to use 192.168.200.0/24: 

    Running Scan...
      Host: 192.168.200.1 ... OPEN - Not a Powerwall
      Host: 192.168.200.6 ... OPEN - Not a Powerwall
      Host: 192.168.200.10 ... OPEN - Not a Powerwall
      Host: 192.168.200.51 ... OPEN - Found Powerwall 3 [Currently Unsupported]
      Host: 192.168.200.145 ... OPEN - Not a Powerwall
      Host: 192.168.200.213 ... OPEN - Not a Powerwall
      Host: 192.168.200.238 ... OPEN - Not a Powerwall
      Done                           

Discovered 1 Powerwall Gateway
     192.168.200.51 [Powerwall-3] Firmware Currently Unsupported - See https://tinyurl.com/pw3support
 
jerry@AOBM Powerwall-Dashboard % python3 -m pypowerwall scan 1  
/Users/jerry/Library/Python/3.9/lib/python/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(

pyPowerwall Network Scanner [0.6.3]
Scan local network for Tesla Powerwall Gateways

    Your network appears to be: 192.168.200.0/24

    Enter Network or press enter to use 192.168.200.0/24: 

    Running Scan...
      Host: 192.168.200.1 ... OPEN - Not a Powerwall
      Host: 192.168.200.6 ... OPEN - Not a Powerwall
      Host: 192.168.200.10 ... OPEN - Not a Powerwall
      Host: 192.168.200.51 ... OPEN - Found Powerwall 3 [Currently Unsupported]
      Host: 192.168.200.145 ... OPEN - Not a Powerwall
      Host: 192.168.200.204 ... OPEN - Found Powerwall 3 [Currently Unsupported]
      Host: 192.168.200.213 ... OPEN - Not a Powerwall
      Host: 192.168.200.238 ... OPEN - Not a Powerwall
      Done                           

Discovered 2 Powerwall Gateway
     192.168.200.51 [Powerwall-3] Firmware Currently Unsupported - See https://tinyurl.com/pw3support
     192.168.200.204 [Powerwall-3] Firmware Currently Unsupported - See https://tinyurl.com/pw3support
 
jerry@AOBM Powerwall-Dashboard % python3 -m pypowerwall scan 1
/Users/jerry/Library/Python/3.9/lib/python/site-packages/urllib3/__init__.py:34: NotOpenSSLWarning: urllib3 v2 only supports OpenSSL 1.1.1+, currently the 'ssl' module is compiled with 'LibreSSL 2.8.3'. See: https://github.com/urllib3/urllib3/issues/3020
  warnings.warn(

pyPowerwall Network Scanner [0.6.3]
Scan local network for Tesla Powerwall Gateways

    Your network appears to be: 192.168.200.0/24

    Enter Network or press enter to use 192.168.200.0/24: 192.168.200.53/32

    Running Scan...
      Host: 192.168.200.53 ... OPEN - Found Powerwall 3 [Currently Unsupported]
      Done                           

Discovered 1 Powerwall Gateway
     192.168.200.53 [Powerwall-3] Firmware Currently Unsupported - See https://tinyurl.com/pw3support
 
`

stevecastaneda, 2023-11-21:
> python3 -m pypowerwall scan 0.5

- 400ms: 4 found successfully (4 successful out of 4 attempts)
- 500ms: 4 found successfully (3 successful out of 3 attempts)

Odd, this time it worked fine.  I would say that 500ms felt quick so it may not be a bad idea just to buffer by another 100ms.

jasonacox, 2023-11-21:
Thanks @jesaf00 ! Feel free to try something even higher... like 2, 5 or even 10. That would likely mean your connection (WiFi) isn't very strong or is dropping packets, but may be worth trying.

jasonacox, 2023-11-21:
@mcbirse - I wonder if the solar-only daemon updates for tesla-history would automatically support PW3? They would use the tesla-solar option but use the standard dashboard.json (but remove the animation and other panels that won't populate). I believe al the Powerwall data is also pulled in via tesla-history (unless I'm reading it wrong) so PW3 users could at least start getting dashboard data.

jasonacox, 2023-11-22:
I added a research discussion thread for /tedapi support (for PW2/+ and PW3): https://github.com/jasonacox/Powerwall-Dashboard/discussions/392

mcbirse, 2023-11-22:
> @mcbirse - I wonder if the solar-only daemon updates for tesla-history would automatically support PW3? They would use the tesla-solar option but use the standard dashboard.json (but remove the [...]

Correct @jasonacox - anyone with a Powerwall system could install the dashboard using the "solar-only" option, and then use the standard `dashboard.json` - pretty much how I've been testing the whole solar-only offshoot.

All of the Powerwall power metrics should be pulled from the cloud, including backup events so grid status data is populated, and of course the analysis data will populate too (power flow animation and extended data, e.g. voltages, alerts, etc. will not however).

- NOTE: run tesla-history using a --start and --end period manually to get the historical data to build analysis data history.
- NOTE2: a "solar-only" install has the advantage this could be easily run via docker, e.g. `docker exec -it tesla-history python3 tesla-history.py --start "start-date-and-time" --end "end-date-and-time"` without even needing to install required python modules).

Can a PW3 user test installing the Powerwall Dashboard in solar-mode mode and see if it works?

Seems timely the solar-only offshoot has now been merged as a setup option just this week! 🚀

jasonacox, 2023-11-22:
Thanks @mcbirse !  I was thinking the same thing (timely release of the solar-only setup that can work for Powerwall 3 owners)! If this works, we should update the script and README to include language that this supports Powerwall 3 (Tesla Cloud version).

@jesaf00 - I know you already installed the Powerwall-Dashboard. Would you be willing to test this?

## Powerwall 3 Dashboard Setup

```bash
# Install Powerwall-Dashboard (if not already installed)
git clone https://github.com/jasonacox/Powerwall-Dashboard.git
cd Powerwall-Dashboard

# If already installed, remove old configuration
rm compose.env

# Run Setup
./setup.sh
```
You will see this:

```
  Powerwall Dashboard (v3.0.0) - SETUP
  -----------------------------------------
  Select configuration profile:

  1 - default     (Powerwall w/ Gateway on LAN)
  2 - solar-only  (No Gateway - data retrieved from Tesla Cloud)
```

For Powerwall 3 owners, select profile "2 - solar-only  (No Gateway - data retrieved from Tesla Cloud)"

When you install the dashboard configuration file, you can use the default (`dashboard.json`). It will have some panels that don't work but after a while you should start seeing data populate.

## Powerwall 3 - Load History

You an load the historic data from the Tesla Cloud into your dashboard as @mcbirse mentions, with something like this:

```bash
docker exec -it tesla-history python3 tesla-history.py --start "2023-01-01 00:00:00" --end "2024-01-01 00:00:00"
```

jesaf00, 2023-11-22:
Here are my results...

`Tesla Account Setup
-------------------
Email address: myemail@gmail.com

Config saved to '/var/lib/tesla-history/tesla-history.conf'

----------------------------------------
Tesla account: myemail@gmail.com
----------------------------------------
Open the below address in your browser to login.

https://auth.tesla.com/oauth2/v3/authorize?response_type=code&client_id=ownerapi&redirect_uri=https%3A%2F%2Fauth.tesla.com%2Fvoid%2Fcallback&scope=openid+email+offline_access&state=d2mvl2SQYIBBdKerwpkUfdfnAzwGqT&code_challenge=lpsbKVScxeDMCMi3d4DKMQXQgoJUGiFuBH1ISmFU7gw&code_challenge_method=S256&login_hint=myemail%40gmail.com

After login, paste the URL of the 'Page Not Found' webpage below.

Enter URL after login: https://auth.tesla.com/void/callback?code=NA_8091ab256d05f3d2f96b8e970f3ee04be42b6dbc32b5038955172702ab91&state=d2mvl2SQYIBBdKerwpkUfdfnAzwGqT&issuer=https%3A%2F%2Fauth.tesla.com%2Foauth2%2Fv3
----------------------------------------
ERROR: Failed to retrieve PRODUCT_LIST - RetryError(MaxRetryError("HTTPSConnectionPool(host='owner-api.teslamotors.com', port=443): Max retries exceeded with url: /api/1/products (Caused by ResponseError('too many 500 error responses'))"))
Restarting tesla-history...
tesla-history
------------------[ Final Setup Instructions ]-----------------

`

jasonacox, 2023-11-22:
Thanks @jesaf00 !  Are you using the same credentials you use for the Tesla App?

@mcbirse Have you seen this before?

jesaf00, 2023-11-22:
Well, IDK what happened. I deleted the container, reran the setup and it succeeded without asking for my PW. and here we are.... 
[image]

jasonacox, 2023-11-22:
Wow! 🚀  Nice job @jesaf00 - First to have a Powerwall 3 Dashboard up and running. ❤️ 

Did you also run the "Load History" steps? I assume that's how you have 24hr of data.

You can delete the "Power Flow" panel. I is based on pypowerwall proxying the Powerwall2/+ portal UI.  The `powerwall-no-animation.json` file replaces that spot with a Powerwall charge graph and piechart.

jesaf00, 2023-11-22:
Yes sir, I did! Thanks for the suggestion about the no animation dashboard. With this working what's next? Need to wait for some API info?

jasonacox, 2023-11-22:
Thanks @jesaf00 ! - Can you share another screenshot of your dashboard with the no-animation version?  

As to "what's next", I'm researching the Tesla Pros API. I have a first pass at the protobuf schema and if you are interested in testing, I have uploaded them here: https://github.com/jasonacox/pypowerwall/tree/main/tools/tedapi#tools 

The `test.py` script will prompt you for your Powerwall Gateway password (the one on the QR sticker) and will attempt to fetch the DIN (device ID number), current Configuration (warning: contains confidential user and password data), and the current status (not yet working 100%).  It will be interesting to see if the Powerwall 3 responds to any of these.

mcbirse, 2023-11-23:
@jesaf00 - this is brilliant, thanks for trying it! Glad to see PW3 owners at least have this option for now.

> ERROR: Failed to retrieve PRODUCT_LIST - RetryError(MaxRetryError("HTTPSConnectionPool(host='owner-api.teslamotors.com', port=443): Max retries exceeded with url: /api/1/products (Caused by ResponseError('too many 500 error responses'))"))

I suspect this was just a transient connection error to the Tesla host 'owner-api.teslamotors.com' for some reason... maybe the site went down or Internet dropped out briefly, and it happened to occur just after (successful) login when the script attempts to get the site details.

> Well, IDK what happened. I deleted the container, reran the setup and it succeeded without asking for my PW.

Yep, login was actually successful and the .conf and .auth files were saved. You would have just been able to re-run setup without removing the container and it would be okay... or even just run `docker restart tesla-history`. You can always run `docker logs tesla-history` to confirm the service is operating correctly as well. If it cannot login or get site details, there will be an error and the script will halt, and will require resolving by a restart or re-authenticating.

> If this works, we should update the script and README to include language that this supports Powerwall 3 (Tesla Cloud version).

@jasonacox - I agree, it makes sense that we now update the setup script and README to advise PW3 owners to use the "solar-only" option. We could even add a 3rd config profile option depending on what is clearer?

jasonacox, 2023-11-24:
@mcbirse - Good idea. But I'm also wondering if we just keep it limited to the two profiles for now while we figure out the final outcome. I'm getting some indication that Tesla won't be interested in providing local API access and pushing for everyone to move the Tesla App as the only location for this data. That could even impact us, as existing PW 2/+ owners (e.g. a firmware that removes the local portal and APIs).  I hope that doesn't happen, but it is worth exploring how we would support that transition and make it easy to switch.  At a minimum, the profile would allow everyone to continue to get basic information into the Dashboard.

Perhaps we just reword the `setup.py` profile selection a bit? We may also want to have the script look for compose.env, and if found, prompt the user if they want to change the configuration profile.

```
Select configuration profile:

  1 - Local Access (Powerwall 1, 2, or + using extended data from Tesla Gateway on LAN) - Default
  2 - Tesla Cloud  (Solar-only, Powerwall 1, 2, +, or 3 using data from Tesla Cloud)
```

If the 2nd option is selected (solar-only), the setup instructions will point users to use the solar-only or no-animation dashboards.

mcbirse, 2023-11-25:
@jasonacox - Sounds good and makes sense to me. I have started a new branch `v3.0.3` and committed some changes which include update to the profile selection descriptions, and allows changing the profile selection when already configured.

I have only done very minimal testing at this point, so if you see any issues feel free to update.

I also had some updates to the `pwstatus` and `weather-history` tools to make them a bit more resilient when errors occur, and have included these changes.

jesaf00, 2023-11-25:
Here is the screenshot you asked for. 

[image]

[43 more replies]