jasonacox/Powerwall-Dashboard #400: 3.0.3 upgrade errors

jgleigh, 2023-12-11:
compose.env: line 35: weather411,default: command not found

Looks like the setup.sh script is corrupting the compose.env file if there are already profiles set.

Original file: COMPOSE_PROFILES=default,weather411

Corrupted file after running upgrade: COMPOSE_PROFILES=default weather411,default

Running upgrade with no compose.env file works correctly.

jgleigh, 2023-12-11:
MacOS 14.1.2 in case there are weird difference between OSes.

jasonacox, 2023-12-12:
Thanks, @jgleigh  - I'll investigate. I thought I had tested an upgrade eon MacOS but likely forgot the weather411 path. @mcbirse If you have a chance, take a look as well in case I'm missing something simple.

mcbirse, 2023-12-12:
Hi @jasonacox - I'm not sure, but it seems it could be a compatibility issue with the separator (IFS) being set to a comma to process the bash array and update COMPOSE_PROFILES.

From what @jgleigh has shown, it looks like IFS does not get set to a comma so it remained as the default (space) when the compose.env file was updated. Must only affect some specific OS's or bash shells, as I tested many upgrade scenarios (but not on MacOS).

At a guess, the below might fix the issue. Have not tested yet but I'll try to do that tonight if I get time.

i.e. replace all occurrences of `IFS=,;` with `IFS=','` in all scripts - example of updated helper function below.

```bash
add_profile() {
    # Create default docker compose env file if needed.
    if [ ! -f ${COMPOSE_ENV_FILE} ]; then
        cp "${COMPOSE_ENV_FILE}.sample" "${COMPOSE_ENV_FILE}"
    fi
    if ! get_profile "${1}"; then
        # Add profile to COMPOSE_PROFILES and save to env file
        PROFILES+=("${1}")
        if [ -z "${COMPOSE_PROFILES}" ]; then
            if grep -q "^#COMPOSE_PROFILES=" "${COMPOSE_ENV_FILE}"; then
                sed -i.bak "s@^#COMPOSE_PROFILES=.*@COMPOSE_PROFILES=$(IFS=',' echo "${PROFILES[*]}")@g" "${COMPOSE_ENV_FILE}"
            else
                echo -e "\nCOMPOSE_PROFILES=$(IFS=',' echo "${PROFILES[*]}")" >> "${COMPOSE_ENV_FILE}"
            fi
        else
            sed -i.bak "s@^COMPOSE_PROFILES=.*@COMPOSE_PROFILES=$(IFS=',' echo "${PROFILES[*]}")@g" "${COMPOSE_ENV_FILE}"
        fi
    fi
}
```

jasonacox, 2023-12-13:
I can confirm that this impacts MacOS. Working on fix...

jasonacox, 2023-12-13:
I found the fix... simple, crazy... line 52:

from:
```
IFS=',' read -a PROFILES <<< ${COMPOSE_PROFILES}
```
to:
```
IFS=',' read -ra PROFILES <<< "${COMPOSE_PROFILES}"
```

I'll submit the PR for 3.0.4.

mcbirse, 2023-12-13:
Oh wow, great find and yes, crazy!

jasonacox, 2023-12-13:
It had me stumped...

 I eventually just pasted it into ChatGPT and it suggested quoting `${COMPOSE_PROFILES}`  (see [chat session](https://chat.openai.com/share/31aea5e4-c562-4a55-a74e-83d2fc1b38c1)). 

😁

mcbirse, 2023-12-13:
Amazing! ChatGPT and AI's in general have become seriously impressive, to the point of being a bit disturbing sometimes. 😄

jgleigh, 2023-12-13:
Fascinating stuff. Thanks for the quick fix.

closed 2023-12-13 by jgleigh