jasonacox/Powerwall-Dashboard #412: upgrade.sh cannot run as-is

cstout1, 2023-12-24:
I am unable to run the upgrade.sh script as is. Are there undocumented prerequisite steps for this script to succeed? Host is macOS 14. Attempt to upgrade produced the error below:
```
service@server Powerwall-Dashboard % sh upgrade.sh 
Upgrade Powerwall-Dashboard from 3.0.5 to 3.0.7
---------------------------------------------------------------------
This script will attempt to upgrade you to the latest version without
removing existing data. A backup is still recommended.

Upgrade - Proceed? [y/N] y

Resetting Timezone to Default...

Pull influxdb.sql, dashboard.json, telegraf.conf, and other changes...

fatal: not a git repository (or any of the parent directories): .git
```
The primary account is a standard (non-admin) user and is the current owner of all files in the working directory.

jasonacox, 2023-12-25:
> fatal: not a git repository (or any of the parent directories): .git

Did you install the files using `git clone` or some other method?  The upgrade script uses git to manage the files. 

Can you run `git status` to see what it shows?

cstout1, 2023-12-25:
Hi Jason, nothing was done to set this up outside of running the setup.sh file from a common directory. I threw it in /Users/Shared in hopes of a simpler install. Install and initial config was effortless with that script.

cstout1, 2023-12-25:
In a desperate attempt to resolve the error I made an empty temp file and tried to commit it but the issue persists. Here’s the output of the git status command. 
`service@server Powerwall-Dashboard % git status
On branch main
Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   .DS_Store
	modified:   grafana/grafana.db
	modified:   influxdb/data/_internal/_series/00/0000
	modified:   influxdb/data/_internal/_series/01/0000
	modified:   influxdb/data/_internal/_series/02/0000
	modified:   influxdb/data/_internal/_series/03/0000
	modified:   influxdb/data/_internal/_series/04/0000
	modified:   influxdb/data/_internal/_series/05/0000
	modified:   influxdb/data/_internal/_series/06/0000
	modified:   influxdb/data/_internal/_series/07/0000
	deleted:    influxdb/data/_internal/monitor/19/000000001-000000001.tsm
	deleted:    influxdb/data/_internal/monitor/19/000000002-000000001.tsm
	deleted:    influxdb/data/_internal/monitor/19/000000003-000000001.tsm
	deleted:    influxdb/data/_internal/monitor/19/000000004-000000001.tsm
	deleted:    influxdb/data/_internal/monitor/19/000000005-000000001.tsm
	deleted:    influxdb/data/powerwall/daily/10/000000033-000000003.tsm
	deleted:    influxdb/data/powerwall/daily/10/000000067-000000003.tsm
	deleted:    influxdb/data/powerwall/daily/10/000000100-000000003.tsm
	deleted:    influxdb/data/powerwall/daily/10/000000108-000000002.tsm
	deleted:    influxdb/data/powerwall/daily/10/000000116-000000002.tsm
	deleted:    influxdb/data/powerwall/monthly/11/000000033-000000003.tsm
	deleted:    influxdb/data/powerwall/monthly/11/000000067-000000003.tsm
	deleted:    influxdb/data/powerwall/monthly/11/000000100-000000003.tsm
	deleted:    influxdb/data/powerwall/monthly/11/000000108-000000002.tsm
	deleted:    influxdb/data/powerwall/monthly/11/000000116-000000002.tsm
	deleted:    influxdb/data/powerwall/raw/12/000000006-000000002.tsm
	deleted:    influxdb/data/powerwall/raw/12/fields.idx
	deleted:    influxdb/data/powerwall/raw/18/000000001-000000001.tsm
	deleted:    influxdb/data/powerwall/raw/18/000000002-000000001.tsm
	deleted:    influxdb/data/powerwall/raw/18/000000003-000000001.tsm
	deleted:    influxdb/data/powerwall/raw/18/000000004-000000001.tsm
	modified:   influxdb/meta/meta.db
	deleted:    influxdb/wal/_internal/monitor/19/_00021.wal
	deleted:    influxdb/wal/powerwall/alerts/9/_00001.wal
	deleted:    influxdb/wal/powerwall/autogen/3/_00001.wal
	deleted:    influxdb/wal/powerwall/daily/10/_00124.wal
	deleted:    influxdb/wal/powerwall/grid/7/_00001.wal
	deleted:    influxdb/wal/powerwall/kwh/4/_00001.wal
	deleted:    influxdb/wal/powerwall/monthly/11/_00124.wal
	deleted:    influxdb/wal/powerwall/pod/8/_00001.wal
	deleted:    influxdb/wal/powerwall/pwtemps/5/_00001.wal
	deleted:    influxdb/wal/powerwall/raw/18/_00009.wal
	deleted:    influxdb/wal/powerwall/vitals/6/_00001.wal

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	influxdb/data/_internal/monitor/19/000000007-000000002.tsm
	influxdb/data/_internal/monitor/30/
	influxdb/data/powerwall/alerts/29/
	influxdb/data/powerwall/alerts/9/000000001-000000001.tsm
	influxdb/data/powerwall/autogen/20/
	influxdb/data/powerwall/autogen/3/000000001-000000001.tsm
	influxdb/data/powerwall/daily/10/000000135-000000002.tsm
	influxdb/data/powerwall/daily/25/
	influxdb/data/powerwall/grid/27/
	influxdb/data/powerwall/grid/7/000000001-000000001.tsm
	influxdb/data/powerwall/kwh/24/
	influxdb/data/powerwall/kwh/4/000000001-000000001.tsm
	influxdb/data/powerwall/monthly/11/000000133-000000004.tsm
	influxdb/data/powerwall/monthly/11/000000141-000000002.tsm
	influxdb/data/powerwall/monthly/11/000000142-000000001.tsm
	influxdb/data/powerwall/monthly/11/000000143-000000001.tsm
	influxdb/data/powerwall/monthly/11/000000144-000000001.tsm
	influxdb/data/powerwall/monthly/11/000000145-000000001.tsm
	influxdb/data/powerwall/pod/28/
	influxdb/data/powerwall/pod/8/000000001-000000001.tsm
	influxdb/data/powerwall/pwtemps/22/
	influxdb/data/powerwall/pwtemps/5/000000001-000000001.tsm
	influxdb/data/powerwall/raw/18/000000006-000000002.tsm
	influxdb/data/powerwall/raw/21/
	influxdb/data/powerwall/vitals/23/
	influxdb/data/powerwall/vitals/6/000000001-000000001.tsm
	influxdb/wal/_internal/monitor/19/_00023.wal
	influxdb/wal/_internal/monitor/30/
	influxdb/wal/powerwall/alerts/29/
	influxdb/wal/powerwall/alerts/9/_00002.wal
	influxdb/wal/powerwall/autogen/20/
	influxdb/wal/powerwall/autogen/3/_00002.wal
	influxdb/wal/powerwall/daily/10/_00136.wal
	influxdb/wal/powerwall/daily/25/
	influxdb/wal/powerwall/grid/27/
	influxdb/wal/powerwall/grid/7/_00002.wal
	influxdb/wal/powerwall/kwh/24/
	influxdb/wal/powerwall/kwh/4/_00002.wal
	influxdb/wal/powerwall/monthly/11/_00146.wal
	influxdb/wal/powerwall/pod/28/
	influxdb/wal/powerwall/pod/8/_00002.wal
	influxdb/wal/powerwall/pwtemps/22/
	influxdb/wal/powerwall/pwtemps/5/_00002.wal
	influxdb/wal/powerwall/raw/18/_00011.wal
	influxdb/wal/powerwall/raw/21/
	influxdb/wal/powerwall/vitals/23/
	influxdb/wal/powerwall/vitals/6/_00002.wal

no changes added to commit (use "git add" and/or "git commit -a") `

jasonacox, 2023-12-25:
Hi @cstout1  - Let's try to replicate the upgrade.sh script manually so we can see what is happening.

I suggest you backup the entire Powerwall-Dashboard directory first.

```bash
# All from the Powerwall-Dashboard director
cd Powerwall-Dashboard

# Let's see what git you are running
git --version
# Should say something like: git version 2.39.3 (Apple Git-145)

# Make sure your user (service?) has full access to modify files
ls -la

# Try a manual pull
git pull

# If that works, you can run upgrade.sh now.
# If that doesn't work, post what it says
```

cstout1, 2023-12-28:
**git version output:**
`git version 2.39.3 (Apple Git-145)`

**ls -la:**
```
Powerwall-Dashboard % ls -la
total 1280
drwxrwxr-x@ 45 service  staff    1440 Dec 24 14:25 .
drwxrwxrwx  11 root     wheel     352 Dec 28 10:26 ..
-rw-r--r--   1 service  staff   10244 Dec 24 14:37 .DS_Store
drwxr-xr-x  14 service  staff     448 Dec 25 10:23 .git
drwxrwxr-x@  3 service  staff      96 Dec 13 21:18 .github
-rw-rw-r--@  1 service  staff    1066 Dec 13 21:18 LICENSE
-rw-rw-r--@  1 service  staff   16552 Dec 13 21:18 README.md
-rw-rw-r--@  1 service  staff   23737 Dec 13 21:18 RELEASE.md
-rw-rw-r--@  1 service  staff       6 Dec 13 21:18 VERSION
-rw-rw-r--@  1 service  staff   10744 Dec 13 21:18 WINDOWS.md
-rw-rw-r--@  1 service  staff      27 Dec 13 21:18 _config.yml
-rw-r--r--   1 service  staff       0 Dec 24 14:25 asdf
drwxrwxr-x@  4 service  staff     128 Dec 19 18:53 backups
-rwxr-xr-x@  1 service  staff    3150 Dec 13 21:18 compose-dash.sh
-rw-r--r--   1 service  staff    1671 Dec 19 16:44 compose.env
-rw-r--r--   1 service  staff    1660 Dec 19 16:31 compose.env.bak
-rw-rw-r--@  1 service  staff    1676 Dec 13 21:18 compose.env.sample
drwxrwxr-x@ 17 service  staff     544 Dec 24 14:25 dashboards
drwxrwxr-x@  8 service  staff     256 Dec 28 10:19 grafana
-rw-r--r--@  1 service  staff    1437 Dec 19 16:32 grafana.env
-rw-rw-r--@  1 service  staff    1437 Dec 13 21:18 grafana.env.sample
drwxrwxr-x@ 14 service  staff     448 Dec 24 14:25 influxdb
-rw-rw-r--@  1 service  staff    3435 Dec 13 21:18 influxdb.conf
-rw-rw-r--@  1 service  staff    1964 Dec 13 21:18 location.html
-rw-rw-r--@  1 service  staff    2046 Dec 13 21:18 powerwall-v1-solar.yml
-rw-rw-r--@  1 service  staff    2509 Dec 13 21:18 powerwall-v1.yml
-rw-rw-r--@  1 service  staff    3276 Dec 13 21:18 powerwall.yml
-rw-r--r--   1 service  staff     169 Dec 24 14:25 pypowerwall.env
-rw-r--r--   1 service  staff     169 Dec 24 14:25 pypowerwall.env.bak
-rw-rw-r--@  1 service  staff     131 Dec 13 21:18 pypowerwall.env.sample
-rwxr-xr-x@  1 service  staff   12227 Dec 13 21:18 setup.sh
-rw-rw-r--   1 service  staff  212602 Dec 24 14:25 telegraf.conf
-rw-rw-r--   1 service  staff  212602 Dec 24 14:25 telegraf.conf.bak
-rw-r--r--@  1 service  staff    1818 Dec 19 16:32 telegraf.local
-rw-rw-r--@  1 service  staff    1818 Dec 13 21:18 telegraf.local.sample
-rw-r--r--   1 service  staff   11547 Dec 24 14:25 tmp.sh
drwxrwxr-x@ 14 service  staff     448 Dec 13 21:18 tools
-rw-rw-r--@  1 service  staff      20 Dec 24 14:25 tz
-rwxr-xr-x@  1 service  staff     861 Dec 13 21:18 tz.sh
-rw-r--r--   1 service  staff   11547 Dec 24 14:25 upgrade-new.sh
-rwxr-xr-x@  1 service  staff   11353 Dec 13 21:18 upgrade.sh
drwxrwxr-x@ 18 service  staff     576 Dec 13 21:18 v6.5.1
-rwxr-xr-x@  1 service  staff   15368 Dec 13 21:18 verify.sh
drwxrwxr-x@ 11 service  staff     352 Dec 19 16:44 weather
-rwxr-xr-x@  1 service  staff    6470 Dec 13 21:18 weather.sh
```
**git pull:**
```
Powerwall-Dashboard % git pull
There is no tracking information for the current branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>

If you wish to set tracking information for this branch you can do so with:

    git branch --set-upstream-to=<remote>/<branch> main```

jasonacox, 2023-12-28:
Something is corrupt or missing in the git configuration. Let's try a few others:

```bash
# Verify origin is set right
git remote -v

# should show this if cloned correctly...
# origin	https://github.com/jasonacox/Powerwall-Dashboard.git (fetch)
# origin	https://github.com/jasonacox/Powerwall-Dashboard.git (push)

# Set to track origin/main
git branch -u origin/main

# Check
git status

# Try again
git pull
```

cstout1, 2023-12-28:
Hi Jason, git remote -v returned nothing and git branch gave a fatal error:
```
git remote -v
Powerwall-Dashboard % git branch -u origin/main     
fatal: the requested upstream branch 'origin/main' does not exist
```

jasonacox, 2023-12-28:
Yikes! I'm not a git guru, but it seems like your setup has completely forgotten its birthday. I don't know if this indicates something corrupt on your global git setup or just this directory.

Perhaps, let's try re-clone the project from scratch to see if it works:

```bash
# Made a new test directory from your home 
cd ~
mkdir Test
cd Test

# Clone project
git clone https://github.com/jasonacox/Powerwall-Dashboard.git
cd Powerwall-Dashboard

# Test git
git status
git remote -v
git pull
```

cstout1, 2023-12-28:
Issue resolved! I made the new folder and cloned successfully and then copied the hidden git files/directories over to the other project folder and was able to run the upgrade script successfully. This was entirely my fault as I didn't see that the first setup step above "Option 1" was a required first step. I ran through the setup.sh/Option 1 process entirely without first cloning. Thank you for your very patient help in resolving this.

jasonacox, 2023-12-29:
Perfect! I'm glad you got it working, good job! 👍 

I'll close this but feel free to reopen if needed.

closed 2023-12-29 by jasonacox