jasonacox/Powerwall-Dashboard #417: Feels Like shows on Graph but has no Legend item

cstout1, 2024-01-05:
I'm not sure if this is intended behavior or a bug but I noticed that the Temperature (orange dashed) and Feels Like (yellow dashed) always show on the Energy Usage graph but there is no Legend item for "Feels Like." 

If you click on the Temperature legend item it will then show a Feels Like legend item that can be selected as well.

Is this maybe a Grafana limitation? Too many legend items currently? Default view has eight clickable (to start), but all show in the graph.

jasonacox, 2024-01-06:
This was on purpose. The legend is crowded as it is. I debated on removing "feels like" altogether but ended up just hiding it to allow users to turn it off if they want. You can edit the panel and unhide the legend if you want.

cstout1, 2024-01-06:
Great! Wasn't sure if it was intentional or not. I definitely don't need the "feels like" data on my graph so I don't mind it not showing in the legend. Thanks for the quick response.

closed 2024-01-06 by cstout1