jasonacox/Powerwall-Dashboard #418: Telegraph Error When Upgrading to 3.07

aesculus, 2024-01-07:
Did the upgrade to V 3.07 and everything seemed OK during the install. I was getting some errors wrt user so I changed all my user entries back to user: "1005:100" which seemed to resolve that issue. Then proceeded to Grafana and updated to the new dashboard and set the weather info. Dashboard came up by my real-time data had an error.  

Note I had created my own server version before that eliminated the negative solar values. Not sure if that is part of the issue now or not.

Looking into the issue I have Telegraph errors up the wazu. Here is the log.

`2024-01-07T00:04:04Z I! Loading config: /etc/telegraf/telegraf.conf
2024-01-07T00:04:04Z I! Loading config: /etc/telegraf/telegraf.d/local.conf
2024-01-07T00:04:04Z I! Starting Telegraf 1.28.2 brought to you by InfluxData th                                                                             e makers of InfluxDB
2024-01-07T00:04:04Z I! Available plugins: 240 inputs, 9 aggregators, 29 process                                                                             ors, 24 parsers, 59 outputs, 5 secret-stores
2024-01-07T00:04:04Z I! Loaded inputs: cpu disk diskio http (2x) kernel mem proc                                                                             esses swap system
2024-01-07T00:04:04Z I! Loaded aggregators:
2024-01-07T00:04:04Z I! Loaded processors: date (2x)
2024-01-07T00:04:04Z I! Loaded secretstores:
2024-01-07T00:04:04Z I! Loaded outputs: influxdb
2024-01-07T00:04:04Z I! Tags enabled: host=telegraf
2024-01-07T00:04:04Z I! [agent] Config: Interval:5s, Quiet:false, Hostname:"tele                                                                             graf", Flush Interval:10s
2024-01-07T00:04:04Z W! [outputs.influxdb] When writing to [http://influxdb:8086                                                                             ]: database "powerwall" creation failed: Post "http://influxdb:8086/query": dial                                                                              tcp 172.29.4.3:8086: connect: connection refused
2024-01-07T00:04:08Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/strings]: Get "http://pypowerwall:8675/strings": dial tcp 172.29.4.2:8675: c                                                                             onnect: no route to host
2024-01-07T00:04:08Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/soe]: Get "http://pypowerwall:8675/soe": dial tcp 172.29.4.2:8675: connect:                                                                              no route to host
2024-01-07T00:04:08Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/aggregates]: Get "http://pypowerwall:8675/aggregates": dial tcp 172.29.4.2:8                                                                             675: connect: no route to host
2024-01-07T00:04:08Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/freq]: Get "http://pypowerwall:8675/freq": dial tcp 172.29.4.2:8675: connect                                                                             : no route to host
2024-01-07T00:04:08Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/pod]: Get "http://pypowerwall:8675/pod": dial tcp 172.29.4.2:8675: connect:                                                                              no route to host
2024-01-07T00:04:08Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/alerts/pw]: Get "http://pypowerwall:8675/alerts/pw": dial tcp 172.29.4.2:867                                                                             5: connect: no route to host
2024-01-07T00:04:08Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/temps/pw]: Get "http://pypowerwall:8675/temps/pw": dial tcp 172.29.4.2:8675:                                                                              connect: no route to host
2024-01-07T00:04:10Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/alerts/pw]: Get "http://pypowerwall:8675/alerts/pw": dial tcp 172.29.4.2:867                                                                             5: connect: connection refused
2024-01-07T00:04:10Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/soe]: Get "http://pypowerwall:8675/soe": dial tcp 172.29.4.2:8675: connect:                                                                              connection refused
2024-01-07T00:04:10Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/temps/pw]: Get "http://pypowerwall:8675/temps/pw": dial tcp 172.29.4.2:8675:                                                                              connect: connection refused
2024-01-07T00:04:10Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/aggregates]: Get "http://pypowerwall:8675/aggregates": dial tcp 172.29.4.2:8                                                                             675: connect: connection refused
2024-01-07T00:04:10Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/freq]: Get "http://pypowerwall:8675/freq": dial tcp 172.29.4.2:8675: connect                                                                             : connection refused
2024-01-07T00:04:10Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/pod]: Get "http://pypowerwall:8675/pod": dial tcp 172.29.4.2:8675: connect:                                                                              connection refused
2024-01-07T00:04:10Z E! [inputs.http] Error in plugin: [url=http://pypowerwall:8                                                                             675/strings]: Get "http://pypowerwall:8675/strings": dial tcp 172.29.4.2:8675: c                                                                             onnect: connection refused
2024-01-07T00:04:14Z E! [outputs.influxdb] When writing to [http://influxdb:8086                                                                             ]: failed doing req: Post "http://influxdb:8086/write?db=powerwall&rp=raw": dial                                                                              tcp 172.29.4.3:8086: connect: connection refused
2024-01-07T00:04:14Z E! [agent] Error writing to outputs.influxdb: could not wri                                                                             te any address

and repeats
`

jasonacox, 2024-01-07:
> no route to host
> connect: connection refused

This would suggest docker compose networking issues. The containers are not able to talk to each other (provided they are all running).  What do you get when you run `verify.sh`? 

Also, can you try to rebuild the docker network?

```bash
# Always good idea to backup your setup before making changes

# Tear down stack
./compose-dash.sh down

# Rebuild stack
./compose-dash.sh up -d
```

Example

[image]

aesculus, 2024-01-07:
OK. We may have identifified the problem and it sounds like it might be related to my prior 'fix' of making my owner server to iliminate the negative solar values. Perhaps my old server is missing something your new version has?

Here is the verify output:
`Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): ERROR: Not Listening
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware Version: Unknown

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking tesla-history
----------------------------------------------------------------------------
 - Skipped: Only required in 'solar-only' configuration

Checking weather411
----------------------------------------------------------------------------
 - Config File weather/weather411.conf: GOOD
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 40.62}
 - Version: 0.2.2

One or more tests failed.
`

aesculus, 2024-01-07:
And I had redone the docker network before I pulled the Telegraph logs.

aesculus, 2024-01-07:
So this appears to be a pwpowerwall problem. For some reason it does not work with the jasonacox/pypowerwall:0.6.4t32 version or my custom version.  

I am seeing this error in the log repeat itself over and over and then the container stops, restarts again.

Traceback (most recent call last):
  File "/app/server.py", line 54, in <module>
    pool_maxsize = int(os.getenv("PW_POOL_MAXSIZE", "15"))
ValueError: invalid literal for int() with base 10: '10PW_STYLE=grafana-dark'

jasonacox, 2024-01-07:
Hi @aesculus  - that looks like an error in your `pypowerwall.env` file, like a missing line break? Can you check that file? If you paste it here, be sure to redact your password and IP.

aesculus, 2024-01-07:
OK. Now I have a convoluted mess. I tried changing the folder file names between my original and the new one and somehow the file system has them all wonked.

I did edit the sample and put my parms in it and now the pwpowerwall does not error (except for some statement about tz but that is set).

But my grafana is all screwed up (probably a mixture of old and new) and won't start.

I am going to run the upgrade again and see if I can get this back on track. Stay tuned.

aesculus, 2024-01-08:
Its working again with the default pwpowerwall server. I will need to add mine back in again. Will the old code from last August be OK or do I need to rebuild it from a current version?

Also now I need to find a way to restore my data I backed up too.

And then I need to look for where my other dashboards are.

All in all I probably just needed to fix that pypowerwall.env file but that is life.

closed 2024-01-08 by aesculus