jasonacox/Powerwall-Dashboard #419: Remove Negative Solar Values in pwpowerwall server

aesculus, 2024-01-08:
Because of the uncertainty in the CT sensors, solar values are often small but negative during non production times. They should never be below 0.

A few of us also have some small loads on circuits measured by the solar CTs and that is even worse. Seeing negative solar values in both the dashboard and real-time display is disconcerting at best.

The fix as you know is very simple and I have it running in my own version of pwpowerwall server. But it would be so much better if it the feature was added to the production pwpowerwall.

Here is the simple code in the if self.path == '/aggregates' or self.path == '/api/meters/aggregates':
`            # Parse JSON string
            data = json.loads(jsonpayload)
            # Update the "instant_power" value in "solar" and "load"
            if data["solar"]["instant_power"] < 0:
                data["load"]["instant_power"] += abs(data["solar"]["instant_power"])
                data["solar"]["instant_power"] = 0
            # Convert back to JSON string
            message = json.dumps(data)
`

BuongiornoTexas, 2024-01-08:
Except that some of us have inverters that do have periodic vampire draws - I've seen mine flick on and off briefly in full dark - and from (possibily faulty) memory, the fronius units are capable of providing data logging pages at anytime of day. So it's not entirely correct to say that solar should never be below zero. It can also be an extremely useful diagnostic in the early days of system setup when stuff is misconfigured (again, personal experience).

Can I suggest that this would be better made an optional functionality?

aesculus, 2024-01-08:
Optional if fine with me. A simple parameter indication you want negative values truncated to 0 would be fine.

mcbirse, 2024-01-09:
There is already a way you can do this by using a Telegraf Starlark script to filter or change values before they are saved to InfluxDB - refer here: https://github.com/jasonacox/Powerwall-Dashboard/discussions/112

I even include an example of exactly what you want - change negative solar to 0.

There was discussion previously about pypowerwall changing the values, but this is not preferred as it should only be returning results without interpretation. The above solution is adaptable to many difference custom requirements anyway.

aesculus, 2024-01-09:
@mcbirse thank you for the tip. I know you said this is before it gets put into the influxDB but is it also before it gets the real-time display? Maybe a nit from a data perspective but ...

Like I said I have already created a special version of pwpowerwall that filters before it gets to the DB or the real-time display, but I would like to be out of that business and I am afraid one day it will bite me.

mcbirse, 2024-01-09:
@aesculus - If the real-time display is querying pypowerwall directly, then unfortunately not, as it is Telegraf that is doing the filtering.

jasonacox, 2024-01-09:
Hi @aesculus  - For the reasons @BuongiornoTexas and @mcbirse , we want the default behavior of pypowerwall to represent the exact data the Powerwall Gateway (or Tesla Cloud for that matter) reports. Data consumers (like telegraph, display devices, or UIs) can make the decision to eliminate the negative solar values if desired.  

However, I'm not opposed to an enhancement to pypowerwall that allows custom filters. If you or someone wanted to add code to create this optional toggle logic, submit it as a pull request and we can review it.

aesculus, 2024-01-09:
Hi Jason: I totally understand the point. But interestingly, Tesla themselves do not show negative solar. On my representative circuit where I have a load on the solar side, they take the energy that is negative and put it onto the home load. Not sure how they do this, but they do. It's probably because I have both load and solar sensors on that circuit so when they see a negative value on one side they dismiss it. And if they see a positive value on the load side they dismiss that. This I am sure, happens in the back end. But it shows correctly on the app.

I don't think I have the background to create a custom filter so perhaps someone else will come buy and be inspired to create one, but there are probably only a handful of us in the world with the issue. :-)

closed 2024-01-09 by aesculus