jasonacox/Powerwall-Dashboard #423: No 'Powerflow' or 'Current State'

llamakarma, 2024-01-17:
**Problem Description**

New, clean installation, does not display anything for 'Current State' and the Powerflow animation is absent.

In Chrome, the Powerflow pane reports 'x.x.x.x refused to connect', in Safari it shows a looping progress bar.

I've tried the manual install and the auto-install. I am using the cloud connection rather than local (not clear if this could be the reason?)

verify.sh looks clean

I can't see port 8675 open on the machine at all - from what I could find, this is required for at leat the Powerflow - I presume also for the current state 

Any help would be very much appreciated!

```
netstat -tl

Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:9000            0.0.0.0:*               LISTEN     
tcp        0      0 localhost:domain        0.0.0.0:*               LISTEN     
tcp        0      0 localhost:36077         0.0.0.0:*               LISTEN     
tcp        0      0 0.0.0.0:8086            0.0.0.0:*               LISTEN     
tcp6       0      0 [::]:ssh                [::]:*                  LISTEN     
tcp6       0      0 [::]:8581               [::]:*                  LISTEN     
tcp6       0      0 [::]:51764              [::]:*                  LISTEN     
tcp6       0      0 [::]:9000               [::]:*                  LISTEN     
tcp6       0      0 [::]:9273               [::]:*                  LISTEN     
tcp6       0      0 [::]:8086               [::]:*                  LISTEN     
```

**System Details**
Describe the host system you are trying to use for the Dashboard:

Ubuntu 22.04.3 VM (under Proxmox) (4 CPU, 4GB) kernel 5.15.0-91-generic

**Verify Tool Output**
```
Verify Powerwall-Dashboard 3.0.8 on Linux - Timezone: Europe/London
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking configuration
----------------------------------------------------------------------------
 - Dashboard configuration: solar-only
 - EnvVar COMPOSE_PROFILES: solar-only

Checking pypowerwall
----------------------------------------------------------------------------
 - Skipped: Only required in 'default' configuration

Checking telegraf
----------------------------------------------------------------------------
 - Skipped: Only required in 'default' configuration

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking tesla-history
----------------------------------------------------------------------------
 - Config File tools/tesla-history/tesla-history.conf: GOOD
 - Auth File tools/tesla-history/tesla-history.auth: GOOD
 - Container (tesla-history): GOOD
 - Version: 0.1.4

Checking weather411
----------------------------------------------------------------------------
 - Skipped: weather411 not set up

All tests succeeded.
```

jasonacox, 2024-01-18:
Hi @llamakarma !

The installation method you picked is for Tesla Solar-Only. This current solar-only version does not use pypowerwall to pull the data so you will not see pypowerwall running on port 8675. That means that the power flow animation or live data meters will not show. There is a solar-only dashboard you can use that removes these panels.  However, there is now another option in development.

We have a new version of pypowerwall that @mcbirse and I are working on that will use pypowerwall for solar-only installs (also works for Powerwall 3 owners which do not get a local API).  Details are here: https://github.com/jasonacox/Powerwall-Dashboard/pull/414

What type of system do you have?  Solar-only, Powerwall 2/+ or Powerwall 3?  If you have a Powerwall, PW2 or PW+, you can use the "local mode" and I would recommend it. If you have Solar-only or. Powerwall 3, you will need to use cloud mode. If you want to try to new setup (beta mode), you can follow the directions here: https://github.com/jasonacox/Powerwall-Dashboard/pull/414#issuecomment-1882355993

mcbirse, 2024-01-18:
Hi @llamakarma

Since you were starting with a clean install anyway, you may want to test a fresh install of the v4.0.0 pre-release. You can do so following the instructions here: https://github.com/jasonacox/Powerwall-Dashboard/pull/414#issuecomment-1897867180

Any feedback or issues you find can be posted in that thread and would be most welcome!

llamakarma, 2024-01-18:
Hi both,

Thank you for your quick reply to what probably seems like a dumb question!

I misinterpreted the setup script:

```
Current: Tesla Cloud

 1 - Local Access (Powerwall 1, 2, or + using extended data from Tesla Gateway on LAN) - Default
 2 - Tesla Cloud  **(Solar-only, Powerwall 1, 2, +, or 3 using data from Tesla Cloud)**
```

It reads to me that option 2 is suitable for all options and my search-fu didn't come up with any other answers. 

Anyway - my setup is PW2 with Backup Gateway2, my solar input is non-Tesla. I have just re-setup with the local only connection and things are working just fine - thank you!

I will take a look at v4.0 over the weekend.

Thanks again for the quick assistance

jasonacox, 2024-01-19:
I'll flag this to update documentation and possibly the setup script.

We should probably switch to this to indicate that the 2nd option is only need if you don't have local access. Perhaps something like this:

 1 - Local Access (Powerwall 1, 2, or + using the Tesla Gateway on LAN) - Default
 2 - Tesla Cloud  (Solar-only systems or Powerwalls without LAN access)