jasonacox/Powerwall-Dashboard #425: Powerwall Vitals metrics stopped updating

ceeeekay, 2024-01-20:
**Problem**
First of all, this dashboard is awesome :) I've only had it installed a few days, and everything has been working great - up until a couple of hours ago. Everything in the PW Vitals panel has completely stopped populating on their respective graphs. The Vitals metrics are the main reason I installed the dashboard -- so I could see individual charge states for each Powerwall.

Every other panel in the Powerwall Dashboard, including the flow meter, and even the combined Powerwall charge level in the Energy Usage graph are all updating fine. I can manually access the gateway API and get stats, e.g., `<gateway>/api/system_status` is responding OK. Networking is good (gateway has an ethernet cable), everything else seems OK, I'm just suddenly missing the most recent Vitals in the dashboard.

**Possibly unrelated:** at around the same time the Vitals stopped updating, I started seeing 401 errors in my Homebridge console for the [Homebridge Tesla Powerwall](https://github.com/datMaffin/homebridge-tesla-powerwall) plugin. These were resolved by simply restarting Homebridge and allowing the plugin to authenticate again. I'm not sure if these 401s are relevant or there has possibly been an upstream change to the gateway at the same time. I'm Including this info just in case it's meaningful.

I've rebooted the server hosting the containers and all containers have restarted, but I still have no recent data for the Vitals panel.

I'm not even sure which container to start looking at for possible errors. Nothing really stands out in the logs that I've checked so far.

**Screenshots**
[image]

As you can see from the screenshot, Vitals have stopped at around 20:48, but Alerts (for example) are still being populated past 23:00.

What do I need to do next? Happy to provide logs if you can tell me which container might be the culprit.

jasonacox, 2024-01-20:
Hi @ceeeekay - I suspect Tesla upgraded your system to 23.44.0. You should see what version you are running in the power flow animation panel.  You can also see it by going to the pypowerwall container API: http://localhost:8675/version (change localhost to the IP address running the container) and you can see what vitals you get back at http://localhost:8675/vitals 

Tesla has removed the vitals API with the firmware 23.44.0 upgrade.  You can join the discussion here:  #402

I understand your interest in these vitals, and you are not alone. The general speculation is that Tesla was planning to remove all local (LAN based) customer accessible APIs and shift to the Tesla App for customers (cloud based) as the only way to get this data and the Tesla Pros app for installers and advanced metrics. The vitals API is the first to go. The Powerwall 3 was released with NO local API and NO web portal access (see #387 ). 

I understand that Tesla needs to upgrade the system to serve all their customers, making the installation and use a better experience for everyone. That may impact those of us automation/data enthusiast, who are using unofficial APIs to get our data or control our systems.  However, I'm also optimistic that we will be able to find a new way to get at least the basic data via local APIs and even hopeful that we may get more with some additional work. 

Some work is already going on in this area:

* [Cloud Based Option](https://github.com/jasonacox/Powerwall-Dashboard/pull/414) - This project and pypowerwall will support a Tesla Cloud based mode very soon (doing last testing with @mcbirse - who has done much of this work! 🙏 ). We originally started this effort to enable Tesla Solar-only owners to use the same dashboard, but this has the advantage of working for every Tesla Energy system (including Powerwall 3). If all the APIs are removed, this will be a path to still gather and display live data via the project.
* [Tesla FleetAPI](https://github.com/jasonacox/pypowerwall/tree/main/tools/fleetapi#tesla-developer---fleetapi-for-powerwall) - I have been exploring and adding tooling to support the [official Tesla API](https://developer.tesla.com/docs/fleet-api) (also cloud based) for accessing Tesla Energy products for monitoring and control. This may be a better approach than the [unofficial Tesla Owners API](https://tesla-api.timdorr.com/) and the [TeslaPy](https://github.com/tdorssers/TeslaPy) python API that many projects like ours are using to access Tesla systems (including Vehicles) via the Tesla Cloud. It require a bit more setup work but at least the APIs are published by Tesla.
* [Tesla Pros API](https://github.com/jasonacox/Powerwall-Dashboard/discussions/392) - This new locally accessible API is not very accessible. It is only available on the  Gateway WiFi access point and oriented toward setup instead of monitoring. Also, it likely has a wide blast radius for destroying your system (and warranty) if used incorrectly as it is made for the installer (always in Wizard mode). I'm hopeful there will be another less risky path.  

More coming. As I said above, I'm still hopeful that a local API will persist (hopefully with some vital like data) or be created soon (for Powerwall 3). Having a local portal/ API set is extremely helpful for power and internet outage events. I believe all Tesla Powerwall owners could benefit from and would appreciate that. If you agree, please 👍  - I'll do what I can to encourage it from my end too.

That was a lot, I'm sorry. Hopefully that was helpful. 😁

ceeeekay, 2024-01-20:
@jasonacox yep, definitely 23.44.0 has caused it.

I've added a comment to the discussion. Thanks.