jasonacox/Powerwall-Dashboard #427: Release 4.0.1 - not seeing voltages

youzer-name, 2024-01-28:
I ran upgrade.sh and saw it do the upgrade from 4.0.0 to 4.0.1. The only non-standard thing in my process is I'm using powerwall.extend.yml to create a powerwall-history container that pulls the data for my Tesla solar site.

I am not seeing any data populating into the voltages panel. I deleted and recreated the dashboard from dashboard.json, and I see the new regex entry in the overrides, so everything looks to be current.

I tried a ./compose-dash down and ./compose-dash up -d which didn't help.

<details>

Upgrade Powerwall-Dashboard from 4.0.0 to 4.0.1
---------------------------------------------------------------------
This script will attempt to upgrade you to the latest version without
removing existing data. A backup is still recommended.

Upgrade - Proceed? [y/N] y

Resetting Timezone to Default...

Pull influxdb.sql, dashboard.json, telegraf.conf, and other changes...

Saved working directory and index state WIP on main: 4b5d0ba Merge pull request #414 from jasonacox/v4.0.0
remote: Enumerating objects: 14, done.
remote: Counting objects: 100% (14/14), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 14 (delta 6), reused 13 (delta 6), pack-reused 0
Unpacking objects: 100% (14/14), 25.18 KiB | 661.00 KiB/s, done.
From https://github.com/jasonacox/Powerwall-Dashboard
   4b5d0ba..7dd1297  main       -> origin/main
 * [new tag]         v4.0.1     -> v4.0.1
Updating 4b5d0ba..7dd1297
Fast-forward
 RELEASE.md                |   6 ++-
 VERSION                   |   2 +-
 dashboards/dashboard.json | 157 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++----
 influxdb/influxdb.sql     |   5 ++
 upgrade.sh                |   2 +-
 5 files changed, 161 insertions(+), 11 deletions(-)

Setting Timezone back to America/New_York...

Updating Powerwall-Dashboard stack...
Including powerwall.extend.yml
Running Docker Compose...
[+] Running 6/6
 ✔ Container pypowerwall    Running                                                         0.0s 
 ✔ Container influxdb       Started                                                         8.8s 
 ✔ Container tesla-history  Started                                                         9.7s 
 ✔ Container weather411     Started                                                         9.9s 
 ✔ Container grafana        Started                                                        10.0s 
 ✔ Container telegraf       Started                                                         9.6s 

Waiting for InfluxDB to start...
....... up!

Add downsample continuous queries to InfluxDB...

Deleting old pyPowerwall...
pypowerwall
pypowerwall

Deleting old telegraf...
telegraf
telegraf

Deleting old weather411...
weather411
weather411

Deleting old tesla-history...
tesla-history
tesla-history

Restarting Powerwall-Dashboard stack...
Including powerwall.extend.yml
Running Docker Compose...
[+] Running 6/6
 ✔ Container pypowerwall    Started                                                         2.6s 
 ✔ Container influxdb       Running                                                         0.0s 
 ✔ Container tesla-history  Started                                                         2.3s 
 ✔ Container weather411     Started                                                         2.7s 
 ✔ Container grafana        Running                                                         0.0s 
 ✔ Container telegraf       Started                                                         4.6s

</details>

I see the new CQ's (vitals 8 - vitals 12) in InfluxDB but I don't see any recent data in vitals.http. The last records are from before my firmware updated. Looking at the source of the CQ, I tried to run:

SELECT PW1_v_out, PW2_v_out, PW3_v_out, PW4_v_out, PW5_v_out, PW6_v_out, PW7_v_out, PW8_v_out, PW9_v_out, PW10_v_out, PW11_v_out, PW12_v_out FROM raw.http order by time desc limit 5

That returned no data. So I don't have the raw data coming in.

select * from raw.http order by time desc limit 1 returns these columns: 

<details>
time                 backup_reserve_percent battery_energy_exported battery_energy_imported battery_frequency battery_i_a_current battery_i_b_current battery_i_c_current battery_instant_apparent_power battery_instant_average_current battery_instant_average_voltage battery_instant_power battery_instant_reactive_power battery_instant_total_current battery_num_meters_aggregated battery_timeout grid_status host     load_energy_exported load_energy_imported load_frequency load_i_a_current load_i_b_current load_i_c_current load_instant_apparent_power load_instant_average_current load_instant_average_voltage load_instant_power load_instant_reactive_power load_instant_total_current load_timeout month nominal_energy_remaining nominal_full_pack_energy percentage site_energy_exported site_energy_imported site_frequency site_i_a_current site_i_b_current site_i_c_current site_instant_apparent_power site_instant_average_current site_instant_average_voltage site_instant_power site_instant_reactive_power site_num_meters_aggregated site_timeout solar_energy_exported solar_energy_imported solar_frequency solar_i_a_current solar_i_b_current solar_i_c_current solar_instant_apparent_power solar_instant_average_current solar_instant_average_voltage solar_instant_power solar_instant_reactive_power solar_instant_total_current solar_num_meters_aggregated solar_timeout source time_remaining_hours url                                year
</ details>

jasonacox, 2024-01-28:
Hi @youzer-name, that response you show is what you will get if you are in Tesla Cloud mode instead of Local Mode. Did you happen to switch to cloud mode?  Run `./verify.sh` or http://pypowerwall:8675/help to see what mode it is running.

If not, try http://pypowerwall:8675/api/system_status - that is where the new voltage data is coming from.

youzer-name, 2024-01-28:
> Hi @youzer-name, that response you show is what you will get if you are in Tesla Cloud mode instead of Local Mode. Did you happen to switch to cloud mode? 

That was it. I think when I did the 4.0.0 update I picked cloud mode, thinking that was required to get all the data after the firmware update. I re-ran setup.sh and am back in local mode and the data is populating.

closed 2024-01-28 by youzer-name