jasonacox/Powerwall-Dashboard #429: Container keeps restarting - Connection error message

marcbaier, 2024-01-30:
I've been using the Powerwall-Dashboard on my Synology NAS for quite some time now and suddenly on January 26th it stopped working. No data is shown and the log shows me this:

[image]

As far as I can tell the IP address of the powerwall has not changed. 

I tried these commands and I am getting an error that the site can not be reached:

http://192.168.86.58:8675/vitals - Powerwall vitals
http://192.168.86.58:8675/stats - pypowerwall stats

jasonacox, 2024-01-31:
HI @marcbaier  - What version are you running? If you have anything before pyPowerwall [v0.7.6](https://github.com/jasonacox/pypowerwall/releases/tag/v0.7.6) you will need to upgrade. There is a bug in pyPowerwall that causes it to continually retry 404 responses from the Powerwall.  The recent Firmware update removed the /api/device/vitals API (resulting in a 404) and pypowerwall will pound the Powerwall until it is gets blocked.  Get the latest by upgrading your Powerwall-Dashboard to at least v3.0.8 (or latest, v4.0.1).

marcbaier, 2024-01-31:
Jason - thanks for the help ! How would I go about doing that ? Do I need to run setup.sh again or is there an easy way to just update the container to the latest version ?

jasonacox, 2024-02-01:
On most systems, `upgrade.sh` would upgrade all the components which is the recommend path. However, I don't know how it works on a Synology. Did you manually set it up? If so, you will need to manually update `powerwall.yml` and make it match the current version - see:

https://github.com/jasonacox/Powerwall-Dashboard/blob/7dd1297642feb8e574f9fb8876fc5d8e9053b0f0/powerwall.yml#L22-L37

marcbaier, 2024-02-01:
OK so I did this:
- Stopped the container
- Edited the powerwall.yml file to this:

- pypowerwall:
        image: jasonacox/pypowerwall:0.7.7t40
        container_name: pypowerwall
        hostname: pypowerwall
        restart: unless-stopped
        user: "1029:1029"
        ports:
            - target: 8675
              published: 8675
              mode: host
        env_file:
            - pypowerwall.env

- restarted the container
- But the container is still called: pypowerwall:latest

I guess I must be doing this wrong ?

marcbaier, 2024-02-01:
Ok I tried ./upgrade.sh and got this error: 
[image]

ran "git pull" and then ./upgrade.sh again:
[image]

Then I tried your optional commands:
[image]

marcbaier, 2024-02-01:
ahh - found a missing ".sh" in the command ./compose-dash.SH up -d
ran that and now I am getting: 

Backupadmin@BackupStation:~/Powerwall-Dashboard$ ./compose-dash.sh up -d
ERROR: Missing compose.env file.
Please run setup.sh or copy compose.env.sample to compose.env.
Backupadmin@BackupStation:~/Powerwall-Dashboard$ 

what puzzles me: when I type "ls": I don't see the compose.env file:
[image]

But there is a file in the folder when I open the folder on the Synology:
[image]

marcbaier, 2024-02-01:
Ok so I figured I have permission issues and just created new versions of pypowerwall.env and compose.env directly via the terminal. 

Now ./upgrade.sh ran all the way through. No errors. 

However, the grafana and telegraf containers are in a restart loop. 
Error message for graphana: 
[image]

Telegraf:
[image]

marcbaier, 2024-02-01:
I have opened Issue 22 again and will do the whole install again from scratch. With new users / permissions etc. Will get back then ;)

marcbaier, 2024-02-01:
Now that the reinstall is done, which worked without any errors, all containers are running withouth error messages. 
[image]

I configured the dashboard as per the instructions but am not getting any data: 

[image]

Is there a way to figure out where the issue is ?

jasonacox, 2024-02-01:
Congrats, @marcbaier !  To keep old data, the key is to preserve the influxes folder and files in there. You can also use the [Tesla History Import Tool](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/tesla-history) to pull history data from the Tesla cloud.

The `./verify.sh` script can be run to see what things are working or not (including file permissions). I don't know if it would have helped in your case. And, you can always re-run `./setup.sh` to ensure all the files are there, but sounds like your permissions were an issue too.

marcbaier, 2024-02-03:
Thanks so much for the hint about the import tool. Used it and now everything is back in order ! Have a great weekend !

jasonacox, 2024-02-04:
Awesome! Huge credit to @mcbirse for creating the tool... its a life saver. 🙏