jasonacox/Powerwall-Dashboard #431: Grid Status, Savings, Self-Powered, Solar Energy Year panels no longer work

sassyass, 2024-01-31:
These were all working fine last year, but about a month ago (the week before Christmas maybe?) these panels stopped working and are now spitting out errors.

[image]
[image]

When I look at them they are all giving the same error: "Status: 500. Message: InfluxDB returned error: error parsing query: found \/, expected identifier, string, number, bool at line 1, char 149"

[image]

I took a peek with Chronograph, and it does look like values are still being populated into InfluxDB:
[image]

There was probably a pypowerwall update before this but not 100% sure; and I was waiting for more updates in case it would magically fix itself, but it's still persisted even after the latest updates.

Any help would be appreciated.

jasonacox, 2024-02-01:
Hi @sassyass - did you import the latest `dashboard.json` and overwrite the existing one?

sassyass, 2024-02-01:
I did import the latest `dashboard.json`, but it still gives the same errors.

jasonacox, 2024-02-01:
Yikes. This seems more like a corruption in the influxdb. Can you try these:

```bash
# Check InfluxDB for errors
docker logs influxdb

# Run setup query to see if there are errors:
docker exec --tty influxdb sh -c "influx -import -path=/var/lib/influxdb/influxdb.sql"
```

sassyass, 2024-02-01:
hmm I was fiddling around with the query, and it looks like it is the time range function.

I changed 
`WHERE  $timeFilter tz('$tz')`
to
`WHERE $timeFilter
GROUP BY time($__interval) fill(null)`

and that seems to mostly have fixed it; in the self-powered panel I just took out the `tz('$tz')`

jasonacox, 2024-02-01:
Ah! Nice troubleshooting. I wonder if your timezone variable was missing. If you import the dashboard.json, the timezone should not be blank or have any spaces:

[image]

sassyass, 2024-02-01:
Hm, no it is populated and looks exactly like that, `America/Los_Angeles`

closed 2024-02-01 by sassyass