jasonacox/Powerwall-Dashboard #432: Use 0.0.0.0 as Ip, or bind to a specific IP

kylebrowning, 2024-02-14:
I might not have found an existing issue but was wondering if there's a way. 

Im trying to use a cloud hosted container to setup this project so I can have the data always coming in, and not have some machine on at home all the time.

The problem im running into is everything is bound to `localhost`, which doesn't work for an iframe.

Is there a way to supply an IP to bind everything to?

[image]

jasonacox, 2024-02-14:
Hi @kylebrowning,

⚠️ I would advise against hosting a public URL for your system. The pypowerwall container is essentially a proxy to your Powerwall. Use with caution.

Tell me a bit more about your setup. Do you have a VPN tunnel set up to get back to your home network (to your Powerwall Gateway)?  Or are you using the cloud mode to fetch your power data?

As to IP address, the docker compose setup will expose the services to whatever IP is available to the host. You should be able to see pypowerwall at http://x.x.x.x:8675/help - But you need to make sure you allow that port (TCP 8675) in your firewall.

kylebrowning, 2024-02-14:
Gotcha! Thanks for getting back to me so quickly. I actually am on Powerwall 3's and for what its worth (I know about the issues with them) , setting up locally things work
[image]

I plan on locking down the grafana interface to a vpn but things didn't go as easily on my hosted container. effectively Nothing worked (as you can see by the screenshot). When I inspected the web page earlier the iframe had `localhost` on it, but I just tried again and it has the right ip (maybe I was looking at the wrong chrome tab haha)

My main goal is to have something be dedicated, as running it from my laptop will probably result in data loss

jasonacox, 2024-02-14:
So the working dashboard is from your laptop?  It looks good. And actually, the `tesla-history` script should work for cloud mode (Powerwall 3).  It uses the same API.

Back to your cloud hosted dashboard, I can see http://172.233.130.10:9000/ but I don't see http://172.233.130.10:8675/ so I suspect either pypowerwall is not actually running or there is a security / firewall setting blocking that. You will need to fix that to see the power flow animation.  However, if you are NOT getting data on the dashboard (it looks blank), that likely means some of the services are not running correctly.  You can run the `verify.sh` from your cloud server to see what is not set up correctly.

kylebrowning, 2024-02-14:
Ahh gotcha, soo
```
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): ERROR: Not Listening
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware Version: Unknown
 - Cloud Mode: YES - ERROR: Not Connected to Tesla Clou
 ```

kylebrowning, 2024-02-14:
Looks like im having the same issues as the others using ubuntu, will close this ticket, thanks

kylebrowning, 2024-02-14:
relates to #110

closed 2024-02-14 by kylebrowning

jasonacox, 2024-02-15:
Can you provide the logs (at least any errors) from pypowerwall?

```bash
docker logs pypowerwall
```