jasonacox/Powerwall-Dashboard #434: Trouble getting data from Powerwall

Thefirebeast, 2024-02-22:
Thank you whoever lends me a life raft!

**Problem Description**
I am unable to connect Grafana to my powerwall. 
I have connected to the Powerwalls wifi and ran "python3 -m pypowerwall scan" and verified its the same ip in the config file, I have also made sure the email/password are correct.

When I run "docker logs -f pypowerwall" I receive this 
_02/21/2024 07:59:59 PM [proxy] [INFO] pyPowerwall [0.7.7] Proxy Server [t40] - HTTP Port 8675
02/21/2024 07:59:59 PM [proxy] [INFO] pyPowerwall Proxy Started
02/21/2024 08:00:01 PM [proxy] [ERROR] Invalid Powerwall Login
02/21/2024 08:00:01 PM [proxy] [ERROR] Fatal Error: Unable to connect. Please fix config and restart.
02/21/2024 08:37:17 PM [proxy] [INFO] pyPowerwall [0.7.7] Proxy Server [t40] - HTTP Port 8675
02/21/2024 08:37:17 PM [proxy] [INFO] pyPowerwall Proxy Started
02/21/2024 08:37:22 PM [proxy] [ERROR] Unable to connect to Powerwall at https://192.168.91.1
02/21/2024 08:37:22 PM [proxy] [ERROR] Fatal Error: Unable to connect. Please fix config and restart._

I have tried the command while connected to the powerwalls wifi and have also ran it when connected to my house wifi, I even ran the command when I had internet disabled yet still got the same error log

**System Details**
Describe the host system you are trying to use for the Dashboard: Lenovo Thinkcentre M53, connected to internet via ethernet
* Hardware: Intel Pentium J2900, 4Gb ram
* OS Version: Linux Mint 21.1 Cinnamon
* Other Specifics:

**Verify Tool Output**
If possible, please run the `verify.sh` found in the Powerwall-Dashboard main folder. It will attempt to verify all the services needed to run Powerwall-Dashboard and identify errors. Please paste the output or screenshot here:

user@user-ThinkCentre-M53:~/Powerwall-Dashboard$ ./verify.sh 
Verify Powerwall-Dashboard 4.0.2 on Linux - Timezone: America/Chicago

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - **Service (port 8675): ERROR: Not Listening**
 - **Version: Unknown**
 - **Powerwall State: ERROR: Not Connected - Firmware Version: Unknown**
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 20.96}
 - Version: 0.2.3

One or more tests failed.
[image]
[image]

Thefirebeast, 2024-02-22:
It is a Powerwall 2. The powerwall and my computer are connected to the same network, the network scan finds nothing unless I connect directly to the Powerwalls network. 

To clarify something, I'm trying to set this up for my dad and cell coverage is currently bonked everywhere it seems so I can't text him to clarify this but he told me before that the email/password he gave me is what he used to setup his Tesla app and later added the power wall to it. Should we be using that login info or is there another set in login credentials we need to use? I will be at his house later today or tomorrow

Thefirebeast, 2024-02-22:
[image]

NickBeacroft, 2024-02-23:
I had a working system and it was stable until the 23.44.0 firmware.
I updated to the v4.0.1 version and managed to get it working again, after logging in to my Powerwall 2.

It ran OK for about a week.

It stopped working again yesterday, but OK again after logging in to my Powerwall again.

Now it has stopped working and I cannot get it running.

I just upgraded to v4.0.2 to see if that would help. But still does not work. Even after connecting and logging in to my Powerwall.

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): ERROR: Not Listening
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware Version: Unknown
 - Cloud Mode: NO

02/23/2024 07:41:19 PM [proxy] [INFO] pyPowerwall [0.7.7] Proxy Server [t40] - HTTP Port 8675
02/23/2024 07:41:19 PM [proxy] [INFO] pyPowerwall Proxy Started
02/23/2024 07:41:22 PM [proxy] [ERROR] Unable to connect to Powerwall at https://192.168.25.101
02/23/2024 07:41:22 PM [proxy] [ERROR] Fatal Error: Unable to connect. Please fix config and restart.

jasonacox, 2024-02-24:
@Thefirebeast  - The portal will prompt you to use (upgrade) to Telsa Pros (now also called Tesla One), but you can bypass that screen and will need to do that: https://github.com/jasonacox/Powerwall-Dashboard/discussions/367#discussioncomment-7278760

@Thefirebeast  - The password for the Tesla App is likely not the same one for the Powerwall (mine are different). I set up the password using the portal (see above) by first going into the Installer screen and updating the customer password.

@NickBeacroft - Some ideas:
* Do you see your Powerwall when you scan for it (`python3 -m pypowerwall scan`)? 
* Can you tell what firmware version it is running (maybe even by looking at Tesla App)?  
* Rate limit? I have seen this happen if you hit the API end point of the Powerwall too much (which is why pypowerwall does caching) so I would recommend stopping pypowerwall (and any other service you have hitting the Powerwall) for 10-30m and see if it clears the rate limit protection that may be causing this.
* The fact that you can log in seems to indicate that it is working. I would check you `pypowerwall.env` file to make sure it has the correct IP, username and password. I know that is unlikley but good to verify. I had a password corrupt and that caused rate limit as well.
* Last straw - You can switch to `cloud mode` by running setup.sh again. You will not get all the data points, but at least you will get core data and the dashboard back.

NickBeacroft, 2024-02-24:
Hi,

Thanks for the advice.

I stopped the pypowerwall and re-checked the other aspects.

This first scan did not find the Powerwall, but I can ping it.
After going through the non-upgrade login procedure again, the scan did find the Powerwall.

Firmware: 23.44.0 eb113390

With the pypowerwall now stopped for 30 minutes, I started it up again.

All looks good again and based on your feedback, it looks perhaps more of an API rate limit.
As I have not changed anything, is it possible the new firmware has a lower rate limit?

jasonacox, 2024-02-25:
Great to hear @NickBeacroft !  I don't know that the new Firmware has more rate limit restrictions, but you probably hit two things: 1) The old pypowerwall version (prior to your upgrade) had a bug that would endlessly retry API endpoints so when Tesla removed the `vitals` URL, pypowerwall would hit the rate limit on retries. 2) The Tesla Pros "don't upgrade" steps on the portal seems to be required in some cases or your dashboard and pypowerwall will continue to loop through trying to login, also causing rate limit.

Thefirebeast, 2024-02-26:
The good: It was indeed an issue with the password, after using the different password I was able to get some data. Thanks for helping me with that.

The bad: I don't know if I'm extremely blind but I did not see anything about the powerwall firmware update screwing the data up in the installation guide. Will have to wait it out in hopes of regaining data access.

jasonacox, 2024-02-26:
> The bad: I don't know if I'm extremely blind but I did not see anything about the powerwall firmware update screwing the data up in the installation guide. Will have to wait it out in hopes of regaining data access.

I'm not following... or is this a joke? 😂

mihailvovk, 2024-04-11:
probably not related, but I'm loosing starting yesterday IP connectivity via wifi. reconnected to another network and then back to regular one, - it worked for some time and not anymore again. Anyone having same symptoms? We had storm yesterday with lots of lightning and heavy winds

jasonacox, 2024-04-11:
Hi @mihailvovk - I had that constantly. My Powerwall gateway is in the garage and evidently on the edge of our WiFi range. It would be solid for days and then with some atmospheric changes (e.g. rain, overcast), stop connecting for hours at a time or would flap every 5 minutes. I gave up trying to keep WiFi working and just hardwired my gateway using an ethernet cable to my LAN switch.  I've had no problems after that.

BuongiornoTexas, 2024-04-11:
I've likewise had issues with wifi, and particularly issues in rainy weather. Wired is not an option for, but I found that setting a static ARP more or less fixed the problem permanently for me. Fix referenced in https://github.com/jasonacox/Powerwall-Dashboard/issues/22#issuecomment-1437664703

mihailvovk, 2024-04-11:
> Hi @mihailvovk - I had that constantly. My Powerwall gateway is in the garage and evidently on the edge of our WiFi range. It would be solid for days and then with some atmospheric changes (e.g. [...]

This is just first time it happened to me after 1.5y. I do my wifi and I am almost certain that wifi is great and I have separate access point in garage. But I am about to swap access points with Aruba decommissioned from workplace and gonna retire my ubiquity. Still thinking that either lightning caused some damage to tesla gateway or to my garage access point

mihailvovk, 2024-04-18:
> I've likewise had issues with wifi, and particularly issues in rainy weather. Wired is not an option for, but I found that setting a static ARP more or less fixed the problem permanently for me. Fix referenced in [#22 (comment)](https://github.com/jasonacox/Powerwall-Dashboard/issues/22#issuecomment-1437664703)

I actually lost two out of 3 access points after last week lightning weather in Houston. Needed to rebuild home wifi on Aruba APs and everythink back to normal. Honestly, - it is quite uncomfortable without proper wifi home.