jasonacox/Powerwall-Dashboard #435: Missing data from specific parts of the dashboard, on one installation only

matellis, 2024-02-23:
**Problem Description**

I have a split system, with 7 batteries divided between two Powerwall 2s. I've setup Powerwall-Dashboard on two separate hosts, once for each side, and some values are missing on one installation: Powerwall Temps, Inverter Power, Voltages & Alerts are completely missing, and House & Grid values are missing from the Frequency and Capacity graphs.

I updated pyPowerwall from 0.6.6 to 0.7.7 and some data began to appear:  Charge and Frequency data for the Powerwalls themselves, but House and Grid values are still missing.

I've tried looking in the docker logs, but no errors there. The systems were, as far as I can tell, setup following the same instructions.

Separately, I presume data for the strings is missing as it isn't available on a Powerwall 2, even with a Tesla inverter? See #320

See screencap showing on the left the dashboard missing data, and on the right the one with the data.

[image]

**System Details**
* Hardware: Raspberry Pi 4Bs
* OS Version: Dietpi 9.1.1
* Firmware on each Powerwall 2 is 23.44.0, pyPowerwall is 0.7.7

**Verify Tool Output**
```
./verify.sh
Verify Powerwall-Dashboard 4.0.2 on Linux - Timezone: America/Los_Angeles
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.7.7 Proxy t40
 - Powerwall State: CONNECTED - Firmware Version: 23.44.0
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 13.94}
 - Version: 0.2.3

All tests succeeded.
```

jasonacox, 2024-02-24:
Hi @matellis !  Thanks for posting and for sharing!

> Firmware on each Powerwall 2 is 23.44.0

Unfortunately, Tesla has removed the device vitals API with this Firmware version (see https://github.com/jasonacox/Powerwall-Dashboard/discussions/402) which means that you are no longer able to get several of these extended data points (island voltage power for grid, home, freq, as well as temp, string and alerts data).

You show that you are still getting this data on one of your devices. That seems odd. Are you sure you are seeing the same firmware on both pypowerwall hosts?  

> I updated pyPowerwall from 0.6.6 to 0.7.7 and some data began to appear

Yes, when this firmware first started to appear, we quickly found an alternative API source to get some of this data. As you noted, it doesn't have all of it. I'm hopeful that we will see some of the API come back with the next firmware release. We are also exploring other options (e.g. [this](https://github.com/jasonacox/Powerwall-Dashboard/discussions/392)) but those options are more risky (could change) and difficult to setup.

matellis, 2024-02-25:
That is unusual, that I'm still getting the data. I've double checked and both Powerwall 2s are on the same 23.44.0 firmware. I've installed the command line API I found at [vloschiavo/powerwall2](https://github.com/vloschiavo/powerwall2). Would you like me to run a few API calls on each one if it might help us get this data back for others?

jasonacox, 2024-02-25:
Actually, can you check the vitals URL for each of the Powerwalls?  You should be able to hit http://localhost:8675/vitals (change "localhost" to the IP address of the host running pypowerwall for each of your systems). It is a large payload, so you don't need to post it. I'm just curious to see if you get the larger payload or the small (or empty) one. With Firmware 23.44.0, others are reporting that it is gone. Ironically, my system is still on 23.36.4 which includes vitals.  

FYI, if you go to http://localhost:8675/version - that will show the Firmware version reported to pypowerwall.

matellis, 2024-02-25:
I get `TIMEOUT!%` for each one. Looking at the graph, I suspect grafana is showing the legend but no data, and for some reason the other unit *never* got any vitals data, for a different reason?

closed 2024-02-25 by matellis

jasonacox, 2024-02-25:
Thanks @matellis  - I was hopeful that we had a case where the Vitals persisted even after the Firmware upgrade. I guess not. 😞

Thanks for helping confirm. Eventually we need to remove the non-functioning panels or stash them into a separate grouping while we figure out how to recover some of the vitals.