jasonacox/Powerwall-Dashboard #437: No data with Neo Gateway

Nordictrack, 2024-02-25:
**Problem Description**
No data in any dashboard. All null. No data in db.

**System Details**
Describe the host system you are trying to use for the Dashboard:
* SolarCity Neo Gateway
* VMWare 8.0, Ubuntu 22.04.4 LTS

**Verify Tool Output**

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.7.7 Proxy t40
 - Powerwall State: CONNECTED - Firmware Version:
 - Cloud Mode: YES - Site ID: 2640038 (Vault 21)

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

(Note, I have a separate grafana instance and don't use the other containers, all data is feeding properly to my central influxdb)

I also keep seeing this in the pypowerwall container logs

2024-02-25T19:08:25.097889627Z 02/25/2024 11:08:25 AM [pypowerwall.cloud] [ERROR] Failed to retrieve ENERGY_SITE_BACKUP_TIME_REMAINING - HTTPError('400 Client Error: https://powergate.prd.sn.tesla.services:443/api/v4/energy_site/backup_time_remaining => {:Message=>"Error getting backup time remaining", :Status=>400} for url: https://owner-api.teslamotors.com/api/1/energy_sites/2640038/backup_time_remaining?language=en')

Telegraf shows no errors and pypowerwall shows this:
pypowerwall | 0.7.7 Proxy t40
gets | 782
errors | 0
timeout | 2
ts | 1708888639
start | 1708888087
clear | 1708888087
uptime | 0:09:12
mem | 35356
site_name | Vault 21
cloudmode | True
siteid | 2640038
counter | 63
authmode | cookie
URI: /aggregates | 112
URI: /soe | 110
URI: /temps/pw | 110
URI: /freq | 110
URI: /strings | 110
URI: /alerts/pw | 110
URI: /pod | 110
URI: /stats | 6

I know the neo sends data because my Tidbyt shows the same data as the tesla app ( at this time: 4kW gen, 2.9kW to home, 1.1kW back to the grid)

Can't think of where I didn't configure something as it's working, just not getting anything.

jasonacox, 2024-02-25:
First let's make sure pypowerwall is getting data from the cloud:

Go to http://localhost:8675/example.html (change localhost to the address of your dashboard, as you did to get stats above) - it should load the power flow animation with power data, etc.

> Powerwall State: CONNECTED - Firmware Version:

It is odd that the Firmware version is missing. As with the error message you posted, it seems you can get some of the APIs but not all of them? Try these different URI's on pypowerwall directly to see which load, like:

http://localhost:8675/aggregates

and these others as you saw ins tats:

URI: /soe | 110
URI: /temps/pw | 110
URI: /freq | 110
URI: /strings | 110
URI: /alerts/pw | 110
URI: /pod | 110
URI: /stats | 6

Nordictrack, 2024-02-25:
Example.html

[image]

SOE:

`{
    "percentage": 5
}`

TEMPS:
Blank

Freq:
`{
    "grid_status": 0
}`

Strings:
Blank

Alerts/pw
`{
    "UnscheduledIslandContactorOpen": 1
}`

Pod
`{
    "backup_reserve_percent": 0,
    "nominal_full_pack_energy": null,
    "nominal_energy_remaining": null,
    "time_remaining_hours": null
}`

Stats
`{
    "pypowerwall": "0.7.7 Proxy t40",
    "gets": 18198,
    "errors": 0,
    "timeout": 42,
    "uri": {
        "/aggregates": 2598,
        "/soe": 2596,
        "/temps/pw": 2596,
        "/freq": 2596,
        "/strings": 2597,
        "/alerts/pw": 2596,
        "/pod": 2597,
        "/stats": 7,
        "/help": 1
    },
    "ts": 1708901064,
    "start": 1708888087,
    "clear": 1708888087,
    "uptime": "3:36:17",
    "mem": 35452,
    "site_name": "Vault 21",
    "cloudmode": true,
    "siteid": 2640038,
    "counter": 58,
    "authmode": "cookie"
}`

jasonacox, 2024-02-25:
Ok, I'm sorry, it took me a while to clue in that you do not have a Powerwall. This is a Solar only system, correct?  The data shown is correct for that since those APIs are about Powerwall metrics. 

Can you please test http://localhost:8675/aggregates and http://localhost:8675/csv ?

The "broken" image for the powerflow animation will show up if a browser refuses to render an IFrame.  You should see the animation outside an iFrame using http://localhost:8675/ by itself.

Nordictrack, 2024-02-25:
Solar Only, correct:

{
    "site": {
        "last_communication_time": "2024-02-25T14:57:52-08:00",
        "instant_power": 560,
        "instant_reactive_power": 0,
        "instant_apparent_power": 0,
        "frequency": 0,
        "energy_exported": 0,
        "energy_imported": 0,
        "instant_average_voltage": 0,
        "instant_average_current": 0,
        "i_a_current": 0,
        "i_b_current": 0,
        "i_c_current": 0,
        "last_phase_voltage_communication_time": "0001-01-01T00:00:00Z",
        "last_phase_power_communication_time": "0001-01-01T00:00:00Z",
        "last_phase_energy_communication_time": "0001-01-01T00:00:00Z",
        "timeout": 1500000000,
        "num_meters_aggregated": 1,
        "instant_total_current": null
    },
    "battery": {
        "last_communication_time": "2024-02-25T14:57:52-08:00",
        "instant_power": null,
        "instant_reactive_power": 0,
        "instant_apparent_power": 0,
        "frequency": 0,
        "energy_exported": 0,
        "energy_imported": 0,
        "instant_average_voltage": 0,
        "instant_average_current": 0,
        "i_a_current": 0,
        "i_b_current": 0,
        "i_c_current": 0,
        "last_phase_voltage_communication_time": "0001-01-01T00:00:00Z",
        "last_phase_power_communication_time": "0001-01-01T00:00:00Z",
        "last_phase_energy_communication_time": "0001-01-01T00:00:00Z",
        "timeout": 1500000000,
        "num_meters_aggregated": null,
        "instant_total_current": 0
    },
    "load": {
        "last_communication_time": "2024-02-25T14:57:52-08:00",
        "instant_power": 3040,
        "instant_reactive_power": 0,
        "instant_apparent_power": 0,
        "frequency": 0,
        "energy_exported": 0,
        "energy_imported": 0,
        "instant_average_voltage": 0,
        "instant_average_current": 0,
        "i_a_current": 0,
        "i_b_current": 0,
        "i_c_current": 0,
        "last_phase_voltage_communication_time": "0001-01-01T00:00:00Z",
        "last_phase_power_communication_time": "0001-01-01T00:00:00Z",
        "last_phase_energy_communication_time": "0001-01-01T00:00:00Z",
        "timeout": 1500000000,
        "instant_total_current": 0
    },
    "solar": {
        "last_communication_time": "2024-02-25T14:57:52-08:00",
        "instant_power": 2480,
        "instant_reactive_power": 0,
        "instant_apparent_power": 0,
        "frequency": 0,
        "energy_exported": 0,
        "energy_imported": 0,
        "instant_average_voltage": 0,
        "instant_average_current": 0,
        "i_a_current": 0,
        "i_b_current": 0,
        "i_c_current": 0,
        "last_phase_voltage_communication_time": "0001-01-01T00:00:00Z",
        "last_phase_power_communication_time": "0001-01-01T00:00:00Z",
        "last_phase_energy_communication_time": "0001-01-01T00:00:00Z",
        "timeout": 1000000000,
        "num_meters_aggregated": 1,
        "instant_total_current": 0
    }
}

jasonacox, 2024-02-25:
>"site": {
>"last_communication_time": "2024-02-25T14:57:52-08:00",
>"instant_power": 560,

The good news is that you are getting power data! I assume http://10.1.200.19:8675/csv also works? 

This could be a bug in the code where it is throwing an uncaught exception.  The page not loading is puzzling.  Can you run `curl http://10.1.200.19:8675/` (assuming that is the host running pypowerwall)? I want to see if anything is coming back.

I could also use your help to find out what is blowing up inside of pypowerwall during those renders. Can you edit the `pypowerwall.env` file and turn on debug mode?

```
PW_DEBUG=yes
```

Then `docker restart pypowerwall` and `docker logs pypowerwall -f` and capture some of the log output for us to analyze.  

Thanks for your help!

Nordictrack, 2024-02-25:
I missed the screenshot for csv, but nothing returns
Curl to 'http://10.1.200.19:8675/'

'curl: (52) Empty reply from server'

Debug:

----------------------------------------
Exception occurred during processing of request from ('10.1.100.9', 61562)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/local/lib/python3.10/http/server.py", line 433, in handle
    self.handle_one_request()
  File "/usr/local/lib/python3.10/http/server.py", line 421, in handle_one_request
    method()
  File "/app/server.py", line 438, in do_GET
    fcontent = fcontent.replace("{VERSION}", status["version"])
TypeError: replace() argument 2 must be str, not None
----------------------------------------
----------------------------------------
Exception occurred during processing of request from ('10.1.100.9', 61563)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/local/lib/python3.10/http/server.py", line 433, in handle
    self.handle_one_request()
  File "/usr/local/lib/python3.10/http/server.py", line 421, in handle_one_request
    method()
  File "/app/server.py", line 438, in do_GET
    fcontent = fcontent.replace("{VERSION}", status["version"])
TypeError: replace() argument 2 must be str, not None
----------------------------------------
02/25/2024 03:18:31 PM [pypowerwall.cloud] [ERROR] Failed to retrieve ENERGY_SITE_BACKUP_TIME_REMAINING - HTTPError('400 Client Error: https://powergate.prd.sn.tesla.services:443/api/v4/energy_site/backup_time_remaining => {:Message=>"Error getting backup time remaining", :Status=>400} for url: https://owner-api.teslamotors.com/api/1/energy_sites/2640038/backup_time_remaining?language=en')
----------------------------------------
Exception occurred during processing of request from ('10.1.100.9', 61571)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/local/lib/python3.10/http/server.py", line 433, in handle
    self.handle_one_request()
  File "/usr/local/lib/python3.10/http/server.py", line 421, in handle_one_request
    method()
  File "/app/server.py", line 438, in do_GET
    fcontent = fcontent.replace("{VERSION}", status["version"])
TypeError: replace() argument 2 must be str, not None
----------------------------------------
02/25/2024 03:18:35 PM [pypowerwall.cloud] [ERROR] Failed to retrieve ENERGY_SITE_BACKUP_TIME_REMAINING - HTTPError('400 Client Error: https://powergate.prd.sn.tesla.services:443/api/v4/energy_site/backup_time_remaining => {:Message=>"Error getting backup time remaining", :Status=>400} for url: https://owner-api.teslamotors.com/api/1/energy_sites/2640038/backup_time_remaining?language=en')
----------------------------------------
Exception occurred during processing of request from ('10.1.100.9', 61575)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/local/lib/python3.10/http/server.py", line 433, in handle
    self.handle_one_request()
  File "/usr/local/lib/python3.10/http/server.py", line 421, in handle_one_request
    method()
  File "/app/server.py", line 208, in do_GET
    message = "%0.2f,%0.2f,%0.2f,%0.2f,%0.2f\n" \
TypeError: must be real number, not NoneType
----------------------------------------
----------------------------------------
Exception occurred during processing of request from ('10.1.100.9', 61578)
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/socketserver.py", line 683, in process_request_thread
    self.finish_request(request, client_address)
  File "/usr/local/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/local/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/local/lib/python3.10/http/server.py", line 433, in handle
    self.handle_one_request()
  File "/usr/local/lib/python3.10/http/server.py", line 421, in handle_one_request
    method()
  File "/app/server.py", line 208, in do_GET
    message = "%0.2f,%0.2f,%0.2f,%0.2f,%0.2f\n" \
TypeError: must be real number, not NoneType
----------------------------------------
02/25/2024 03:18:40 PM [pypowerwall.cloud] [ERROR] Failed to retrieve ENERGY_SITE_BACKUP_TIME_REMAINING - HTTPError('400 Client Error: https://powergate.prd.sn.tesla.services:443/api/v4/energy_site/backup_time_remaining => {:Message=>"Error getting backup time remaining", :Status=>400} for url: https://owner-api.teslamotors.com/api/1/energy_sites/2640038/backup_time_remaining?language=en')

Nordictrack, 2024-02-25:
Here is startup logs:

2024-02-25T23:39:44.341130532Z 02/25/2024 03:39:44 PM [proxy] [INFO] pyPowerwall Proxy Server - Cloud Mode
2024-02-25T23:39:44.341144185Z 02/25/2024 03:39:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/site_info/site_name
2024-02-25T23:39:44.462076044Z 02/25/2024 03:39:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Retrieved SITE_CONFIG data
2024-02-25T23:39:44.462267879Z 02/25/2024 03:39:44 PM [proxy] [INFO] Connected to Site ID 2640038 (Vault 21)
2024-02-25T23:39:45.005520802Z 02/25/2024 03:39:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /alerts/pw HTTP/1.1" 200 -
2024-02-25T23:39:45.005634868Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /vitals
2024-02-25T23:39:45.005786037Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-25T23:39:45.008962075Z 02/25/2024 03:39:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /strings HTTP/1.1" 200 -
2024-02-25T23:39:45.009000960Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /vitals
2024-02-25T23:39:45.009196077Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-25T23:39:45.011816770Z 02/25/2024 03:39:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /temps/pw HTTP/1.1" 200 -
2024-02-25T23:39:45.012382447Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /vitals
2024-02-25T23:39:45.012411465Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-25T23:39:45.013067545Z 02/25/2024 03:39:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /soe HTTP/1.1" 200 -
2024-02-25T23:39:45.013158113Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/system_status/soe
2024-02-25T23:39:45.014555682Z 02/25/2024 03:39:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /freq HTTP/1.1" 200 -
2024-02-25T23:39:45.014789842Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/system_status
2024-02-25T23:39:45.016392529Z 02/25/2024 03:39:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /pod HTTP/1.1" 200 -
2024-02-25T23:39:45.016412190Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/system_status
2024-02-25T23:39:45.016984436Z 02/25/2024 03:39:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /aggregates HTTP/1.1" 200 -
2024-02-25T23:39:45.017305173Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/meters/aggregates
2024-02-25T23:39:45.017664297Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-25T23:39:45.177195811Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Retrieved SITE_DATA data
2024-02-25T23:39:45.210043042Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-25T23:39:45.212954106Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-25T23:39:45.216313774Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-25T23:39:45.216343117Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-25T23:39:45.216675340Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-25T23:39:45.216690848Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-25T23:39:45.217772644Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-25T23:39:45.351444989Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Retrieved SITE_SUMMARY data
2024-02-25T23:39:45.417368835Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_SUMMARY data
2024-02-25T23:39:45.418025846Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_SUMMARY data
2024-02-25T23:39:45.418332478Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /vitals
2024-02-25T23:39:45.418748958Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /vitals
2024-02-25T23:39:45.419243324Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-25T23:39:45.419459883Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-25T23:39:45.419494046Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-25T23:39:45.419709243Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-25T23:39:45.419939775Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/system_status/grid_status
2024-02-25T23:39:45.420296254Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/operation
2024-02-25T23:39:45.420659567Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-25T23:39:45.420977956Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-25T23:39:45.480051305Z 02/25/2024 03:39:45 PM [pypowerwall.cloud] [ERROR] Failed to retrieve ENERGY_SITE_BACKUP_TIME_REMAINING - HTTPError('400 Client Error: https://powergate.prd.sn.tesla.services:443/api/v4/energy_site/backup_time_remaining => {:Message=>"Error getting backup time remaining", :Status=>400} for url: https://owner-api.teslamotors.com/api/1/energy_sites/2640038/backup_time_remaining?language=en')

jasonacox, 2024-02-25:
Ok, I see the bug in the code. Version is expected but we are getting none back for your installation. Additionally, the csv code is likely getting a null back for battery so I can catch that as well.

I can make the change. You mention you are running a custom setup. I'll upload a test pypowerwall container to try if you are willing.

Nordictrack, 2024-02-26:
Absolutely willing to assist.

jasonacox, 2024-02-26:
Ok, I have a test image that hopefully has the right fixes:  jasonacox/pypowerwall:0.7.7t41

If you are using the `powerwal.yml` file (or simliar) with docker-compose, you can edit that file and change the image for pypowerwall to `jasonacox/pypowerwall:0.7.7t41` and then run `./compose-dash.sh up -d` or equivalent.

Nordictrack, 2024-02-26:
So far so good:
http://10.1.200.19:8675/example.html shows:
[image]
As does the root

CSV output:
2520.00,3000.00,480.00,0.00,5.00

Debug output:

2024-02-26T01:22:43.194570281Z 02/25/2024 05:22:43 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-26T01:22:44.211838380Z 02/25/2024 05:22:44 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/sitemaster HTTP/1.1" 200 -
2024-02-26T01:22:44.211918410Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/sitemaster
2024-02-26T01:22:44.214535805Z 02/25/2024 05:22:44 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/meters/aggregates HTTP/1.1" 200 -
2024-02-26T01:22:44.214575120Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/meters/aggregates
2024-02-26T01:22:44.214588036Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-26T01:22:44.214599035Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-26T01:22:44.215134053Z 02/25/2024 05:22:44 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/system_status/soe HTTP/1.1" 200 -
2024-02-26T01:22:44.215441707Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/system_status/soe
2024-02-26T01:22:44.219823143Z 02/25/2024 05:22:44 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/system_status/grid_status HTTP/1.1" 200 -
2024-02-26T01:22:44.221788054Z 02/25/2024 05:22:44 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/powerwalls HTTP/1.1" 200 -
2024-02-26T01:22:44.222275770Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/system_status/grid_status
2024-02-26T01:22:44.222496484Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/powerwalls
2024-02-26T01:22:44.223256732Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-26T01:22:44.306983978Z 02/25/2024 05:22:44 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/sitemaster HTTP/1.1" 200 -
2024-02-26T01:22:44.307135346Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/sitemaster
2024-02-26T01:22:44.308417533Z 02/25/2024 05:22:44 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/meters/aggregates HTTP/1.1" 200 -
2024-02-26T01:22:44.308498410Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/meters/aggregates
2024-02-26T01:22:44.308516270Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-26T01:22:44.308519511Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-26T01:22:44.310877022Z 02/25/2024 05:22:44 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/system_status/soe HTTP/1.1" 200 -
2024-02-26T01:22:44.310944314Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/system_status/soe
2024-02-26T01:22:44.311146770Z 02/25/2024 05:22:44 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/system_status/grid_status HTTP/1.1" 200 -
2024-02-26T01:22:44.311452649Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/system_status/grid_status
2024-02-26T01:22:44.311520696Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-26T01:22:44.315074956Z 02/25/2024 05:22:44 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/powerwalls HTTP/1.1" 200 -
2024-02-26T01:22:44.315357761Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/powerwalls
2024-02-26T01:22:44.431536782Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Retrieved SITE_SUMMARY data
2024-02-26T01:22:44.511843969Z 02/25/2024 05:22:44 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_SUMMARY data
2024-02-26T01:22:45.005465168Z 02/25/2024 05:22:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /alerts/pw HTTP/1.1" 200 -
2024-02-26T01:22:45.005724828Z 02/25/2024 05:22:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /vitals
2024-02-26T01:22:45.006867134Z 02/25/2024 05:22:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-26T01:22:45.006901680Z 02/25/2024 05:22:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-26T01:22:45.007771608Z 02/25/2024 05:22:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /temps/pw HTTP/1.1" 200 -
2024-02-26T01:22:45.008024360Z 02/25/2024 05:22:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /vitals
2024-02-26T01:22:45.008254351Z 02/25/2024 05:22:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_CONFIG data
2024-02-26T01:22:45.008359936Z 02/25/2024 05:22:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-26T01:22:45.009697458Z 02/25/2024 05:22:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /soe HTTP/1.1" 200 -
2024-02-26T01:22:45.010423478Z 02/25/2024 05:22:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/system_status/soe
2024-02-26T01:22:45.010459265Z 02/25/2024 05:22:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_SUMMARY data
2024-02-26T01:22:45.011969765Z 02/25/2024 05:22:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /pod HTTP/1.1" 200 -
2024-02-26T01:22:45.012261248Z 02/25/2024 05:22:45 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/system_status
2024-02-26T01:22:45.014907817Z 02/25/2024 05:22:45 PM [proxy] [DEBUG] 172.18.0.4 "GET /strings HTTP/1.1" 200 -
[... cut ...]
2024-02-26T01:22:47.028377698Z 02/25/2024 05:22:47 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-26T01:22:47.151187036Z 02/25/2024 05:22:47 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Returning cached SITE_DATA data
2024-02-26T01:22:48.145728274Z 02/25/2024 05:22:48 PM [proxy] [DEBUG] 10.1.100.9 "GET /api/troubleshooting/problems HTTP/1.1" 200 -
2024-02-26T01:22:48.145769293Z 02/25/2024 05:22:48 PM [pypowerwall.cloud] [DEBUG]  -- cloud: Request for /api/troubleshooting/problems

Waiting on dashboard data to populate

jasonacox, 2024-02-26:
> Waiting on dashboard data to populate

How is looking? Is telegraf transferring the data?

Nordictrack, 2024-02-26:
Upon checking, it looks like the following is populating:

[image]

I'm sure I'm reading a metric wrong. When I'm back in front of my PC, I'll output the tables telegraf made. But the data is coming in.

Nordictrack, 2024-02-26:
Happy to report no issues. Here's a sample (from an overview dash I'm making)
[image]

jasonacox, 2024-02-27:
That's great news @Nordictrack !  Thanks for confirming and nice dashboard!! 

If you don't mind helping, you may not want to add the power flow animation to your dashboard (you could if you wanted) but could you check it as well? URL http://10.1.200.19:8675/example.html ? The one you posted earlier showed "grid offline" and numbers that didn't add up.  I would like to confirm if we still have an issue with some of the data we are creating for the animation. 

[image]

Thanks for your help!

Nordictrack, 2024-02-27:
I did notice yesterday that the grid shows no data on that, but the load_instant_power data does have the correct number when I compare it to the Tesla & Tidbyt app. Right now, it's showing this when it should be pulling from the grid:

[image]
[image]

jasonacox, 2024-02-28:
Thanks @Nordictrack ! This is helpful.  I believe I have the wrong mapping on the grid status for Solar only systems.

 Can you provide the results of these calls?  Replace localhost with the host running pypowerwall:

http://localhost:8675/api/site_info
http://localhost:8675/api/sitemaster
http://localhost:8675/api/system_status/soe
http://localhost:8675/api/system_status/grid_status
http://localhost:8675/api/powerwalls
http://localhost:8675/api/troubleshooting/problems

I'm also going to see if I can either remove the Powerwall icon in the power flow animation, or at least remove the kW and % values.

Nordictrack, 2024-02-28:
Site Info
{
    "max_system_energy_kWh": 0,
    "max_system_power_kW": 0,
    "site_name": "Vault 21",
    "timezone": null,
    "max_site_meter_power_kW": null,
    "min_site_meter_power_kW": null,
    "nominal_system_energy_kWh": 0,
    "nominal_system_power_kW": 0,
    "panel_max_current": null,
    "grid_code": {
        "grid_code": null,
        "grid_voltage_setting": null,
        "grid_freq_setting": null,
        "grid_phase_setting": null,
        "country": null,
        "state": null,
        "utility": "NV Energy"
    }
}

Sitemaster
{
    "status": "StatusUp",
    "running": true,
    "connected_to_tesla": true,
    "power_supply_mode": false,
    "can_reboot": "Yes"
}

system_status/soe
{
    "percentage": 0
}

system_status/grid_status
{
    "grid_status": "SystemIslandedActive",
    "grid_services_active": false
}

powerwalls
{
    "enumerating": false,
    "updating": false,
    "checking_if_offgrid": false,
    "running_phase_detection": false,
    "phase_detection_last_error": "no phase information",
    "bubble_shedding": false,
    "on_grid_check_error": "on grid check not run",
    "grid_qualifying": false,
    "grid_code_validating": false,
    "phase_detection_not_available": true,
    "powerwalls": [
        {
            "Type": "",
            "PackagePartNumber": "2012170-25-E",
            "PackageSerialNumber": "TG1234567890G1",
            "type": "SolarPowerwall",
            "grid_state": "Grid_Uncompliant",
            "grid_reconnection_time_seconds": 0,
            "under_phase_detection": false,
            "updating": false,
            "commissioning_diagnostic": {
                "name": "Commissioning",
                "category": "InternalComms",
                "disruptive": false,
                "inputs": null,
                "checks": [
[... 291 lines cut ...]
                    "debug": null,
                    "checks": null
                },
                {
                    "name": "Grid code",
                    "status": "not_run",
                    "start_time": null,
                    "end_time": null,
                    "progress": 0,
                    "results": null,
                    "debug": null,
                    "checks": null
                }
            ],
            "alert": false
        }
    },
    "msa": null,
    "states": null
}

troubleshooting/problems
{
    "problems": []
}

jasonacox, 2024-02-28:
Thanks!  This is a bug - I'll work on a fix...

> system_status/grid_status
```json
{
    "grid_status": "SystemIslandedActive",
    "grid_services_active": false
}
```

Nordictrack, 2024-02-28:
If it's worth anything, this is a pre-tesla solarcity install (tesla installed it but gave me solar city gear) and those use a "power blaster" to sense grid usage.

jasonacox, 2024-03-03:
Hi @Nordictrack - I made an update to better handles the grid status data for cloud-mode configurations.  If you update, it would be interesting to see if the power flow animation now shows the grid data correctly.

If you are using the powerwall.yml file (or simliar) with docker-compose, you can edit that file and change the image for pypowerwall to `jasonacox/pypowerwall:0.7.9t41` and then run `./compose-dash.sh up -d` or equivalent.

Nordictrack, 2024-03-03:
Success on the Power flow!

[image]

[image]

jasonacox, 2024-03-03:
That's great news!!  Thanks for the help. I think we could create a `PW_STYLE` file to remove the Powerwall icon and data so you only have grid, solar and home as part of the animation if that sounds interesting.

Something like...

[image]

Also, nice looking dashboard! Thanks for sharing that.

jasonacox, 2024-03-04:
Ok, I have another update.  I created a solar.js file to update the animation for solar-only:

[image]

How to set it up:
* Edit `powerwall.yml` file (or simliar) and change the image for pypowerwall to `jasonacox/pypowerwall:0.7.9t42`
* Edit `pypowerwall.env` and edit the style line to be `PW_STYLE=solar`
* Run `./compose-dash.sh up -d` or equivalent.

Nordictrack, 2024-03-05:
Test was successful.

[image]

jasonacox, 2024-03-06:
That's great! If you ever build a dashboard that includes that, I would love to see it. 😁 

Thank for all your help. This means that any Tesla Solar-only customers can use our dashboards, including the power flow animation.