jasonacox/Powerwall-Dashboard #446: Dashboard appears unable to communicate with Gateway

GeneralSprint, 2024-03-10:
**Problem**
I woke up this morning to find that my graphs had all stopped updating as of 11:26pm last night (GMT). I checked for updates to the dashboard and found one had been published not one hour ago, figured someone else must already had had the same issue and fixed it, so i updated. Once updated, the live power flow would still not load(times out eventually), I attempted to connect directly to my gateway (I checked the Telsa app was working which it was), and found i couldn't even open the webpage to the local IP. I bounced the network port, and now the page does load, it lets me log in (once i get pass the "would you like to upgrade to pro installer", and I can see the powerflow directly on the gateway, but the dashboard still doesn't seem to be able to talk to it and retrieve any data... Firmware is showing as 23.44.0 eb113390.

[image]

[image]

**Host System**
 - Runs as a docker in Ubuntu in Proxmox
 - Everything is wired via ethernet. 

Not sure what other information would be useful, happy to provide, just let me know :)

GeneralSprint, 2024-03-10:
Decided todo some patching, and patched my docker server, after a reboot, everything came back... doesn't explain the powerwall going off-network. Was tempted to delete but decided to leave this here incase it proves useful to anyone else... Reboot all the things! 

Sprint

closed 2024-03-10 by GeneralSprint

jasonacox, 2024-03-10:
> Firmware is showing as 23.44.0 eb113390.

This Firmware version removes the /vitals endpoint. The Powerwall-Dashboard versions prior to v3.0.8 (see [here](https://github.com/jasonacox/Powerwall-Dashboard/releases/tag/v3.0.8)) would not handle this well would continually try to connect to that API until the Powerwall would rate-limit BLOCK connections. I suspect you may have been on an older version. The only fix is to upgrade and wait 10m or so for the rate-limit block to be auto-removed by the Powerwall.