jasonacox/Powerwall-Dashboard #448: New Firmware 24.4.0

simonrb2000, 2024-03-15:
**Problem**
Received new firmware today, 23.4.0 0fe780c9

**To Reproduce**
Doesn't connect

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Host System**
 - Raspberry Pi

**Additional context**
I have redone the setup, I use the local control. I can log into the gateway using its IP Address however I can't get this to log in. I'll keep an eye on it and was just getting this out there as I haven't seen any mention of 24.4.0 firmware anywhere on the internet.

closed 2024-03-15 by simonrb2000

jasonacox, 2024-03-16:
Hi @simonrb2000  -  Thanks for reporting this. I will add to our Firmware discussion https://github.com/jasonacox/Powerwall-Dashboard/discussions/109

Would you be able to perform some test for us about this new firmware?  It's good that pypowerwall is still able to connect and put data. Do you see device vital data?  Try: http://localhost:8675/vitals or http://localhost:8675/alerts.

Any odd behavior or anything missing from the Dashboard?

longzheng, 2024-03-16:
> Do you see device vital data? Try: http://localhost:8675/vitals or http://localhost:8675/alerts.

I've also been updated to 24.4.0.

Nothing in `/vitals` or `/alerts` unfortunately.

jasonacox, 2024-03-16:
Thanks @longzheng  - Have you spotted anything new with this version?

I'm still on 23.36.4 which I'm fine with since vitals still works. :)  But it does make it more difficult for me to explore options for data.

longzheng, 2024-03-16:
I had a quick run through the various API endpoints and they looked the same.

I didn't get a chance to reverse engineer any of the JavaScript.

jasonacox, 2024-03-17:
Thanks @longzheng .

I'm going to proceed with the expectation that vitals is not going to return. I've updated the latest pypowerwall to better handle that condition and stop polling for vitals if it detects the latest Firmware upgrades. That should help with some odd errors and timeout/rate limits people are seeing (see #449 ). I'll also work on adding the Tesla Pros API support to pypowerwall to allow people to use that to get vitals data. It is a more complex setup but seems to be working and would be a viable option for those who choose that adventure. 😉

simonrb2000, 2024-03-17:
Hey,

Yes as @longzheng has described. Everything has been the same as previous firmware 23.44. I haven’t seen anything added or removed so far.

Happy to test the Tesla Pro API if that helps, sounds interesting. Would be nice to get temps back.

Cheers

Simon

jasonacox, 2024-03-17:
@simonrb2000 - Sadly, we still haven't found the temps. The Tesla Pros API doesn't have it either unfortunately. However, I'm also investigating the API at http://localhost8675/api/solar_powerwall - It has string data, and is available on the standard local-network APIs but seems to be reporting only a single inverter. I only have one inverter so I'm limited on what I can explore. If anyone finds a good way to get additional data from other inverters, please let me know.