jasonacox/Powerwall-Dashboard #451: Local setup does not pull up any data

DOCNY, 2024-03-16:
I am not sure if this is a new change from Tesla but I had local LAN setup and working fine but now it does not pull any data after the last update. Previously I lost string data. Nothing changed in setup on my end. 

If I change to cloud setup the data is enabled again but no historic data.

jasonacox, 2024-03-17:
Hi @DOCNY 

> does not pull any data after the last update

Did it stop working immediately after the update?

Also, can share the logs you see in `docker logs pypowerwall`?  Also run `verify.sh` to see if everything shows setup correctly.

DOCNY, 2024-03-17:
Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): ERROR: Not Listening
 - Version: Unknown
 - Powerwall State: ERROR: Not Connected - Firmware Version: Unknown
 - Cloud Mode: NO

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 2.89}
 - Version: 0.2.3

One or more tests failed.

------------------------

2024/03/16 22:29:49 | stderr | 03/16/2024 10:29:49 PM [proxy] [ERROR] Fatal Error: Unable to connect. Please fix config and restart.
-- | -- | --
2024/03/16 22:29:49 | stderr | 03/16/2024 10:29:49 PM [proxy] [ERROR] Unable to connect to Powerwall at https://192.168..x.x
2024/03/16 22:29:44 | stderr | 03/16/2024 10:29:44 PM [proxy] [INFO] pyPowerwall Proxy Started
2024/03/16 22:29:44 | stderr | 03/16/2024 10:29:44 PM [proxy] [INFO] pyPowerwall [0.7.9] Proxy Server [t42] - HTTP Port 8675

DOCNY, 2024-03-17:
Running as cloud

Verify Powerwall-Dashboard 4.1.1 on Linux - Timezone: America/New_York
----------------------------------------------------------------------------
This script will attempt to verify all the services needed to run
Powerwall-Dashboard. Use this output when you open an issue for help:
https://github.com/jasonacox/Powerwall-Dashboard/issues/new

Checking pypowerwall
----------------------------------------------------------------------------
 - Config File pypowerwall.env: GOOD
 - Container (pypowerwall): GOOD
 - Service (port 8675): GOOD
 - Version: 0.7.9 Proxy t42
 - Powerwall State: CONNECTED - Firmware Version: 23.44.0
 - Cloud Mode: YES - Site ID: 2252238472343092 (765 Ewing Ave)

Checking telegraf
----------------------------------------------------------------------------
 - Config File telegraf.conf: GOOD
 - Local Config File telegraf.local: GOOD
 - Container (telegraf): GOOD
 - Version: Telegraf 1.28.2 (git: HEAD@8d9cf395)

Checking influxdb
----------------------------------------------------------------------------
 - Config File influxdb.conf: GOOD
 - Environment File influxdb.env: GOOD
 - Container (influxdb): GOOD
 - Service (port 8086): GOOD
 - Filesystem (./influxdb): GOOD
 - Version: InfluxDB shell version: 1.8.10

Checking grafana
----------------------------------------------------------------------------
 - Config File grafana.env: GOOD
 - Container (grafana): GOOD
 - Service (port 9000): GOOD
 - Filesystem (./grafana): GOOD
 - Version: Grafana CLI version 9.1.2

Checking weather411
----------------------------------------------------------------------------
 - Container (weather411): GOOD
 - Service (port 8676): GOOD
 - Weather: {"temperature": 47.3}
 - Version: 0.2.3

All tests succeeded.

jasonacox, 2024-03-17:
> Unable to connect to Powerwall at https://192.168..x.x/

This is the core issue. A few thoughts:
* Double check the IP address - `pip3 install pypowerwall; python3 -m pypowerwall scan` can be used to find it.
* Are you able to open the web portal of the Powerwall at https://192.168..x.x/ in a browser?
* Check firewall and network settings on you host running pypowerwall. The above "scan" would help show if it is working.

> CONNECTED - Firmware Version: 23.44.0

It is good you are able to connect to the Tesla Cloud, but this version (23.44.0) should still have local APIs. The key issue seems to be that pypowerwall is not able to connect to the Powerwall.

DOCNY, 2024-03-17:
I can access via local ip and can log in with same credentials. Interestingly, I have an home assistant integration and that is down as well. That's why I think it may Tesla limiting local API etc. I get the following error there

A connection error occurred while connecting to the Powerwall: Powerwall is unreachable: Cannot connect to host 192.168.X.X:443 ssl:False [Connect call failed ('192.168.X.X', 443)]

jasonacox, 2024-03-17:
I recommend you stop the home assistant integration and stop pypowerwall. Wait 10m for the block to expire and try pypowerwall again.

DOCNY, 2024-03-17:
it did not help. 

I am using synology docker for both pypowerwall and home assistant. No change in firewall. DSM 7.2 for at least two months or so. I see data drop from 3/9

jasonacox, 2024-03-17:
Do you have a shell you can run on the synology? If so, you could try to manually connect by running a python script (of course, replace 192.168.x.x with the actual address):

```python
import pypowerwall
pypowerwall.set_debug(True)
pw = pypowerwall.Powerwall("192.168.x.x","password")
print(pw.status())
print(pw.power())
```

Also test network from there from the shell:

```bash
ping 192.168.x.x
curl -k https://192.168.x.x
```

DOCNY, 2024-03-18:
it has error connecting. On home assistant it seems to indicate it has Powerwall Self signed certificate issue.

443 ssl:False [Connect call failed

jasonacox, 2024-03-19:
The Powerwall uses a self-signed cert so that is valid, but it has always been that way.

If you are having troubles connecting using the `ping` and `curl`, that indicates you have a network or host (server) issue. Try rebooting or checking your local network for any issues.

DOCNY, 2024-03-20:
thanks for all the support. it turned out to be a vlan assignment for Synology port on Unifi. Previous versions used to have option of all Vlans and with the current one, that's not an option and if you choose anything other that default it will do that for some reason.

alexmccoy, 2024-04-08:
I am experiencing this issue as well.  The web interface loads, but no longer provides an option to log in.  It only has the teslapros redirect link.  Digging in you can find an "upgrade later" button at the bottom, but I am no longer able to reset the password on the gateway, and receive an error about the switch toggle despite me having just toggled it.  Is this dead forever?

[image]

jasonacox, 2024-04-09:
Hi @alexmccoy 

This may help you navigate past the upgrade screens: https://github.com/jasonacox/Powerwall-Dashboard/discussions/367#discussioncomment-7278760

You should first stop the Dashboard (`docker stop pypowerwall`) in case the Powerwall is restricting access based on rate limits (too many login attempts).  Let it sit for 10 minutes then try to log in to the Powerwall gateway portal again. If it still says your password is bad, try to reset it.