jasonacox/Powerwall-Dashboard #453: Docker Compose version v2.25.0 reports version is obsolete

BJReplay, 2024-03-20:
**Problem**
Docker Compose version 2.25.0 now reports that version is obsolete.

This appears when using the compose-dash.sh helper - for example:
```

./compose-dash.sh down && ./compose-dash.sh up -d
Including powerwall.extend.yml
Running Docker Compose...
WARN[0000] powerwall.yml: `version` is obsolete
[+] Running 10/10
 ✔ Container pwdusage                   Removed                                                                   10.7s
 ✔ Container homeassistant              Removed                                                                    5.1s
 ✔ Container weather411                 Removed                                                                    1.2s
 ✔ Container grafana                    Removed                                                                    1.2s
 ✔ Container ecowitt                    Removed                                                                   10.8s
 ✔ Container telegraf                   Removed                                                                    0.8s
 ✔ Container weatherdesktop             Removed                                                                    3.8s
 ✔ Container pypowerwall                Removed                                                                    0.5s
 ✔ Container influxdb                   Removed                                                                    0.5s
 ✔ Network powerwall-dashboard_default  Removed                                                                    0.3s
Including powerwall.extend.yml
Running Docker Compose...
WARN[0000] powerwall.yml: `version` is obsolete
[+] Running 9/10
 ⠸ Network powerwall-dashboard_default  Created                                                                    3.4s
 ✔ Container homeassistant              Started                                                                    0.8s
 ✔ Container pypowerwall                Started                                                                    1.9s
 ✔ Container influxdb                   Started                                                                    1.9s
 ✔ Container weatherdesktop             Started                                                                    1.4s
 ✔ Container ecowitt                    Started                                                                    2.8s
 ✔ Container weather411                 Started                                                                    3.0s
 ✔ Container grafana                    Started                                                                    3.1s
 ✔ Container pwdusage                   Started                                                                    2.6s
 ✔ Container telegraf                   Started                                                                    2.7s
```

**To Reproduce**
1. Update to the latest versions of docker compose (e.g. sudo apt update && sudo apt dist-upgrade -y)
2. cd Powerwall-Dashboard
3. ./compose-dash.sh down && ./compose-dash.sh up -d
4. See error

**Host System**
 - WSL2 (Ubuntu 22.10) on Windows 11
 - Also reproducible on Ubuntu 22.04 LTS

**Additional context**
Noticed this on all my Docker Compose examples (Powerwall-Dashboard and others) after this update to Docker.

Checking shows that Version has been optional for a long time.

jasonacox, 2024-03-20:
Thanks @BJReplay  - but correct me if I'm wrong, this is a "warning" not an error, or are you seeing this no longer work?

In any case, we can explore removing it. If you remove the `version: "3.5"` line do you see any problems?

BJReplay, 2024-03-20:
Thanks @jasonacox I was updating my issue (as I accidently) submitted before I'd finished while you commented.

Yes, it's just a warning - it still works.

If you remove the version line entirely, the warning goes away.

I'll do a little more research and see if I can find out when it became optional and see if I can work out the earliest version of Docker or Docker Compose that would break if we removed it.  I'll get back here, and if you agree, I can put together a PR to remove it.

BuongiornoTexas, 2024-03-20:
I suspect it will need to stay as long powerwall-dashboard continues to support older versions of compose. I've got a vague memory that version 3.8 of the yaml spec is the last one to need it?

BJReplay, 2024-03-20:
I'm almost certain we're using version 2 format (because it uses the services element).

It's supported by docker 1.10 released Feb 4, 2016.

https://docs.docker.com/compose/intro/history/#compose-file-format-versioning describes the history: 

_Version one of the Docker Compose command-line binary was first released in 2014. It was written in Python, and is invoked with docker-compose. Typically, Compose V1 projects include a top-level_ **version element** _in the compose.yml file, with values ranging from 2.0 to 3.8, which refer to the specific file formats._

_Version two of the Docker Compose command-line binary was announced in 2020, is written in Go, and is invoked with docker compose. Compose V2_ **ignores the version top-level element** _in the compose.yml file._

https://docs.docker.com/compose/compose-file/compose-file-v2/

BJReplay, 2024-03-20:
Note there are a couple of potentially confusing versions here: Versions of Docker Compose (V1 and V2) and versions of the compose file format (V1, V2 and V3).

This is a good jumping off point: https://docs.docker.com/compose/intro/history/

My gut feel is that unless someone is running a very old (pre Compose 1.6.0 in 2016) system, we can remove version.

BJReplay, 2024-03-20:
> I've got a vague memory that version 3.8 of the yaml spec is the last one to need it?

From my reading, not actually needed after 2.0 - was optional from that point on.

BJReplay, 2024-03-20:
I've put together a PR - feel free to accept or reject based on your assessment :)

My take on it is that:

1. version should be removed
2. docker compose (rather than docker-compose) and the set up for docker-compose (rather than the python module - which is V1 docker compose) probably should be part of the main instructions (hence the change to the readme) but I haven't changed anything else.

Ellen-Gu, 2024-03-20:
I encountered this warning today. 
As BJReplay stated, I removed the line from my yml file then the warning disappeared.

The line I commented out is as the following:
#version: "3.8"

Ellen-Gu, 2024-03-20:
> I've put together a PR - feel free to accept or reject based on your assessment :)
>
> My take on it is that:
>
> 1. version should be removed
> 2. docker compose (rather than docker-compose) and the set up [...]

I encountered this warning today.
As BJReplay stated, I removed the line from my yml file then the warning disappeared.

The line I commented out is as the following:
#version: "3.8"

BJReplay, 2024-03-23:
Closing as fix merged.

closed 2024-03-23 by BJReplay