jasonacox/Powerwall-Dashboard #455: tesla-history.py not pulling all data

lukemcguire, 2024-03-21:
**Problem**
When running the [tesla-history import tool](https://github.com/jasonacox/Powerwall-Dashboard/tree/main/tools/tesla-history#tesla-history-import-tool) not all of the data is being imported to the database.  Specifically, the end of each day is missing data.  When running the tool it'll find the data gap but it won't 

**To Reproduce**
Please paste a screenshot or steps to reproduce the behavior:
1. Find time period with data gap (e.g. 2023-06-26 17:00:00 - 2023-06-26 23:59:59)
2. Run command 'python3 tesla-history.py --start "2023-06-26 00:00:00" --end "2023-06-27 23:59:59"'
3. Receive error 'ERROR: No data returned for this date/time range'

Here's the output of the same command with --debug flag - [debug.txt](https://github.com/jasonacox/Powerwall-Dashboard/files/14704381/debug.txt)

**Screenshots**
[image]

**Host System**
 - Hardware - Synology DS920+
 - OS - DSM 7.2.1-69057 Update 3
 - Powerwall-Dashboard stack is running in Container Manager (Synology's docker tool)

**Additional context**
I think it may be something to do with time zones since it appears to not pull the last 7 hours of each day and I'm in tz America/Los_Angeles.  Looking through the debug output it seems that it finds the gap but when it tries to pull the data it's pulling from the wrong time window.

jasonacox, 2024-03-22:
Hi @lukemcguire  - Does the host your are running have your timezone set correctly? There is code in the script to do the timezone calculation but it does use your localhost settings. It should be something like:

```bash
$ timedatectl
                      Local time: Thu 2024-03-21 19:48:56 PDT
                  Universal time: Fri 2024-03-22 02:48:56 UTC
                        RTC time: Fri 2024-03-22 02:48:56
                       Time zone: America/Los_Angeles (PDT, -0700)
```

lukemcguire, 2024-03-22:
Well, add `timedatectl` to the list of useful things that Synology neglected to include in their distro.  Anyways...

```shell
$ date ; date -u
Fri Mar 22 09:26:35 AM PDT 2024
Fri Mar 22 04:26:35 PM UTC 20244
``` 
[image]

So that looks like the system time is ok. Is it possible that one of the containers isn't in the right time zone?  I don't know if that's being handled in the script or not.

Looking at the screenshot I provided, the data gap would be from 2023-06-26 17:00-23:59 PDT. However, based on the debug output that I provided, it says `* Found data gap: [2023-06-26 00:00:00+00:00] - [2023-06-26 07:00:00+00:00] (7:00:00s)`. This wouldn't seem to be a timezone related issue as the amount of the time error isn't a factor of 7.  Maybe I'm grasping at straws here.  

Is there something else I can check?

lukemcguire, 2024-03-22:
I guess I should also note that I've run the weather-history tool and it is working just fine and is not returning gaps.

jasonacox, 2024-03-23:
These aren't running in containers, just scripts on your localhost.  The weather history tool isn't using the Tesla Cloud API so it wouldn't be impacted by the timezone translation issues.  One final thought, can you confirm your Tesla App shows the correct timezone selected?

I'll flag this as a bug to look at. If you want to debug the code is here: https://github.com/jasonacox/Powerwall-Dashboard/blob/main/tools/tesla-history/tesla-history.py

lukemcguire, 2024-03-23:
Yeah, it seems like it's the right time zone.

I'll probably take a look at the script, but my coding skills are pretty
rusty.

lukemcguire, 2024-03-23:
Screenshot attached.

[image]

mcbirse, 2024-03-24:
Hi @lukemcguire - there seems to be some issue detecting your local timezone correctly when interpreting the `--start` and `--end` times.

The giveaway is below - for the period of history data being retrieved, it should show correctly your timezone offset in the timestamps (i.e -07:00), but it does not.

```
Running for period: [2023-06-26 00:00:00+00:00] - [2023-06-27 23:59:59+00:00] (1 day, 23:59:59s)
```

It's possibly more of a system compatibility issue rather than a bug in the script. Synology NAS (and NAS systems in general) seem to be a problem.

There's an easy workaround. You can actually specify full formed timestamps including the offset with the `--start` and `--end` arguments.

i.e. try this instead (add your timezone offset after the times):

```bash
python3 tesla-history.py --start "2023-06-26 00:00:00-07:00" --end "2023-06-27 23:59:59-07:00"
```

Hopefully that would work for you. Let us know how it goes.

lukemcguire, 2024-03-25:
Interesting idea, but unfortunately still no joy.

```
$ python3 tesla-history.py --start "2023-06-26 00:00:00-07:00" --end "2023-06-27 23:59:59-07:00"
----------------------------------------
Tesla account: XXXX
----------------------------------------
      Site ID: XXXX
    Site type: Powerwall x1
    Site name: My Home
     Timezone: America/Los_Angeles
    Installed: 2023-05-15 14:11:56-07:00
  System time: 2024-03-25 10:49:16-07:00
----------------------------------------
Running for period: [2023-06-26 07:00:00+00:00] - [2023-06-28 06:59:59+00:00] (1 day, 23:59:59s)

Searching InfluxDB for data gaps (power usage)
* Found data gap: [2023-06-26 23:55:00+00:00] - [2023-06-27 07:00:00+00:00] (7:05:00s)
* Found data gap: [2023-06-27 23:55:00+00:00] - [2023-06-28 06:59:59+00:00] (7:04:59s)

Searching InfluxDB for data gaps (grid status)
* None found

Retrieving data for gap: [2023-06-26 23:56:00+00:00] - [2023-06-27 06:59:59+00:00] (7:03:59s)
* Loading daily history: [2023-06-26] (America/Los_Angeles)
* Loading daily history: [2023-06-27] (America/Los_Angeles)
Retrieving data for gap: [2023-06-27 23:56:00+00:00] - [2023-06-28 06:59:59+00:00] (7:03:59s)
* Loading daily history: [2023-06-28] (America/Los_Angeles)

ERROR: No data returned for this date/time range
```

mcbirse, 2024-03-26:
What is your TZ value set to in your tesla-history.conf file? Can you post a copy of the config file.

lukemcguire, 2024-03-26:
Here you go.  Renamed so that Github allows posting.

[tesla-history.conf.txt](https://github.com/jasonacox/Powerwall-Dashboard/files/14751509/tesla-history.conf.txt)

jasonacox, 2024-03-26:
This is so odd. I  have a test system that has a lot of gaps. I found a good 5 day gap:

* Loading daily history: [2024-02-26] (America/Los_Angeles)
Retrieving data for gap: [2024-02-26 02:14:00-08:00] - [2024-03-02 22:56:59-08:00] (5 days, 20:42:59s)
* Loading daily history: [2024-02-27] (America/Los_Angeles)

It filled in the full days... here is where it filled in the gap to the recorded data (the weather data shows the gap):

[image]

@lukemcguire do you have another system that could be used to run the script?  You would just edit the conf file to point to the IP address of your Dashboard host (for Influxdb).

lukemcguire, 2024-03-26:
Huh... good catch @mcbirse. I didn't notice that the timezone call was basically giving +07 instead of -07.  Something's clearly cracker jacks.  

### Weather History Tool

First, let's take a look at the weather-history tool.  

```
$ python3 weather-history.py --start "2023-06-26 00:00:00" --end "2023-06-27 23:59:59"
---------------------------------------------------
Weather History Import Tool for Powerwall-Dashboard
---------------------------------------------------
Running for period: [2023-06-26 00:00:00+00:00] - [2023-06-27 23:59:59+00:00] (1 day, 23:59:59s)
```

It doesn't appear to be getting the timezone data correctly either so I'm not sure why that's working when the tesla-history tool isn't. I can see that the `check_datetime()` function appears to be the same in both tools.

----------  

### Python Shell

Here's the output from the python shell

```
$ python3
Python 3.9.14 (main, Mar 24 2023, 10:04:27) 
[GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from dateutil.parser import isoparse
>>> from dateutil import tz
>>> 
>>> start = isoparse("2023-06-26 00:00:00")
>>> start
datetime.datetime(2023, 6, 26, 0, 0)
>>> print(start)
2023-06-26 00:00:00
>>> type(start)
<class 'datetime.datetime'>
>>> type(start.utcoffset())
<class 'NoneType'>
>>> 
>>> influxtz = tz.gettz("America/Los_Angeles")
>>> influxtz
tzfile('/usr/share/zoneinfo/America/Los_Angeles')
>>> print(influxtz)
tzfile('/usr/share/zoneinfo/America/Los_Angeles')
>>> type(influxtz)
<class 'dateutil.tz.tz.tzfile'>
>>> 
>>> start = start.replace(tzinfo=influxtz)
>>> start
datetime.datetime(2023, 6, 26, 0, 0, tzinfo=tzfile('/usr/share/zoneinfo/America/Los_Angeles'))
>>> print(start)
2023-06-26 00:00:00+00:00
>>> type(start)
<class 'datetime.datetime'>
>>> type(start.utcoffset())
<class 'datetime.timedelta'>
>>> 
>>> utctz = tz.tzutc()
>>> utctz
tzutc()
>>> print(utctz)
tzutc()
>>> type(utctz)
<class 'dateutil.tz.tz.tzutc'>
>>> 
>>> startutc = start.astimezone(utctz)
>>> startutc
datetime.datetime(2023, 6, 26, 0, 0, tzinfo=tzutc())
>>> print(startutc)
2023-06-26 00:00:00+00:00
>>> type(startutc)
<class 'datetime.datetime'>
>>> type(startutc.utcoffset())
<class 'datetime.timedelta'>
>>> 
>>> startlocal = start.astimezone(influxtz)
>>> startlocal
datetime.datetime(2023, 6, 26, 0, 0, tzinfo=tzfile('/usr/share/zoneinfo/America/Los_Angeles'))
>>> print(startlocal)
2023-06-26 00:00:00+00:00
>>> type(startlocal)
<class 'datetime.datetime'>
>>> type(startlocal.utcoffset())
<class 'datetime.timedelta'>
``` 
It's clearly doing something different from your installation and it looks like it's breaking around the `influxtz` command (different class type and what not).

----------

### Solution?

However, I was able to get some progress here thanks to the suggestion from @jasonacox .

> @lukemcguire do you have another system that could be used to run the script? You would just edit the conf file to point to the IP address of your Dashboard host (for Influxdb).

Yes, I was able to run the script from a laptop and sure enough, it found the data gaps and was able to import the data.

```
$ python3 tesla-history.py --start "2023-06-26 00:00:00" --end "2023-06-27 23:59:59"
----------------------------------------
Tesla account: XXXX@XXXX.com
----------------------------------------
      Site ID: XXXX
    Site type: Powerwall x1
    Site name: My Home
     Timezone: America/Los_Angeles
    Installed: 2023-05-15 14:11:56-07:00
  System time: 2024-03-26 11:58:12-07:00
----------------------------------------
Running for period: [2023-06-26 00:00:00-07:00] - [2023-06-27 23:59:59-07:00] (1 day, 23:59:59s)

Searching InfluxDB for data gaps (power usage)
* Found data gap: [2023-06-26 16:55:00-07:00] - [2023-06-27 00:00:00-07:00] (7:05:00s)
* Found data gap: [2023-06-27 16:55:00-07:00] - [2023-06-27 23:59:59-07:00] (7:04:59s)

Searching InfluxDB for data gaps (grid status)
* None found

Retrieving data for gap: [2023-06-26 16:56:00-07:00] - [2023-06-26 23:59:59-07:00] (7:03:59s)
* Loading daily history: [2023-06-26] (America/Los_Angeles)
Retrieving data for gap: [2023-06-27 16:56:00-07:00] - [2023-06-27 23:59:59-07:00] (7:03:59s)
* Loading daily history: [2023-06-27] (America/Los_Angeles)

Writing to InfluxDB
Updating InfluxDB
Done.
``` 

So, while I can get my data imported, the script still isn't working when run on the NAS.  🤷

mcbirse, 2024-04-07:
Thanks for testing and sorry for late reply!

So to me this does not appear to be an bug with the script, but more of an issue with either the host system (Synology NAS) and/or the python dateutil module.

What's the deal with the timezone files on Synology NAS?

python dateutil is interpreting the below as UTC, and not the timezone specified.

```
>>> start = start.replace(tzinfo=influxtz)
>>> start
datetime.datetime(2023, 6, 26, 0, 0, tzinfo=tzfile('/usr/share/zoneinfo/America/Los_Angeles'))
>>> print(start)
2023-06-26 00:00:00+00:00
```

Above should show the offset as -07:00 if it was working correctly.

Can you run the below test on your Synology NAS system and see what is returned? This is using the built-in zoneinfo module (available since python 3.9), instead of dateutil.

```py
from dateutil.parser import isoparse
from zoneinfo import ZoneInfo

start = isoparse("2023-06-26 00:00:00")
start
print(start)

influxtz = ZoneInfo("America/Los_Angeles")
influxtz
print(influxtz)
type(influxtz)

start = start.replace(tzinfo=influxtz)
start
print(start)
```

lukemcguire, 2024-04-07:
It looks like this might resolve the issue. 

```python
$ python3
Python 3.9.14 (main, Mar 24 2023, 10:04:27) 
[GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from dateutil.parser import isoparse
>>> from zoneinfo import ZoneInfo
>>> 
>>> start = isoparse("2023-06-26 00:00:00")
>>> start
datetime.datetime(2023, 6, 26, 0, 0)
>>> print(start)
2023-06-26 00:00:00
>>> 
>>> influxtz = ZoneInfo("America/Los_Angeles")
>>> influxtz
zoneinfo.ZoneInfo(key='America/Los_Angeles')
>>> print(influxtz)
America/Los_Angeles
>>> type(influxtz)
<class 'zoneinfo.ZoneInfo'>
>>> 
>>> start = start.replace(tzinfo=influxtz)
>>> start
datetime.datetime(2023, 6, 26, 0, 0, tzinfo=zoneinfo.ZoneInfo(key='America/Los_Angeles'))
>>> print(start)
2023-06-26 00:00:00-07:00
```

dJOS1475, 2024-05-27:
@lukemcguire I gave up trying to get the history tools working on my DS1621+ and just fired up an Ubuntu Server VM instead. Aside from [Ubuntu breaking some of the PIP install instructions](https://github.com/jasonacox/Powerwall-Dashboard/issues/477), it's been smooth sailing to get everything running.