jasonacox/Powerwall-Dashboard #461: Grafana provisioning environment variable affects performance badly

BuongiornoTexas, 2024-04-05:
**Problem**

Currently `grafana.env` includes a line for a provisioning path that doesn't exist. 

```
GF_PATHS_PROVISIONING=/var/lib/grafana/provisions
```
This causes errors during grafana startup (show up in logs) and causes timeouts, hangs and other problems with grafana dashboards. 
Until one of us has time to sort proper provisioning, the simplest fix is to comment this line out. Grafana dashboard load times and responsiveness improve dramatically. 

**Host System**
 - Intel NUC
 - Debian

jasonacox, 2024-04-06:
Sounds good to me!

BuongiornoTexas, 2024-04-06:
I'll leave it for you to add in the next release - raising a PR for a single `#` seems a bit overkill!

BuongiornoTexas, 2024-04-10:
Closed by #462

closed 2024-04-10 by BuongiornoTexas