jasonacox/Powerwall-Dashboard #471: Upgrade to 4.2.1 failure

jeffkruczynski, 2024-04-28:
**Problem**
While running upgrade.sh, this error occurred:
`Error response from daemon: container 42a94b6d803a859e7cefc6c7755985edcb4dbca1edd759cf2d1b29281b3c19bc: driver "btrfs" failed to remove root filesystem: Failed to destroy btrfs snapshot /volume1/@docker/btrfs/subvolumes for e5a9459c11afd6b7b9bd3ae999c7e570b88a998694379e60323a78f12716d534: device or resource busy`

The full dashboard is still functioning, but the Power Flow widget is not, and data collection isn't happening either.

(I believe I was upgrading from 4.2.0)

**Host System**
 - Synology DSM 7.2.1-69057

**Additional context**
Haven't had any other similar issues when running upgrade.sh.
I've tried re-running upgrade.sh, but still get this error:
```
WARNING: You already have the latest version (v4.3.0).

Upgrade Powerwall-Dashboard from 4.3.0 to 4.3.0
---------------------------------------------------------------------
This script will attempt to upgrade you to the latest version without
removing existing data. A backup is still recommended.

Checking Docker Compose...
INFO: Docker Compose V2 Found: OK to Continue
Upgrade - Proceed? [y/N] y

Resetting Timezone to Default...

Pull influxdb.sql, dashboard.json, telegraf.conf, and other changes...

No local changes to save
Already up to date.

Setting Timezone back to America/Los_Angeles...

Updating Powerwall-Dashboard stack...
Running Docker Compose...
[+] Running 3/0
 ⠿ Container influxdb     Running                                                                                                                                           0.0s
 ⠋ Container pypowerwall  Restart                                                                                                                                           0.0s
 ⠿ Container weather411   Running                                                                                                                                           0.0s
 ⠿ Container grafana      Running                                                                                                                                           0.0s
Error response from daemon: container is marked for removal and cannot be started
```

I did also try updating the pypowerwall image to the latest, which was in itself successful, but the dashboard widget is still not functioning.

Any suggestions? 

(Maybe a side note but: because I have a backup done right before running the upgrade, what would be the right uninstall / reinstall / bring back old data steps?)

jeffkruczynski, 2024-04-28:
Update — resolved this. Found [this thread](https://meta.discourse.org/t/how-to-fix-container-is-marked-for-removal-and-cannot-be-started-without-reboot/136301/2) and following the commands to remove the marked "dead" entry allowed me to re-run upgrade.sh successfully.

closed 2024-04-28 by jeffkruczynski

jasonacox, 2024-04-28:
Hi @jeffkruczynski - It is odd as it seems to be a docker issue on the Synology. And it looks like the pypowerwall is the one with an issue. Some suggestions to try to help us figure out what happened:

```bash
# run Verify
./verify.sh

# check Logs - especially any with errors in the Verify step
docker logs pypowerwall

# recreate pypowerwall - with Force removal
docker stop pypowerwall
docker rm -f pypowerwall

# restart
./compose-dash.sh up -d

# try upgrade
./upgrade.sh
```

jasonacox, 2024-04-28:
Awesome! What commands did you run?

jeffkruczynski, 2024-04-28:
From the site I linked above, I did:
`docker ps -a`

In the list of images, pypowerwall was marked as "Dead." Then I ran
`docker rm <CONTAINER ID>`
where <CONTAINER ID> was the ID number of the dead image.

After doing that, I re-ran upgrade.sh, and all went smoothly!