jasonacox/Powerwall-Dashboard #472: not showing Powerwall Vitals

silvestrob, 2024-04-29:
**Problem**
The Dashboard doesn`t show powerwall vitals. It shows only pw1 

**Screenshots**
[image]

**Host System**
[image] 

Thank for the help you cloud give me.
paolo

jasonacox, 2024-04-30:
Hi @silvestrob - A few questions:

1. Can you tell us what Powerwall Firmware you are running?  You can run the `./vitals.sh` script to get that as well as other important information that may be helpful.
2. What Dashboard version are you running ( have you run `./upgrade.sh` to get latest) - or is this a new installation?
3. Check the direct API call to see what is rendering: http://localhost:8675/pod (replace localhost with the IP address of the Dashboard host).

silvestrob, 2024-04-30:
Hi, Jason thanks for your reply here the information requested:

- firmware 24.4.0 unfortunately ./vitals.sh doesn’t work (command not found)
- after many tries I provided a new installation and I’ve updated the
dashboard to the latest version
- see attached file for pod information

Thanks for your support
Paolo

{"PW1_name": null, "PW1_POD_ActiveHeating": null, "PW1_POD_ChargeComplete": null, "PW1_POD_ChargeRequest": null, "PW1_POD_DischargeComplete": null, "PW1_POD_PermanentlyFaulted": null, "PW1_POD_PersistentlyFaulted": null, "PW1_POD_enable_line": null, "PW1_POD_available_charge_power": null, "PW1_POD_available_dischg_power": null, "PW1_POD_nom_energy_remaining": 10052, "PW1_POD_nom_energy_to_be_charged": null, "PW1_POD_nom_full_pack_energy": 14858, "PW1_PackagePartNumber": "3012170-05-C", "PW1_PackageSerialNumber": "TG122280002C9A", "PW1_pinv_state": "PINV_GridFollowing", "PW1_pinv_grid_state": "Grid_Compliant", "PW1_p_out": 450, "PW1_q_out": 0, "PW1_v_out": 232.4, "PW1_f_out": 50.007, "PW1_i_out": -10.700000000000001, "PW1_energy_charged": 2155903, "PW1_energy_discharged": 1881704, "PW1_off_grid": 0, "PW1_vf_mode": 0, "PW1_wobble_detected": 0, "PW1_charge_power_clamped": 0, "PW1_backup_ready": 1, "PW1_OpSeqState": "Active", "PW1_version": "0fe780c991e052", "time_remaining_hours": 24.75862068965517, "backup_reserve_percent": 20.0, "nominal_full_pack_energy": 14858, "nominal_energy_remaining": 10052}

jasonacox, 2024-04-30:
> ./vitals.sh doesn’t work (command not found)

My bad!  I mean `./verify.sh`

The pod payload show only a single Powerwall is coming back from the Tesla Powerwall Gateway.  That is not expected.  Is it possible that you have two gateways?

```bash
# Have pypowerwall scan for gateways

pip install pypowerwall

python -m pypowerwall scan
```

silvestrob, 2024-04-30:
Hi, thanks no problem.
I'm out of home. I'll back this evening (italy time). I only have one
powerwall and one gateway.

jasonacox, 2024-04-30:
Ha!  I misunderstood your issue.  Bsaed on your srcreenshot, you are seeing what you should see for a single Powerwall (PW1).  A few things to note:

* With the firmware version you are running, Tesla has removed the API data that provides temperature data so that graph will be blank. 
* The rest of the graphs you show are showing the correct data - that is what you should see. It will eventually fill up as time passes:

[image]

silvestrob, 2024-04-30:
ok thanks but I can't see even the three phases value in terms of V,and Hz.
depending on the firmware?

[image: image.png]

jasonacox, 2024-04-30:
Ah! The third phase is hidden - you can unhide it:

[image]

Scroll down to "C" and click unhide:

[image]

Click "Apply" and "Save"

silvestrob, 2024-04-30:
Unfortunately I can’t see even the second. The dashboard shows only pw1 V and HZ not showing Home L1,L2 and L3I’ve tried to un shows but no effectInviato da iPhone

image.png (view on web)
image.png (view on web)

silvestrob, 2024-04-30:
for better undestand what I mean

[image: image.png]

BuongiornoTexas, 2024-04-30:
I lost 3 phase frequency reporting with 24.4 - Now I only get frequency data on the phase the powerwall sits on. I haven't checked the API to see if this can be picked up elsewhere.

jasonacox, 2024-05-01:
Ugh. Thanks @BuongiornoTexas. Another casualty of the Firmware upgrade.  The raw data is being sourced from:

* http://localhost:8675/freq
* http://localhost:8675/pod

(replace localhost with the IP address of your Dashboard host if not local)

BuongiornoTexas, 2024-05-01:
It's possible this is due to variations in multi-phase supply between countries - US installations may well continue to get 3 phase data (120V/phase?), while Europe, Australia and others may not (240V/phase).

freq dump:

```json
{
  "PW1_name": null,
  "PW1_PINV_Fout": 50.061,
  "PW1_PINV_VSplit1": null,
  "PW1_PINV_VSplit2": null,
  "PW1_PackagePartNumber": "XXXXX",
  "PW1_PackageSerialNumber": "XXXXXX",
  "PW1_p_out": 0,
  "PW1_q_out": 340,
  "PW1_v_out": 238.8,
  "PW1_f_out": 50.061,
  "PW1_i_out": -0.2,
  "grid_status": 1
}
```
and pod

```json
{
  "PW1_name": null,
  "PW1_POD_ActiveHeating": null,
  "PW1_POD_ChargeComplete": null,
  "PW1_POD_ChargeRequest": null,
  "PW1_POD_DischargeComplete": null,
  "PW1_POD_PermanentlyFaulted": null,
  "PW1_POD_PersistentlyFaulted": null,
  "PW1_POD_enable_line": null,
  "PW1_POD_available_charge_power": null,
  "PW1_POD_available_dischg_power": null,
  "PW1_POD_nom_energy_remaining": 15127,
  "PW1_POD_nom_energy_to_be_charged": null,
  "PW1_POD_nom_full_pack_energy": 15127,
  "PW1_PackagePartNumber": "XXXXXXXXXX",
  "PW1_PackageSerialNumber": "XXXXXXXXXXX",
  "PW1_pinv_state": "PINV_GridFollowing",
  "PW1_pinv_grid_state": "Grid_Compliant",
  "PW1_p_out": -20,
  "PW1_q_out": 330,
  "PW1_v_out": 238.60000000000002,
  "PW1_f_out": 49.947,
  "PW1_i_out": -0.2,
  "PW1_energy_charged": 3380972,
  "PW1_energy_discharged": 2842412,
  "PW1_off_grid": 0,
  "PW1_vf_mode": 0,
  "PW1_wobble_detected": 0,
  "PW1_charge_power_clamped": 0,
  "PW1_backup_ready": 1,
  "PW1_OpSeqState": "Active",
  "PW1_version": "0fe780c991e052",
  "time_remaining_hours": 36.9176327028676,
  "backup_reserve_percent": 20,
  "nominal_full_pack_energy": 15127,
  "nominal_energy_remaining": 15127
}
```

silvestrob, 2024-05-01:
Thanks, here my value at freq and pod

{
  "PW1_name": null,
  "PW1_PINV_Fout": 49.981,
  "PW1_PINV_VSplit1": null,
  "PW1_PINV_VSplit2": null,
  "PW1_PackagePartNumber": "3012170-05-C",
  "PW1_PackageSerialNumber": "TG122280002C9A",
  "PW1_p_out": -50,
  "PW1_q_out": -10,
  "PW1_v_out": 234.8,
  "PW1_f_out": 49.981,
  "PW1_i_out": 0,
  "grid_status": 1
}

{
  "PW1_name": null,
  "PW1_POD_ActiveHeating": null,
  "PW1_POD_ChargeComplete": null,
  "PW1_POD_ChargeRequest": null,
  "PW1_POD_DischargeComplete": null,
  "PW1_POD_PermanentlyFaulted": null,
  "PW1_POD_PersistentlyFaulted": null,
  "PW1_POD_enable_line": null,
  "PW1_POD_available_charge_power": null,
  "PW1_POD_available_dischg_power": null,
  "PW1_POD_nom_energy_remaining": 9290,
  "PW1_POD_nom_energy_to_be_charged": null,
  "PW1_POD_nom_full_pack_energy": 14861,
  "PW1_PackagePartNumber": "3012170-05-C",
  "PW1_PackageSerialNumber": "TG122280002C9A",
  "PW1_pinv_state": "PINV_GridFollowing",
  "PW1_pinv_grid_state": "Grid_Compliant",
  "PW1_p_out": -90,
  "PW1_q_out": 0,
  "PW1_v_out": 234.7,
  "PW1_f_out": 49.966,
  "PW1_i_out": -0.2,
  "PW1_energy_charged": 2161374,
  "PW1_energy_discharged": 1887176,
  "PW1_off_grid": 0,
  "PW1_vf_mode": 0,
  "PW1_wobble_detected": 0,
  "PW1_charge_power_clamped": 0,
  "PW1_backup_ready": 1,
  "PW1_OpSeqState": "Active",
  "PW1_version": "0fe780c991e052",
  "time_remaining_hours": 22.4531722054381,
  "backup_reserve_percent": 20,
  "nominal_full_pack_energy": 14861,
  "nominal_energy_remaining": 9290
}

jasonacox, 2024-05-02:
@BuongiornoTexas and @silvestrob - Do you see the same data at http://localhost:8675/api/solar_powerwall ? If you get more data on that call, we may be able to capture that.

With the new Firmware, in the US we no longer get the two phases (VSplit, each 120V) but get the aggregate (240v) only:

```json
{
"PW1_name": null,
"PW1_PINV_Fout": 60.021,
"PW1_PINV_VSplit1": null,
"PW1_PINV_VSplit2": null,
"PW1_PackagePartNumber": "2012170-25-E",
"PW1_PackageSerialNumber": "XXXXXXXXXXX",
"PW1_p_out": 590,
"PW1_q_out": -10,
"PW1_v_out": 240.20000000000002,
"PW1_f_out": 60.021,
"PW1_i_out": -12.700000000000001,
"PW2_name": null,
"PW2_PINV_Fout": 60.021,
"PW2_PINV_VSplit1": null,
"PW2_PINV_VSplit2": null,
"PW2_PackagePartNumber": "3012170-05-B",
"PW2_PackageSerialNumber": "XXXXXXXXXXX",
"PW2_p_out": 560,
"PW2_q_out": 0,
"PW2_v_out": 240.4,
"PW2_f_out": 60.021,
"PW2_i_out": -12.3,
"grid_status": 1
}
```

BuongiornoTexas, 2024-05-02:
Unfortunately, that's a no go for people with non-Telsa solar (me and many others!). If this solar_powerwall does offer additional data, then we may want to offer an either/or for users. While I'd like to get more frequency info, I definitely don't want to lose the limited data I get at the moment!

silvestrob, 2024-05-02:
Unfortunately the return was "null"

It means that I will settle for the data that are still available.
p