jasonacox/Powerwall-Dashboard #473: Loss of data logging

shareef777, 2024-04-29:
**Problem**
A failed power wall required a replacement be installed and Tesla techs re-installed the existing inverter with a different wiring layout. Now the rewired inverter is not showing ANY data.

**To Reproduce**
The screenshot shows Inverter2 not reporting any power. However, when looking at the total power being generated, the metric reported by Inverter1 is not enough (indicating that Inverter2 IS generating power).

**Screenshots**
[image]

**Host System**
Mac Mini M1
macOS 14.4.1

jasonacox, 2024-04-30:
Hi @shareef777 

The inverter data is coming from the `strings` API (http://localhost:8675/strings) - The Dashboard then sums up every set of string values to determine the inverter power.

Did the technicians happen to upgrade the firmware on your system? If so, I'm sad to report that the new firmware removes a key API (`/vitals`) that was used to assemble the string data. We had to move to the http://localhost:8675/api/solar_powerwall API which only includes the first set of strings. In your case, you had two strings so with the upgraded firmware, you will only get the first set.

Check the URL's above. It could be something else, but I have a hunch it was a firmware upgrade that got you.

shareef777, 2024-04-30:
Thanks for the feedback.

First link (/strings) responds back with:

> {
>     "A": {
>         "Connected": true,
>         "Current": 4.88,
>         "Power": 1415,
>         "Voltage": 289.3
>     },
>     "B": {
>         "Connected": false,
>         "Current": 0.03,
>         "Power": 0,
>         "Voltage": -0.1999999999999993
>     },
>     "C": {
>         "Connected": true,
>         "Current": 6.98,
>         "Power": 1560,
>         "Voltage": 223.70000000000002
>     },
>     "D": {
>         "Connected": true,
>         "Current": 7.55,
>         "Power": 1688,
>         "Voltage": 224.10000000000002
>     }
> }

The second also responds, with:

> {"pvac_status": {"state": "PVAC_Active", "disabled": false, "disabled_reasons": [], "grid_state": "Grid_Compliant", "inv_state": "INV_Grid_Connected", "v_out": 239.60000000000002, "f_out": 59.980000000000004, "p_out": 4710, "q_out": -30, "i_out": 19.240000000000002, "string_vitals": [{"string_id": 1, "connected": true, "measured_voltage": 289.3, "current": 4.9, "measured_power": 1420}, {"string_id": 2, "connected": false, "measured_voltage": -0.09999999999999964, "current": 0.03, "measured_power": 0}, {"string_id": 3, "connected": true, "measured_voltage": 224, "current": 6.97, "measured_power": 1560}, {"string_id": 4, "connected": true, "measured_voltage": 224.3, "current": 7.55, "measured_power": 1689}]}, "pvs_status": {"state": "PVS_Active", "disabled": false, "enable_output": true, "v_ll": 239.70000000000002, "self_test_state": "PVS_SelfTestOff"}, "pv_power_limit": 5762.340185821082, "power_status_setpoint": "on", "pvac_alerts": {"LastRxTime": "2024-04-30T10:56:44.186725-05:00", "ReceivedMuxBitmask": 1, "PVAC_alertMatrixIndex": 0, "PVAC_a001_inv_L1_HW_overcurrent": false, "PVAC_a002_inv_L2_HW_overcurrent": false, "PVAC_a003_inv_HVBus_HW_overvoltage": false, "PVAC_a004_pv_HW_CMPSS_OC_STGA": false, "PVAC_a005_pv_HW_CMPSS_OC_STGB": false, "PVAC_a006_pv_HW_CMPSS_OC_STGC": false, "PVAC_a007_pv_HW_CMPSS_OC_STGD": false, "PVAC_a008_inv_HVBus_undervoltage": false, "PVAC_a009_SwAppBoot": false, "PVAC_a010_inv_AC_overvoltage": false, "PVAC_a011_inv_AC_undervoltage": false, "PVAC_a012_inv_AC_overfrequency": false, "PVAC_a013_inv_AC_underfrequency": false, "PVAC_a014_PVS_disabled_relay": false, "PVAC_a015_pv_HW_Allegro_OC_STGA": false, "PVAC_a016_pv_HW_Allegro_OC_STGB": false, "PVAC_a017_pv_HW_Allegro_OC_STGC": false, "PVAC_a018_pv_HW_Allegro_OC_STGD": false, "PVAC_a019_ambient_overtemperature": false, "PVAC_a020_dsp_overtemperature": false, "PVAC_a021_dcac_heatsink_overtemperature": false, "PVAC_a022_mppt_heatsink_overtemperature": false, "PVAC_a023_unused": false, "PVAC_a024_PVACrx_Command_mia": false, "PVAC_a025_PVS_Status_mia": false, "PVAC_a026_inv_AC_peak_overvoltage": false, "PVAC_a027_inv_K1_relay_welded": false, "PVAC_a028_inv_K2_relay_welded": false, "PVAC_a029_pump_faulted": false, "PVAC_a030_fan_faulted": false, "PVAC_a031_VFCheck_OV": false, "PVAC_a032_VFCheck_UV": false, "PVAC_a033_VFCheck_OF": false, "PVAC_a034_VFCheck_UF": false, "PVAC_a035_VFCheck_RoCoF": false, "PVAC_a036_inv_lost_iL_control": false, "PVAC_a037_PVS_processor_nERROR": false, "PVAC_a038_inv_failed_xcap_precharge": false, "PVAC_a039_inv_HVBus_SW_overvoltage": false, "PVAC_a040_pump_correction_saturated": false, "PVAC_a041_excess_PV_clamp_triggered": false, "PVAC_a042_mppt_curve_scan_completed": false, "PVAC_a043_fan_speed_mismatch_detected": false, "PVAC_a044_fan_deadband_toggled": false, "PVAC_a045_max_thermal_current_exceeded": false}, "pvs_alerts": {"LastRxTime": "2024-04-30T10:56:44.06576-05:00", "ReceivedMuxBitmask": 0, "PVS_a001_WatchdogReset": false, "PVS_a002_SW_App_Boot": false, "PVS_a003_V12vOutOfBounds": false, "PVS_a004_V1v5OutOfBounds": false, "PVS_a005_VAfdRefOutOfBounds": false, "PVS_a006_GfOvercurrent300": false, "PVS_a007_V12vPowerOutOfBounds": false, "PVS_a008_UNUSED_8": false, "PVS_a009_GfOvercurrent030": false, "PVS_a010_PvIsolationTotal": false, "PVS_a011_PvIsolationStringA": false, "PVS_a012_PvIsolationStringB": false, "PVS_a013_PvIsolationStringC": false, "PVS_a014_PvIsolationStringD": false, "PVS_a015_SelfTestGroundFault": false, "PVS_a016_ESMFault": false, "PVS_a017_MciStringA": false, "PVS_a018_MciStringB": true, "PVS_a019_MciStringC": false, "PVS_a020_MciStringD": false, "PVS_a021_RapidShutdown": false, "PVS_a022_Mci1SignalLevel": false, "PVS_a023_Mci2SignalLevel": false, "PVS_a024_Mci3SignalLevel": false, "PVS_a025_Mci4SignalLevel": false, "PVS_a026_Mci1PvVoltage": false, "PVS_a027_Mci2PvVoltage": false, "PVS_a028_systemInitFailed": false, "PVS_a029_PvArcFault": false, "PVS_a030_VDcOv": false, "PVS_a031_Mci3PvVoltage": false, "PVS_a032_Mci4PvVoltage": false, "PVS_a033_dataException": false, "PVS_a034_PeImpedance": false, "PVS_a035_PvArcDetected": false, "PVS_a036_PvArcLockout": false, "PVS_a037_PvArcFaultData1": false, "PVS_a038_PvArcFault_SelfTest": false, "PVS_a039_SelfTestRelayFault": false, "PVS_a040_LEDIrrationalFault": false, "PVS_a041_MciPowerSwitch": false, "PVS_a042_MciPowerFault": false, "PVS_a043_InactiveUnsafePvStrings": false, "PVS_a044_FaultStatePvStringSafety": false, "PVS_a045_RelayCoilIrrationalFault": false, "PVS_a046_RelayCoilIrrationalLockout": false, "PVS_a047_AcSensorIrrationalFault": false, "PVS_a048_DcSensorIrrationalFault": false, "PVS_a049_arcSignalMibspiHealth": false, "PVS_a050_RelayCoilIrrationalWarning": false, "PVS_a051_DcBusShortCircuitDetected": false, "PVS_a052_PvArcFault_PreSelfTest": false, "PVS_a053_PvA [... cut ...]
A bit of backstory. I have two PW+ installed, each with a set of inverters and 3 strings on each inverter. The first one (PW1) failed, though the inverter continued to work. Tesla came out and replaced the failed PW1 (kept the same inverter). Yes, they did say they completed a firmware upgrade as part of the repair, but not sure if that was done on both systems (had assumed it was on only the one they replaced). Does the fact that I get a response to both endpoints suggest that I'm running two different firmwares?

jasonacox, 2024-05-01:
You can see what firmware you have by running the `./verify.sh` script.

If you have two different gateways, you should see them both doing this:

```bash
# Have pypowerwall scan for gateways

pip install pypowerwall

python -m pypowerwall scan
```

If you do indeed have multiple gateways, you could install two instances of Powerwall Dashboard (one for each gateway) to get all the data. Ideally we would update this project to better support multiple gateways. I keep hoping that someone with multiple gateways will help us figure it out. 😁   

On the other hand, if you only see one gateway, unfortunately it seems it's Firmware was upgraded so you will only get the first set of strings. There is a way to get all of them but it requires a more elaborate setup using https://github.com/ygelfand/go-powerwall. Eventually I would like to add that to the base setup but it requires a dual-homed device able to communicate with the Powerwall access point.

shareef777, 2024-05-01:
Sorry if I wasn't clear, I only have one gateway. I was under the impression that the first set of strings shown would be per inverter (not per gateway). Makes sense now, thanks for the clarification.

commit ef3c70b9 (2024-03-03): Cloud grid status fix #473
closed 2024-05-01 by shareef777