jasonacox/Powerwall-Dashboard #477: Additional Python modules for some Tools can no longer be installed via PIP as instructed

dJOS1475, 2024-05-27:
**Problem**
The various instructions to install prerequisites for ecowitt and tesla history (and maybe others) are no longer correct on Ubuntu 22.x +

**To Reproduce**
Attempt to install python modules: dateutil teslapy influxdb etc

`pip install python-dateutil teslapy influxdb`

**Screenshots**

[image]

**Host System**
 - OS Ubuntu 22.x +

**Additional context**
The workaround is to add `--break-system-packages`
eg
`pip install python-dateutil teslapy influxdb --break-system-packages`

jasonacox, 2024-05-27:
Thanks @dJOS1475 

I flagged this for documentation so we can update with those helps. However, using --break-system-packages is probably not the best option for everyone. There are a few other options:

## Docker

We could have the tools run inside of docker with its own python environment already set up.  @mcbirse Didn't we already set this up?  Perhaps we change the tool runs to be `docker run` instead.

## Python Virtual Environment

A recommended approach (when you can't `apt install` packages which is most of them) is to create a python virtual environment:

```bash
sudo apt install python3-venv
python3 -m venv .venv
source .venv/bin/activate
pip install python-dateutil teslapy influxdb 
```

You would need to `source .venv/bin/activate` for any session that needed to use those installed libraries (eg. put it in ~/.bashrc or your cron script).

dJOS1475, 2024-05-27:
Cheers, I was able to `apt install` everything except teslapy and had to restore to the brute force method for that. I run my dashboard in a dedicated vm, so I wasn’t bothered if I broke it.