jasonacox/Powerwall-Dashboard #478: Power Flow Always Shows Off-Grid - Solar Only

lsgc123, 2024-06-02:
First of all, I do appreciate @jasonacox to provide us this awesome tool!

I am using Solar-Only dashboard and it works well except the Power Flow animation and Energy Usage chart.

[image]
For the above Power Flow animation, my solar system generates 0.7 kW and my home uses 0.5 kW. That is, I am sending around 0.2 kW to the Grid. However, this animation indicates the Grid is off and no interaction between Solar and Grid. (Nothing between Home and Grid when consumption is bigger than production, either). Also, Grid Status is RED all the time on the Grid Status.

When I checked the Key power data(http://localhost:8675/csv), it says '-137.65,522.35,660.00,0.00,5.00' which I sent 137.65 kW to Grid.
Is there any way to fix this? In fact, it is just a minor issue and does not affect to the main purpose of this tool, but I am just wondering if I could fix this on my side. 

I can still see the correct data from the Current State as below(The data is different because I took this screenshot later): 
[image]

Please advise.

Thank you.

jasonacox, 2024-06-02:
Hi @lsgc123 , thanks for opening this! It looks like the animation things that your grid is offline, but clearly it is not. I would love to fix this.

* First, are you using the latest version (v4.3.2 )?  You can see what you have installed by looking at the VERSION file. If needed,  I would recommend upgrading (run `upgrade.sh`) as we have upgraded a few things along the way to help with the Solar Only setups (https://github.com/jasonacox/Powerwall-Dashboard/discussions/475 https://github.com/jasonacox/Powerwall-Dashboard/issues/437 https://github.com/jasonacox/Powerwall-Dashboard/pull/438 )
* You can also update your `pypowerwall.env` file and set `PW_STYLE=solar` and run `./compose-dash.sh up -d` it should remove the Powerwall Icon from the power flow animation.  See [instructions](https://github.com/jasonacox/Powerwall-Dashboard/issues/437#issuecomment-1975481959). I just realized that setup currently doesn't do this by default (TODO: fix).
* If none of the above works, I would like to get your help to find out where it is breaking down. Here are the URLs used by the animation - can you post these?

  * http://localhost:8675/api/sitemaster
  * http://localhost:8675/api/powerwalls
  * http://localhost:8675/api/system_status/grid_status 
  * http://localhost:8675/api/system_status/soe
  * http://localhost:8675/api/auth/toggle/supported
  * http://localhost:8675/api/troubleshooting/problems
  * http://localhost:8675/version

lsgc123, 2024-06-02:
Hi @jasonacox ,

Thank you for the reply.

I confirmed the version I installed was 4.3.2
[image]

Applied the change for the PyPowerWall and confirmed the PowerWall icon was gone but still same istuation for Grid.
[image]

Below is the information from my local server:

sitemaster: 
{"status": "StatusUp", "running": true, "connected_to_tesla": true, "power_supply_mode": false, "can_reboot": "Yes"}

powerwalls: 
{"enumerating": false, "updating": false, "checking_if_offgrid": false, "running_phase_detection": false, "phase_detection_last_error": "no phase information", "bubble_shedding": false, "on_grid_check_error": "on grid check not run", "grid_qualifying": false, "grid_code_validating": false, "phase_detection_not_available": true, "powerwalls": [{"Type": "", "PackagePartNumber": "2012170-25-E", "PackageSerialNumber": "TG1234567890G1", "type": "SolarPowerwall", "grid_state": "Grid_Uncompliant", "grid_reconnection_time_seconds": 0, "under_phase_detection": false, "updating": false, "commissioning_diagnostic": {"name": "Commissioning", "category": "InternalComms", "disruptive": false, "inputs": null, "checks": [{"name": "CAN connectivity", "status": "fail", "start_time": "2023-12-16T08:34:17.3068631-08:00", "end_time": "2023-12-16T08:34:17.3068696-08:00", "message": "Cannot perform this action with site controller running. From landing page, either \"STOP SYSTEM\" or \"RUN WIZARD\" to proceed.", "results": {}, "debug": {}, "checks": null}, {"name": "Enable switch", "status": "fail", "start_time": "2023-12-16T08:34:17.306875474-08:00", "end_time": "2023-12-16T08:34:17.306880724-08:00", "message": "Cannot perform this action with site controller running. From landing page, either \"STOP SYSTEM\" or \"RUN WIZARD\" to proceed.", "results": {}, "debug": {}, "checks": null}, {"name": "Internal communications", "status": "fail", "start_time": "2023-12-16T08:34:17.306886099-08:00", "end_time": "2023-12-16T08:34:17.306891223-08:00", "message": "Cannot perform this action with site controller running. From landing page, either \"STOP SYSTEM\" or \"RUN WIZARD\" to proceed.", "results": {}, "debug": {}, "checks": null}, {"name": "Firmware up-to-date", "status": "fail", "start_time": "2023-12-16T08:34:17.306896598-08:00", "end_time": "2023-12-16T08:34:17.306901723-08:00", "message": "Cannot perform this action with site controller running. From landing page, either \"STOP SYSTEM\" or \"RUN WIZARD\" to proceed.", "results": {}, "debug": {}, "checks": null}], "alert": false}, "update_diagnostic": {"name": "Firmware Update", "category": "InternalComms", "disruptive": true, "inputs": null, "checks": [{"name": "Solar Inverter firmware", "status": "not_run", "start_time": null, "end_time": null, "progress": 0, "results": null, "debug": null, "checks": null}, {"name": "Solar Safety firmware", "status": "not_run", "start_time": null, "end_time": null, "progress": 0, "results": null, "debug": null, "checks": null}, {"name": "Grid code", "status": "not_run", "start_time": null, "end_time": null, "progress": 0, "results": null, "debug": null, "checks": null}, {"name": "Powerwall firmware", "status": "not_run", "start_time": null, "end_time": null, "progress": 0, "results": null, "debug": null, "checks": null}, {"name": "Battery firmware", "status": "not_run", "start_time": null, "end_time": null, "progress": 0, "results": null, "debug": null, "checks": null}, {"name": "Inverter firmware", "status": "not_run", "start_time": null, "end_time": null, "progress": 0, "results": null, "debug": null, "checks": null}, {"name": "Grid code", "status": "not_run", "start_time": null, "end_time": null, "progress": 0, "results": null, "debug": null, "checks": null}], "alert": false}, "bc_type": null, "in_config": true}, {"Type": "", "PackagePartNumber": "3012170-05-B", "PackageSerialNumber": "TG1234567890G1", "type": "ACPW", "grid_state": "Grid_Uncompliant", "grid_reconnection_time_seconds": 0, "under_phase_detection": false, "updating": false, "commissioning_diagnostic": {"name": "Commissioning", "category": "InternalComms", "disruptive": false, "inputs": null, "checks": [{"name": "CAN connectivity", "status": "fail", "start_time": "2023-12-16T08:34:17.320856307-08:00", "end_time": "2023-12-16T08:34:17.320940302-08:00", "message": "Cannot perform this action with site controller running. From landing page, either \"STOP SYSTEM\" or \"RUN WIZARD\" to proceed.", "results": {}, "debug": {}, "checks": null}, {"name": "Enable switch", "status": "fail", "start_time": "2023-12-16T08:34:17.320949301-08:00", "end_time": "2023-12-16T08:34:17.320955301-08:00", "message": "Cannot perform this action with site controller running. From landing page, either \"STOP SYSTEM\" or \"RUN WIZARD\" to proceed.", "results": {}, "debug": {}, "checks": null}, {"name": "Internal communications", "status": "fail", "start_time": "2023-12-16T08:34:17.320960676-08:00", "end_time": "2023-12-16T08:34:17.320966176-08:00", "message": "Cannot perform this action with site controller running. From landing page, either \"STOP SYSTEM\" or \"RUN WIZARD\" to proceed.", "results": {}, "debug": {}, "checks": null}, {"name": "Firmware up-to-date", "status": "fail", "start_time": "2023-12-16T08:34:17.32097155-08:00", "end_time": "2023-12-16T08:34:17.3209768-08:00", "message": "Cannot perform this action with site controller running. From landing page, either \"STOP SYSTEM\" or \"RUN WIZARD\" to proceed.", "results": {}, "debug": {}, "checks": null}], "alert": false}, "update_di [... cut ...]  "debug": null, "checks": null}, {"name": "Grid code", "status": "not_run", "start_time": null, "end_time": null, "progress": 0, "results": null, "debug": null, "checks": null}], "alert": false}}, "msa": null, "states": null}

grid_status(suspicious grid_services_active: false):
{"grid_status": "SystemIslandedActive", "grid_services_active": false}

SOE: 
{"percentage": 0.0}

auth supported:
{"toggle_auth_supported": true}

problems:
{"problems": []}

version:
{"version": "SolarOnly", "vint": 0}

jasonacox, 2024-06-02:
Ok, thanks @lsgc123 - 

> grid_status(suspicious grid_services_active: false):
{"grid_status": "SystemIslandedActive", "grid_services_active": false}

Yes, this is a problem. I'll dig in to why. This would cause the animation to think it is in island mode (grid down).

Which dashboard.json file are you using?  The note below the powerwall "Firmware" should say "Solar Only". You might try [dashboard-solor-only.json](https://github.com/jasonacox/Powerwall-Dashboard/blob/main/dashboards/dashboard-solar-only.json) if you haven't already. It won't fix the grid-down issue but might remove the N/A "Firmware" note.

jasonacox, 2024-06-02:
I might have found it.  When the system polls the Telsa cloud for status, if the grid_status is not "Active" it assumes you are in island mode. I created a beta proxy if you are willing to test it out for me. It has the new logic in it and also adds more debug information.  Here is what you will need to do:

Edit the `powerwall.yml` file and replace the image name for pypowerwall to `jasonacox/pypowerwall:0.10.0t58-beta7`

```
# from this
    pypowerwall:
        image: jasonacox/pypowerwall:0.9.1t57
        container_name: pypowerwall
        hostname: pypowerwall
        restart: unless-stopped

# to this
    pypowerwall:
        image: jasonacox/pypowerwall:0.10.0t58-beta7
        container_name: pypowerwall
        hostname: pypowerwall
        restart: unless-stopped
```

Same that and then run this:

```
./compose-dash.sh up -d 
```

If nothing changes, I need you to turn on debug.  Edit the `pypowerwall.env` file and change:

```
# this
PW_DEBUG=no

# to this
PW_DEBUG=yes
```
 
Load in the new settings:

```
./compose-dash.sh up -d 
# Run this to see logs - ^C to end
docker logs pypowerwall -f

```

lsgc123, 2024-06-02:
Hi @jasonacox 

I am pretty sure I am using the dashboard, dashboard-solar-only.json.
Since the word 'Firmware' is a part of pypowerwall, it does not matter with the dashboard, I believe.

Unfortunately, neither worked at this moment, so I have turned on the Debug mode. Should I send the logs to you or you can see the logs on your side?

jasonacox, 2024-06-02:
As for the logs, go to http://localhost:8675/api/system_status/grid_status and then capture the section of the logs that should look something like this:

```
06/02/2024 01:45:02 PM [pypowerwall.cloud.pypowerwall_cloud] [DEBUG]  -- cloud: get_api_system_status_grid_status: {
    "response": {
        "solar_power": 7690,
        "percentage_charged": 100,
        "backup_capable": true,
        "battery_power": 0,
        "load_power": 769,
        "grid_status": "Active",
        "grid_services_active": false,
        "grid_power": -6921,
        "grid_services_power": 0,
        "generator_power": 0,
        "island_status": "on_grid",
        "storm_mode_active": false,
        "timestamp": "2024-06-02T13:45:02-07:00",
        "wall_connectors": []
    }
}
```

or this

```
06/02/2024 01:48:42 PM [pypowerwall.fleetapi.pypowerwall_fleetapi] [DEBUG]  -- fleetapi: get_api_system_status_grid_status: {'solar_power': 3700, 'percentage_charged': 100, 'backup_capable': True, 'battery_power': 0, 'load_power': 903, 'grid_status': 'Active', 'grid_services_active': False, 'grid_power': -2797, 'grid_services_power': 0, 'generator_power': 0, 'island_status': 'on_grid', 'storm_mode_active': False, 'timestamp': '2024-06-02T13:48:42-07:00', 'wall_connectors': []}
```

On the dashboard, we updated that dashboard so you may want to install the newer one.

Thanks for your help with this.

lsgc123, 2024-06-02:
Here we go

06/02/2024 05:59:35 PM [pypowerwall.cloud.pypowerwall_cloud] [DEBUG]  -- cloud: get_api_system_status_grid_status: {
    "response": {
        "solar_power": 1230,
        "load_power": 722.0999755859375,
        "grid_status": "Unknown",
        "grid_services_active": false,
        "grid_power": -507.9000244140625,
        "grid_services_power": 0,
        "generator_power": 0,
        "island_status": "island_status_unknown",
        "storm_mode_active": false,
        "timestamp": "2024-06-02T17:59:35-04:00",
        "wall_connectors": [
            {
                "din": "1734412-02-D--B7S23353J00983",
                "wall_connector_state": 2,
                "wall_connector_fault_state": 2,
                "wall_connector_power": 0,
                "ocpp_status": 1
            }
        ]
    }
}

jasonacox, 2024-06-02:
Thanks!  

> "grid_status": "Unknown",

Well, there it is. :)  Okay, easy fix.  I just pushed a new version. 

Edit the `powerwall.yml` file and replace the image name for pypowerwall to `jasonacox/pypowerwall:0.10.0t58-beta8`

```
...
    pypowerwall:
        image: jasonacox/pypowerwall:0.10.0t58-beta8
        container_name: pypowerwall
...
```

Then...

```
./compose-dash.sh up -d 
```

If that fixes it, you can edit `pypowerwall.env` and set `PW_DEBUG=no` to save your disk space.

lsgc123, 2024-06-03:
Works perfectly! Great!

Thank you very much!
Still shows Grid Status(which is a part of Dashboard not pypowerwall) as off grid(Red Graph), but it does not matter.

jasonacox, 2024-06-03:
Oh, I'm so glad you noticed that!  I took another look and sure enough, there is another place where grid_status is getting set and would ignore the Solar Only "Unknown" response. I've updated it and you can try using this:

```
jasonacox/pypowerwall:0.10.0t58-beta9
```

Thanks for your help with this! I'll get this included in the next release for our other solar-only members.

Thank you! 🙏